awslabs/amazon-kinesis-video-streams-producer-sdk-java #3: No Lib Folder

arunned, 2017-12-02:
Hello, As per the readme note, there should be a library folder with producerJNI library. But this is missing. When I tried compiling the C++ project in macOS I got an error. Hence not able to continue. Please direct me to how to get this JNI file to run this. 

ReadMe Notes:
Note: NativeLibraryPath must contain your "KinesisVideoProducerJNI" library. (File name depends on your Operating System, "libKinesisVideoProducerJNI.so" for Linux, "libKinesisVideoProducerJNI.dylib" for Mac OS, "KinesisVideoProducerJNI.dll" for Windows)

MushMal, 2017-12-02:
Hi there,

We will be adding the pre-built libraries for a few platforms shortly. Meanwhile, could we ask you to report an issue in the amazon-kinesis-video-streams-producer-sdk-cpp/issues for the failed build on Mac. We made some fixes to the install script to handle a compilation error on later Mac OS versions with curl library and updated the build-time dependencies in the Readme.

arunned, 2017-12-03:
great. Thanks for the update. Will try compiling it again and log an issue if I still encounter this problem. Can't wait to start using kinesis video stream ingest.

closed 2017-12-08 by zhiyua-git