awslabs/amazon-kinesis-video-streams-producer-sdk-java #5: Integration for pi Camera

bvasan, 2017-12-05:
Hi,
I got the Demo App in Java SDK working by building the libKinesisVideoProducerJNI.dylib on mac and placing it in the lib directory. I was able to successfully stream the frame files (h264) to kinesis video. I believe the current implementation of the SDK supports only frame files as the MediaSource. Are there plans for supporting Raspberry Pi Camera (picamera) natively in the Java SDK? If so, what is the ETA?

I couldn't attend the specific workshop @ reinvent where AWS showed streaming the data from Raspberry Pi to Kinesis video realtime. Does it mean that it is already implemented in C++ sdk? Could you please share the details?

Thank you for the support. 

Thanks
Babu Srinivasan

MushMal, 2017-12-06:
Hi Babu,

We are super-thrilled you were able to stream using Java SDK. 

We are working on publishing updates to the C++ SDK build script that will build the GStreamer demo app for Raspberry Pi. Basically, when you launch the build scrip it will build the dependencies and then will build the core libraries then the sample application - in this case the GStreamer demo app.

We are also interested in learning about your scenarios. Are you interested in streaming form the camera (through GStreamer) from Java?

bvasan, 2017-12-06:
Hi 
Thank you for the quick response. Great to know the C++ SDK update for Raspberry Pi is on the way. C++ or Java does not matter to me, as long as I am able to have the producer application in Raspberry Pi stream the video from the camera attached to Raspberry Pi to Kinesis Video.  I just need the straight forward producer application that directly take the feed from the camera attached to Raspberry Pi and make it available in Kinesis Video. The main use case is to have Amazon Rekognition video source the stream from Kinesis video to perform real time identification of faces and match it to the known faces in the Rekognition collection. Based on the results, send notifications (lambda, sns etc.). The flow is this:-
Camera+RaspberryPi3 ==> Kinesis Video ==> Amazon Rekognition ==> S3 ==> Lambda ==> SNS/SES

Thanks
Babu Srinivasan

MushMal, 2017-12-06:
Babu, we just pushed an update to the C++ SDK which should help with Raspberry PI. Please give it a try.

closed 2017-12-06 by MushMal