awslabs/amazon-kinesis-video-streams-producer-sdk-java #6: Undefined methods after

backdoorcodr, 2017-12-09:
Hello,

I have followed the README.md of the [cpp-sdk](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp) to build the C++ wrapper for JNI. Access key and access key id have also been specified to my JVM environment. 

I download this java project and it starts to download the dependencies specified in pom.xml. However, the after maven have downloaded necessary libraries and built the workspace I get a number of methods to be undefined. Screenshot has been attached. Are there any missing dependencies in the pom.xml? How this can be resolved? 
[image]

zhiyua-git, 2017-12-10:
Hi,

This issue is likely caused by missing lombok dependency. You can check your pom.xml and try to update your maven dependencies.

For you Eclipse case, uncheck option in **Window -> Preferences -> Maven -> Do not automatically update dependencies from remote repositories**. This should be able to help you get those dependencies.

backdoorcodr, 2017-12-10:
@zhiyua-git perfect. That resolved this issue. Many thanks :+1: :beer:

closed 2017-12-10 by backdoorcodr