awslabs/amazon-kinesis-video-streams-producer-sdk-java #7: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory (althought it exists in local/lib/)

backdoorcodr, 2017-12-10:
Hello,

I am trying to execute the demo application on my linux machine from eclipse. The import was successful and the project builds well. Before this, I have also build the [producer-sdk-cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp) and libKinesisVideoProducerJNI.so also resides in the "kinesis-video-native-build" directory. 

I have also adjusted the path to this native library within my eclipse IDE. However, runtime exception is being thrown : 

`ERROR / KinesisVideo: 2017-12-10T12:30Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: /home/workspace/kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory`

Where this native library needs to reside? Is `kinesis-video-native-build/downloads/local/lib/`a relevant place for that? I am saying this because `kinesis-video-native-build/downloads/local/lib/`consists of liblog4cplus-1.2.so.5.

zhiyua-git, 2017-12-10:
Hi,

This is a known linking issue we are trying to solve. 

Can you take a look at  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#troubleshooting. The first one workaround should fix your issue.

backdoorcodr, 2017-12-11:
@zhiyua-git okay, let me give it a try and let you guys know !

backdoorcodr, 2017-12-12:
@zhiyua-git That works :) Thank you :+1: :beer:

thorodi, 2019-02-27:
I have the same issue with liblog4cplus-2.0.so.3: cannot open shared object file: No such file or directory. Even though the file is in the output library folder after running "make install". Any idea why I get this error message?

thorodi, 2019-02-27:
Solved the issue. My LD_LIBRARY_PATH environment variable wasn't set correctly.

709280939, 2021-12-10:
> Solved the issue. My LD_LIBRARY_PATH environment variable wasn't set correctly.
How did you solve?