awslabs/amazon-kinesis-video-streams-producer-sdk-java #8: ClassNotFoundException: com.amazonaws.regions.Regions

TerrificMao, 2017-12-12:
I have built the jar package: kinesisvideo-java-demo-1.0-SNAPSHOT.jar successfully.
when I run comand: java -jar kinesisvideo-java-demo-1.0-SNAPSHOT.jar -Daws.accessKeyId=xxxxxxxxxxxxx -Daws.secretKey=xxxxxxxxxxxxxx -Djava.library.path=libKinesisVideoProducerJNI.dylib

Exception in thread "main" java.lang.NoClassDefFoundError: com/amazonaws/regions/Regions
        at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:30)
Caused by: java.lang.ClassNotFoundException: com.amazonaws.regions.Regions
        at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:335)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        ... 1 more

I have set my region with US_EAST_1 and changed my own stream name with kinesis-video-stream.

closed 2017-12-12 by TerrificMao

sumitbadaya, 2018-01-03:
Facing same problem. Any solution?

zhiyua-git, 2018-01-03:
Update your maven dependency will help you get all classes needed to build java package.

sumitbadaya, 2018-01-03:
Updated many times but it is not downloading any dependency and the errors persist

jafetmorales, 2018-03-06:
I was getting the same exception for `us-east-2` for the "Android PubSub with WebSockets Sample." I had to modify my `build.gradle` to add the following dependency: `'com.amazonaws:aws-android-sdk-cognitoidentityprovider:2.3.2'` and the following repository: `"https://maven.fabric.io/public"`. It works like a charm after that. Should I do a pull request?

mkhajuriwala, 2018-04-08:
I am facing the same problem. is there a solution to this

zhiyua-git, 2018-04-09:
@mkhajuriwala If you are seeing this issue when you are running from command line, can you try following [instruction](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/README.md#run-the-demo-application-from-command-line) here and see if problem persists? Thank you and look forward to your reply.