awslabs/amazon-kinesis-video-streams-producer-sdk-java #9: Exception thrown on sending thread: Exception while sending encoded chunk in MKV stream

TerrificMao, 2017-12-12:
When I run PutMediaDemo.java, I can see the sample video on AWS Kinesis Video console, but I can't get all video, here is my detailed infomation about error:

Dec 13, 2017 1:33:20 AM com.amazonaws.internal.DefaultServiceEndpointBuilder getServiceEndpoint
信息: {kinesisvideo, us-west-2} was not found in region metadata, trying to construct an endpoint using the standard pattern for this region: 'kinesisvideo.us-west-2.amazonaws.com'.
HTTP/1.1 200 OK
x-amzn-RequestId: 8f5e4b32-df62-11e7-acb6-5954fc2c0a6b
Content-Type: application/json
Transfer-Encoding: chunked
Date: Tue, 12 Dec 2017 17:33:26 GMT

72
{"EventType":"BUFFERING","FragmentTimecode":0,"FragmentNumber":"91343852333193120789800267782983005990846373393"}

71
{"EventType":"RECEIVED","FragmentTimecode":0,"FragmentNumber":"91343852333193120789800267782983005990846373393"}

75
{"EventType":"BUFFERING","FragmentTimecode":2433,"FragmentNumber":"91343852333193120794752027940124532060778351310"}

72
{"EventType":"PERSISTED","FragmentTimecode":0,"FragmentNumber":"91343852333193120789800267782983005990846373393"}

**_## b5
{"EventType":"ERROR","FragmentTimecode":2433,"FragmentNumber":"91343852333193120794752027940124532060778351310","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId":4004}

0
Received EOF for HTTP chunked encoding data.
ERROR: KinesisVideoStreams	2017-12-13T01:33Z T12: EXCEPTION: RuntimeException: Exception thrown on sending thread: Exception while sending encoded chunk in MKV stream !_** 
HTTP/1.1 200 OK
x-amzn-RequestId: 9c39ec29-df62-11e7-b2ca-3d51b9948330
Content-Type: application/json
Transfer-Encoding: chunked
Date: Tue, 12 Dec 2017 17:33:48 GMT

bdhandap, 2017-12-13:
Hi TerrificMao,

Demo app had an issue with the sample mkv file as it shared GOP between the clusters. I've updated the sample mkv file in the demo app and now the demo works just fine. Please try with the new file and let us know.

Thanks
Babu

TerrificMao, 2017-12-13:
Hi Babu,
Thanks so much for your help. I have run the PutMediaDemo.java successfully!
But when I run DemoAppMain.java, I still meet the error about native library libKinesisVideoProducerJNI.dylib.

WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
ERROR / KinesisVideo: 2017-12-13T10:40Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: no KinesisVideoProducerJNI in java.library.path
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:45)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.intellij.rt.execution.application.AppMain.main(AppMain.java:147)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.initializeLibrary(NativeKinesisVideoProducerJni.java:1045)
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:213)
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:186)
	at com.amazonaws.kinesisvideo.client.NativeKinesisVideoClient.initialize(NativeKinesisVideoClient.java:112)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:106)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:79)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:31)
	... 5 more

Process finished with exit code 1

closed 2017-12-13 by TerrificMao

roccomuso, 2018-06-21:
@bdhandap @unicornss Is it fixed now? can you provide a valid mkv file or the ffmpeg cmd to generate it?

bdhandap, 2018-06-21:
@roccomuso Yes. The [MKV file](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/tree/master/src/main/resources/data/mkv) is latest and should work with PutMediaDemo.

roccomuso, 2018-06-22:
@bdhandap the `run-java-demoapp.sh` that runs the java demo app do the video stream upload of that mkv file but on the video overview page I can't see the live stream... is it a bug? Are you able to run the PutMediaDemo and see the video stream on the overview?