awslabs/amazon-kinesis-video-streams-producer-sdk-java #12: Loading native library KinesisVideoProducerJNI failed

sambithota, 2017-12-21:
I am facing the same issue while running the sample in my local. I am getting this.
EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError
Because libKinesisVideoProducerJNI.dll is not present. How to generate this on windows machine?

zhiyua-git, 2017-12-21:
Thanks for reaching out, we are still looking into the WIndows build and hope to get the Windows solution out soon.

unicornss, 2018-01-06:
We will publish once we get the Windows Build for Kinesis Video Streams. in our roadmap.  

Closing this for now.

Thanks
-SS

closed 2018-01-06 by unicornss

mocampo78, 2018-04-04:
Hi...

Do you have a date to publish libKinesisVideoProducerJNI.dll ?

Still is not working in Windows 

Thanks

MushMal, 2018-04-04:
@mocampo78 thanks for your feedback. We are working on it. Unfortunately, we don't have an exact date currently. We will communicate as soon as we have a set date.

CedszxcJulian, 2018-07-03:
Hi, 

Still working on KinesisVideoProducerJNI.dll? :) 

Thank you & Kind Regards,
Ced

zhiyua-git, 2018-07-06:
@CedszxcJulian 

Yes, windows build is on our roadmap. We will communicate as soon as we have a set date.

Thanks,
-ZH

arunsridher, 2018-12-05:
Hey it got resolved for me by doing this
Right click on your project and go to **Java Build Path** then **Libraries** and under **Maven Dependencies** go to **Native Library Location**
Click on it and go to **Edit** and set the **Location Path** to /absolute path/to/amazon-kinesis-video-streams-producer-sdk-java/src/main/resources/lib/windows

I was initially **None**
I'm using windows and eclipse