awslabs/amazon-kinesis-video-streams-producer-sdk-java #13: Will have python sdk?

imyoungyang, 2017-12-25:
Hi

Will have python sdk?
Currently, I used python and opencv to generate mkv file. Then upload mkv file via java sdk putMedia. If you can provide python sdk, it will be very helpful.

Thank you.

unicornss, 2017-12-29:
Hi.

Glad you were able to use OpenCV to generate frames and PutMedia API to send to Kinesis Video Streams service.

We will certainly update you when we get the Python SDK. 

Thanks
- ss

unicornss, 2018-01-06:
We will publish once we get the Python SDK for Kinesis Video Streams. in our roadmap.  Closing this for now.

Thanks
-SS

closed 2018-01-06 by unicornss

krishan1456, 2018-11-23:
> Hi
> 
> Will have python sdk?
> Currently, I used python and opencv to generate mkv file. Then upload mkv file via java sdk putMedia. If you can provide python sdk, it will be very helpful.
> 
> Thank you.

Hi, Yang
Can you please share the python file that you have used to generate mkv file from video stream I have a requirement to do this.

imyoungyang, 2018-11-23:
@krishan1456 here is the github repo:
https://github.com/imyoungyang/video-streaming/blob/master/face-detection-multi-files.py#L33

krishan1456, 2018-11-29:
> @krishan1456 here is the github repo:
> https://github.com/imyoungyang/video-streaming/blob/master/face-detection-multi-files.py#L33

Thank you @imyoungyang for this help.

exquisitebuy, 2019-12-15:
I have written this python file which is able to push fragments .
But the video gets stuck after a while . not sure what is the issue

from App.AWSDeps.AWSBase import AWSBase
from aws_request_signer import AwsRequestSigner
import hashlib,requests,natsort,glob,os,time

class StreamVideoToKinesis(AWSBase):

    def __init__(self,regionName,serviceType):
        self.session = requests.Session()
        super(StreamVideoToKinesis, self).__init__(regionName=regionName,serviceType=serviceType)

    def sendToKinesis(self,streaARN,videoFile):
        print("DPDPDPDPDPDPD")
        getEPRSP = self._getEndPoint(streaARN)
        endPOint = getEPRSP["DataEndpoint"] + "/putMedia"
        headers = {
            "x-amzn-stream-arn": streaARN,
            "x-amzn-fragment-timecode-type": "ABSOLUTE",
        }
        files_hash = hashlib.sha256(('').encode('utf-8')).hexdigest()
        reqSigner = AwsRequestSigner(self.regionName, self.accessKey,self.secretKey, self.serviceType)
        headers.update(reqSigner.sign_with_headers("POST", endPOint, headers, content_hash=files_hash))
        # with requests.Session() as session:
        self.session.headers = headers
        resp = self.session.post(endPOint, data=open("fromWebCamCodded.mkv", "rb"), stream=True)
        # print(resp.json())
        print(resp.text)
        time.sleep(2)

    def _getEndPoint(self,streamArn):
        ret = self.botoClient.get_data_endpoint(StreamARN=streamArn, APIName="PUT_MEDIA")
        return ret

    def get_video_fragments(self):
        video_path = r"C:\Users\Mitrasax\personalRepo\staticUI\QuizProwessBackend\App\livestream\frames\VideoFragments"
        return natsort.natsorted(glob.glob(os.path.join(video_path, "*.mkv")))

if __name__ == "__main__":
    obj = StreamVideoToKinesis(regionName=AWSBase.AWSKeys["region_oregon"],serviceType="kinesisvideo")
    obj.sendToKinesis(streaARN="",videoFile="")
    # print(obj.get_video_fragments())

prem48science, 2020-06-25:
@exquisitebuy tried your method but i am not able to read the data sent to server in a browser nor I am able to get the data by using get media method. It throws an error that it is not in codec "V_MPEG/ISO/AVC". did you not face this problem? 

moreover, the `[documentation ](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_PutMedia.html)`says 
_"To be able to play the media on the console or via HLS, track 1 of each fragment should contain h.264 encoded video, the CodecID in the fragment metadata should be "V_MPEG/ISO/AVC", and the fragment metadata should include AVCC formatted h.264 codec private data. Optionally, track 2 of each fragment should contain AAC encoded audio, the CodecID in the fragment metadata should be "A_AAC", and the fragment metadata should include AAC codec private data."_

assman, 2020-07-28:
Is there any progress for the Python SDK for producers?

MushMal, 2020-07-28:
Sorry, no update on it yet

glods, 2020-08-16:
> I have written this python file which is able to push fragments .
> But the video gets stuck after a while . not sure what is the issue
>
> from App.AWSDeps.AWSBase import AWSBase
> from aws_request_signer [...]

------------------------------------*************************************
Hello @exquisitebuy , this looks interesting. I am trying to send video to kinesis video with python but get troubles with it. Please can you share the code you are using for that so that I can try it myself.   Thanks in advance

glods, 2020-08-16:
@imyoungyang thanks for your code. However I get the following error everytime I try to upload a mkv file.

Please how can I solve this ?
 
 Aug 07, 2020 9:01:12 AM com.amazonaws.internal.DefaultServiceEndpointBuilder getServiceEndpoint
INFO: {kinesisvideo, us-east-1} was not found in region metadata, trying to construct an endpoint using the standard pattern for this region: 'kinesisvideo.us-east-1.amazonaws.com'.
file name:../outputStream/1.mkv
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.fasterxml.jackson.databind.util.ClassUtil (file:/home/pi/Documents/pi-home-surveillance-v1/tryit/kinesisvideo-java-demo-1.0-SNAPSHOT.jar) to field java.lang.Throwable.cause
WARNING: Please consider reporting this to the maintainers of com.fasterxml.jackson.databind.util.ClassUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Exception in thread "main" java.lang.RuntimeException: Exception while connecting to the server ! 
	at com.amazonaws.kinesisvideo.http.ParallelSimpleHttpClient.connectAndProcessInBackground(ParallelSimpleHttpClient.java:137)
	at com.amazonaws.kinesisvideo.client.PutMediaClient.putMediaInBackground(PutMediaClient.java:89)
	at com.amazonaws.kinesisvideo.demoapp.PutMediaDemo.main(PutMediaDemo.java:142)
Caused by: javax.net.ssl.SSLException: Connection reset
	at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:127)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:326)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:269)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:264)
	at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:137)
	at java.base/sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1144)
	at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1055)
	at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:395)
	at java.base/sun.security.ssl.SSLSocketImpl.ensureNegotiated(SSLSocketImpl.java:709)
	at java.base/sun.security.ssl.SSLSocketImpl$AppOutputStream.write(SSLSocketImpl.java:962)
	at java.base/sun.nio.cs.StreamEncoder.writeBytes(StreamEncoder.java:233)
	at java.base/sun.nio.cs.StreamEncoder.implFlushBuffer(StreamEncoder.java:312)
	at java.base/sun.nio.cs.StreamEncoder.implFlush(StreamEncoder.java:316)
	at java.base/sun.nio.cs.StreamEncoder.flush(StreamEncoder.java:153)
	at java.base/java.io.OutputStreamWriter.flush(OutputStreamWriter.java:254)
	at java.base/java.io.BufferedWriter.flush(BufferedWriter.java:257)
	at com.amazonaws.kinesisvideo.http.ParallelSimpleHttpClient.sendInitRequest(ParallelSimpleHttpClient.java:177)
	at com.amazonaws.kinesisvideo.http.ParallelSimpleHttpClient.startCommunication(ParallelSimpleHttpClient.java:161)
	at com.amazonaws.kinesisvideo.http.ParallelSimpleHttpClient.connectAndProcessInBackground(ParallelSimpleHttpClient.java:135)
	... 2 more
	Suppressed: java.net.SocketException: Broken pipe (Write failed)
		at java.base/java.net.SocketOutputStream.socketWrite0(Native Method)
		at java.base/java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:110)
		at java.base/java.net.SocketOutputStream.write(SocketOutputStream.java:150)
		at java.base/sun.security.ssl.SSLSocketOutputRecord.encodeAlert(SSLSocketOutputRecord.java:81)
		at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:357)
		... 19 more
Caused by: java.net.SocketException: Connection reset
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:186)
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:140)
	at java.base/sun.security.ssl.SSLSocketInputRecord.read(SSLSocketInputRecord.java:448)
	at java.base/sun.security.ssl.SSLSocketInputRecord.decode(SSLSocketInputRecord.java:165)
	at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:108)
	... 16 more
done & rm: ../outputStream/1.mkv