awslabs/amazon-kinesis-video-streams-producer-sdk-java #14: CLI instructions

rnzsgh, 2017-12-29:
Please create CLI instructions; not everyone uses an IDE. Thanks!

After compiling the C++ producer SDK, these are the steps I took:

sudo apt-get install maven

#!/bin/bash

mvn package

jar_files=$(mktemp)

mvn -Dmdep.outputFile=$jar_files dependency:build-classpath

export LD_LIBRARY_PATH=/home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib:$LD_LIBRARY_PATH

classpath_values=$(cat $jar_files)

java -classpath target/kinesisvideo-java-demo-1.0-SNAPSHOT.jar:$classpath_values \
  -Daws.accessKeyId=ACCESS_KEY  \
  -Daws.secretKey=SECRET_KEY \
  -Djava.library.path=/home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build \
  com.amazonaws.kinesisvideo.demoapp.DemoAppMain

rm -Rf $jar_files

zhiyua-git, 2017-12-29:
Thanks for the information, I will update README to include CLI instruction soon.

unicornss, 2018-02-27:
Closing this as we have updated the Readme to include CLI instructions. Thanks for the input.

closed 2018-02-27 by unicornss