awslabs/amazon-kinesis-video-streams-producer-sdk-java #15: Clarify params in ClientDemoApp

rnzsgh, 2017-12-29:
Greetings, I am trying to use the [ClientDemoApp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/client/ClientDemoApp.java), but running into problems.

I do not know what the "apiUrl" should be and I do not know what "material set" should be configured as.

zhiyua-git, 2017-12-29:
Thanks for reaching out and sorry for the confusion caused by ClientDemoApp, it used some internal configurations and tools, so it would not work on GitHub. We will remove it to avoid distraction on this.

For demo purpose, can you please try out DemoAppMain and PutMediaDemo? Thanks.

rnzsgh, 2017-12-29:
Thanks for the clarification. I ran DemoAppMain to write data and that worked well, but I am looking for an example that showcases how to read MPEG-4 data from the stream for display to the user.

zhiyua-git, 2017-12-29:
The sample for stream data consumption is in this repo: https://github.com/aws/amazon-kinesis-video-streams-parser-library

Let me know how it goes. Thank you.

closed 2018-01-06 by zhiyua-git