awslabs/amazon-kinesis-video-streams-producer-sdk-java #16: KinesisVideoProducerJNI UnsatisfiedLinkError

lyi2012, 2017-12-31:
OS: ubuntu 16.04 LTS (on DeepLens)
JAVA:9.0.1

I also tried to follow the troubleshoot instruction at https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp
And it did not work.

Any suggestions?

ERROR / KinesisVideo: 2017-12-31T14:24Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: /home/aws_cam/kvspj/amazon-kinesis-video-streams-producer-sdk-java/target/libKinesisVideoProducerJNI.so: libproducer.so: cannot open shared object file: No such file or directory: /home/aws_cam/kvspj/amazon-kinesis-video-streams-producer-sdk-java/target/libKinesisVideoProducerJNI.so: libproducer.so: cannot open shared object file: No such file or directory
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:45)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.initializeLibrary(NativeKinesisVideoProducerJni.java:1061)
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:213)
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:186)
	at com.amazonaws.kinesisvideo.client.NativeKinesisVideoClient.initialize(NativeKinesisVideoClient.java:112)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:106)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:79)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:31)

unicornss, 2018-01-02:
Hi, 

If you have followed the instructions from https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp  then please verify that the library 
`libKinesisVideoProducerJNI.so` is present in the sub-directory `kinesis-video-native-build` within `amazon-kinesis-video-streams-producer-sdk-cpp`  directory. 

- Setting that directory location of library (`libKinesisVideoProducerJNI.so`)  in Kinesis Video Streams Java  SDK.
-- You can then use the following VM options to specify the directory location of the library similar to the one below.
 `-Djava.library.path=/Users/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build`

lyi2012, 2018-01-03:
Unicornss,

Thank you. I got it resolved. It turns out that the original directory structure was

home/aws_cam/**kvsp**/amazon-kinesis-video-streams-producer-sdk-java

after running the installation script, I renamed it to 

home/aws_cam/**kvspj**/amazon-kinesis-video-streams-producer-sdk-java

and for some reason, the .so file remembers the original **kvsp** directory and going back there to load its own dependencies.

I used strace to figure this out, and created a symbolic link from kvspj to kvsp, then everything works fine. 

-lyi

closed 2018-01-03 by lyi2012

sumitbadaya, 2018-01-03:
I have the same problem with same ubuntu version. And unfortunately the steps given in comment above didnt solve the issue. Can you please help how to resolve the same issue?

lyi2012, 2018-01-03:
Try the following steps, 

1. make sure the library libKinesisVideoProducerJNI.so is under 
amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/
2. try to use commandline to reproduce the problem, replace ... with the right classpath
java -classpath= ... com.amazonaws.kinesisvideo.demoapp.DemoAppMain
3. add strace in front of the command line
strace java -classpath= ... com.amazonaws.kinesisvideo.demoapp.DemoAppMain

Then you should be able to see which internal dependancy file has failure

- lyi

bbrandt94, 2018-02-08:
Hi there!

I'm having the same issue. When I go in the sub-directory kinesis-video-native-build within amazon-kinesis-video-streams-producer-sdk-cpp directory, I do NOT see the libKinesisVideoProducerJNI.so file. I am guessing this is because then install-script, when ran, does not complete because it runs into an error concerning bison (see attached image).
What can I do about this?

[image]

zhiyua-git, 2018-02-08:
Hi bbrandt94,

You will need to install bison first. According to the [install steps](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#install-steps-for-ubuntu-17x), you can try to Install bison: **sudo apt-get install bison**. And then rerun the install-script, if install-script finished successfully, you should be able to get the libKinesisVideoProducerJNI.so file.

Thanks,
-ZH

bbrandt94, 2018-02-08:
Hi zhiyua-git,

I should have mentioned this before but I am running Mac OS 10.13.3, so sudo apt-get is not what I would use. I do believe, however, that I have installed bison correctly because I can see it in the amazon-kinesis-video-streams-producer-sdk/kinesis-video-native-build/downloads folder. The issue I am having is that when I run the install script, I am getting this error (screenshot below) about a minute into the install-script running.

[image]

unicornss, 2018-02-08:
Hi bbrandt94. 
It has been working in  macOS High Sierra 10.13.2 (17C205): https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/18

Not sure if something has changed in 10.13.3.

Could you please try commenting out the bison from 

```https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/install-script#L255```

to
```https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/install-script#L266  ```

Please let us know how it goes when running install-script after this.

Thanks
ss

bbrandt94, 2018-02-08:
Hi unicornss,

Thank you for your response. I commented out the code in the bison file. I then ran the install-script and it ran for much longer than it had before, but it again ended with an error. However its a different error than before. See screenshot below.

[image]

unicornss, 2018-02-08:
Would you please run in the `/Users/macbook1/KinesisTest123/amazon-kinesis-video-streams-producer-sdk-java//kinesis-video-native-build/downloads/local/bin` folder the following version checks for these build tools?
```
./m4 --version
./autoconf --version
./automake --version
./aclocal --version 
./autom4te --version
```

bbrandt94, 2018-02-08:
Here are the results.

[image]

unicornss, 2018-02-09:
Thanks @bbrandt94 . Looks like m4 was installed correctly but could not get the autoconf installed in your environment. 
We will check in High Sierra 10.13.3  and keep you updated.  -SS

unicornss, 2018-02-13:
Hi @bbrandt94  
Bison build still seems to be an issue  (https://github.com/spack/spack/issues/5521)  in High Sierra  

To resolve the bison and autoconf quickly, would suggest installing these open source build tools/ dependencies through `brew`. Installing brew is straightforward and you can find instructions in https://brew.sh/  .

Once `brew` is installed you can install bison, `autoconfig` and `pkgconfig` using.

```
brew install bison 
brew install autoconf
brew install pkgconfig
```

Then  you can try commening the `autoconf` section in the `install-script` and  run it.  Let us know how it goes. 
Thanks
-SS

ramneetsingh, 2020-02-04:
Hi,
I am facing the same issue as
java.lang.UnsatisfiedLinkError: com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.getNativeLibraryVersion()Ljava/lang/String;

I've tried to set the native library path and LD_LIBRARY_PATH through command line/VM arguments, but nothing seems to work. Any help will be appreciated.

Below is the screenshot from the error

[image]