awslabs/amazon-kinesis-video-streams-producer-sdk-java #17: KinesisVideoProducerJNI UnsatisfiedLinkError on Ubuntu 16.04

sumitbadaya, 2018-01-03:
OS: Ubuntu 16.04 TLS
Java 8
JDK 1.8

I also tried to follow the troubleshoot instruction at https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp
And it did not work. I have followed the solution in the comment posted on issue #16 and made it sure that "libKinesisVideoProducerJNI.so" file is under "kinesis-video-native-build"  directory.
But it's still not working.

Getting runtime exception:

> WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
> WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
> WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
> WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
> WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
> WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
> ERROR / KinesisVideo: 2018-01-03T14:26Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: /home/sysquare/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build/libKinesisVideoProducerJNI.so: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory: /home/sysquare/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build/libKinesisVideoProducerJNI.so: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory
> Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
> 	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:45)
> Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
> 	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.initializeLibrary(NativeKinesisVideoProducerJni.java:1061)
> 	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:213)
> 	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:186)
> 	at com.amazonaws.kinesisvideo.client.NativeKinesisVideoClient.initialize(NativeKinesisVideoClient.java:112)
> 	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:106)
> 	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:79)
> 	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:31)
>

unicornss, 2018-01-03:
Hi Sumit,

Looks like the libraries (downloaded and installed during the native SDK build) is not in the path in your environment, which should not be happening.  In the meantime would you be able to try setting the environment variable using  `export LD_LIBRARY_PATH=/home/myawslabs/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib:$LD_LIBRARY_PATH`? Please make sure you are setting the path matching to your the folder location (where you have downloaded the Kinesis Video Stream CPP SDK and build the libraries using install-script).   You can also check the folder where you have downloaded and built the libraries for  liblog4cplus-1.2.so.5 .  If you have downloaded the native SDK in  /home/myawslabs then the libraries for native support It should be within the  /home/myawslabs/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib.   Once the environment variable is able to identify the path for the libraries, then you should be able to get the Java SDK demo app working. Let us know if you have questions. Thanks -SS

sumitbadaya, 2018-01-04:
I do not see any "downloads" directory or "lib" folder under amazon-kinesis-video-streams-producer-sdk-cpp directory

zhiyua-git, 2018-01-04:
Hi Sumit,

The "downloads" directory is under "{YourDir}/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/".

sumitbadaya, 2018-01-04:
thanks and its working now. But there are other issues like the sample code shows several errors like "Regions" could not be resolved, "builder" could not be resolved etc. And many such errors. I believe maven is unable to download/find dependencies needed for some reason. 
Can you please help me download all the needed libraries?

zhiyua-git, 2018-01-04:
Hi Sumit,

Which IDE are you using or are you trying to compile on CLI?

sumitbadaya, 2018-01-05:
Eclipse Oxygen 2.2

zhiyua-git, 2018-01-05:
Right-click on your project(s) --> Maven --> Update Project... UPDATE: In latest versions of Eclipse: For you Eclipse case, uncheck option in Window -> Preferences -> Maven -> Do not automatically update dependencies from remote repositories. This should be able to help you get those dependencies. Let me know if it works for your case. Thanks.

sumitbadaya, 2018-01-05:
I have read this solution earlier too and tried it several times but unfortunately it doesn't download any library and the errors of "xxx import cannot be resolved" persists.

zhiyua-git, 2018-01-05:
I just tried with 
Eclipse Java EE IDE for Web Developers.

Version: Oxygen.2 Release (4.7.2)
Build id: 20171218-0600

There is an issue with lombok integration. Solution is:
1. Downloaded jar from https://projectlombok.org/download
2. java -jar /path/to/lombok.jar
3. cd /path/to/eclipse & ./eclipse
4. lock Eclipse to the Launcher

If you use Eclipse shortcut to launch, it would crash after installing lombok, so please follow the steps to start Eclipse. Thanks.

zhiyua-git, 2018-01-05:
After installing lombok, do a project clean should trigger auto-rebuild and will resolve the issues.

sumitbadaya, 2018-01-05:
Thank you very much for the solution. Yes this did solve the Import issues.

closed 2018-01-05 by sumitbadaya