awslabs/amazon-kinesis-video-streams-producer-sdk-java #18: UnsatisfiedLinkError after building on macOS High Sierra (10.13.2)

hjobrien, 2018-01-09:
Hi, I'm getting the UnsatisfiedLinkError [others](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/16) have posted about as well, but I'm seeing it on my mac. I ran the instructions here: [https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp), and I see `amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.dylib`, which seems right, and I'm including the absolute path to `kinesis-video-native-build` as my `-Djava.library.path` in the JVM arguments (running in IntelliJ), any help would be appreciated!

On a related note, [this page](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-javaapi.html) says I should find a `KinesisVideoProducerJNI.dylib` file, whereas I'm finding a `libKinesisVideoProducerJNI.dylib` file (note the **lib** prefix), is that what the docs are talking about, or is there some other issue here?

unicornss, 2018-01-10:
Hi hjobrien ,
It should be to the **directory** where the file "libKinesisVideoProducerJNI.dylib" exists. Example:
-Djava.library.path=/home/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build

Can you provide the details of the exception you are seeing?
Thanks
-SS

hjobrien, 2018-01-10:
Here is my argument: `-Djava.library.path=~/Documents/Launchpad/yolonet/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build`

If I run `ls ~/Documents/Launchpad/yolonet/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build` I see `libKinesisVideoProducerJNI.dylib`

My full error text is here: https://gist.github.com/hjobrien/235833be5e45dda01b179e1f2dab0667, which seems like there is some path error as it looks like it can't find the `.dylib`

zhiyua-git, 2018-01-10:
Hi hjobrien,

The issue is you cannot specify "~/{path}" as your java.library.path. You can either use absolute path like "/Users/username/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build" or relative path to your work directory.

To get your absolute path, you can run "cd ~/Documents/Launchpad/yolonet/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build && pwd". With this absolute path set in java.library.path, the sample app should work.

Thanks,
-ZH

hjobrien, 2018-01-10:
Okay, thanks for that tip zhiyua! However, when I run that command, and I replace my path with its absolute one (`/Users/Hank/Documents/Launchpad/yolonet/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build`) I get what appears to be the same error: https://gist.github.com/hjobrien/6fef2c28226a585cab444c4c22629aab

unicornss, 2018-01-10:
This works fine in the Sierra version 10.12.6.  Not sure if anything different is happening in the High Sierra.  
Another debug suggestion is to set the library path in the https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/producer/jni/NativeLibraryLoader.java#L55  as `System.load("/Users/Hank/Documents/Launchpad/yolonet/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build")` and see if that loads successfully. 

 Note that this is in addition to the `-Djava.library.path=/Users/Hank/Documents/Launchpad/yolonet/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build` along with the AWS credentials `(-Daws.accessKeyId=<accessKey>
-Daws.secretKey=<secrectKey>)` you provided in the arguments. 

This is only ***debugging*** purposes and not ideal for regular usage.

hjobrien, 2018-01-10:
So if I replace that line with that suggestion, I get the same error again: https://gist.github.com/hjobrien/1bc6cc6bcdd0c82a662760ef8b0bd220. 

My program arguments are unchanged.

unicornss, 2018-01-17:
In macOS High Sierra 10.13.2 (17C205): 

After setting the path in the VM options of IDE ` -Djava.library.path=/Users/ss/Projects/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build` able to see the library is loaded and succeeded in creating the Kinesis Video Client.

**Output:**
`Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_createKinesisVideoClient(): Enter
Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_createKinesisVideoClient(): Creating Kinesis Video client.
createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 1073741824, spill ratio 90% and flags 0x00000001
heapInitialize(): Creating AIV heap.
heapInitialize(): Heap is initialized OK`

vijay-chavakula, 2018-01-29:
When i gave the properties in the VM arguments for the library path without quotes ("") the library is getting loaded fine. 
After setting the path in the VM options of IDE 

    -Djava.library.path=/Users/ss/Projects/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
Is this the fix or is it like we should be passing java.library.path property in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/producer/jni/NativeKinesisVideoProducerJni.java#L186 

    public void create(final @Nonnull DeviceInfo deviceInfo) throws ProducerException
    {
        create(deviceInfo,  System.getProperty("java.library.path"));
    }

and also make necessary changes in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/producer/jni/NativeLibraryLoader.java#L34

    final String dylibLibraryFileName = fullPath + File.separator + "lib" + libraryName + ".dylib";

Please suggest.

zhiyua-git, 2018-01-29:
Hi vijay-chavakula,

Setting java.library.path in the VM options of IDE is expected way to load library. You don't need to change any code to load library if VM options work for you.

Thank you.
-ZH

arunc4u91, 2018-02-21:
Hello 
Facing the similar issue when trying on Ubuntu and tried all the basic checks. The error message are 
[unsatisfied link error.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/1745831/unsatisfied.link.error.txt). 

I am trying the demo app in eclipse with the vm values as  -Djava.library.path=/home/alex/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build and with necessary credentials. Can anyone help me on this?

unicornss, 2018-02-21:
Hi @arunc4u91 
Can you check if `libKinesisVideoProducerJNI.so` file exists in the `/home/alex/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build` ?
if not, you have to checkout and build the Producer SDK (cpp)  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/ .  
As you mentioned you are using Ubuntu ...there is a section on readme for dependencies and how to build the native SDK in Ubuntu.
Also, can you tell us which version of Ubuntu you are using?

Thanks

arunc4u91, 2018-02-21:
yes the file exists in that path. I did successfully make that dependencies.
Ubuntu version
Description:	Ubuntu 16.04.3 LTS
Release:	16.04

unicornss, 2018-02-22:
Can you post the output of :  `ldd libKinesisVideoProducerJNI.so`
Also, since you have mentioned you are using Eclipse you may want to see how to resolve lombok in Eclipse.
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/17

arunc4u91, 2018-02-22:
[LDD command output.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/1746193/LDD.command.output.txt)

I have included lombok.jar to my eclipse and followed that solution. Still getting unsatisfied link error.
In the meanwhile, ill try once on intelliJ and update.

zhiyua-git, 2018-02-22:
Hi arunc4u91,

We tested with latest Eclipse Oxygen 4.7.2 just downloaded on Ubuntu 16.04, after resolving that Lombok issue, it worked well. Even with an empty file libKinesisVideoProducerJNI.so, it would report issues about that file itself instead of file not found.

Can you run "ls -l" and "pwd" in /home/alex/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build and provide the output for us to better understand what could be wrong (file size, file path, etc.)?

Thanks,
-ZH

arunc4u91, 2018-02-22:
Output for those commands and lombok in the eclipse configuration.
[ls command.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/1746404/ls.command.txt)

unicornss, 2018-02-22:
@arunc4u91  the ldd command output shows 
```
        liblog4cplus-1.2.so.5 => not found
	libcrypto.so.1.1 => not found
	libgtest.so => not found
```
Just to ensure the libraries are linked correctly,  can you follow these two steps below (as documented in the troubleshooting guide of https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/

`rm -rf ./kinesis-video-native-build/CMakeCache.txt ./kinesis-video-native-build/CMakeFiles  ` 
and run 
`./kinesis-video-native-build/install-script`

Also, could you try starting the demo app from command line  (ref:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/14 )
 
```
mvn package
jar_files=$(mktemp)
# Create classpath string of dependencies from the local repository to a file
mvn -Dmdep.outputFile=$jar_files dependency:build-classpath
# Add library path for all open source library dependencies
export LD_LIBRARY_PATH=/home/alex/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib:$LD_LIBRARY_PATH
classpath_values=$(cat $jar_files)
# Start the demo app 
java -classpath target/kinesisvideo-java-demo-1.0-SNAPSHOT.jar:$classpath_values
-Daws.accessKeyId=ACCESS_KEY -Daws.secretKey=SECRET_KEY -Djava.library.path=/home/alex/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build com.amazonaws.kinesisvideo.demoapp.DemoAppMain
```

arunc4u91, 2018-02-22:
After I run "rm -rf ./kinesis-video-native-build/CMakeCache.txt ./kinesis-video-native-build/CMakeFiles"
the ldd output shown all found and the unsatisfied link error is gone. Thanks guyz.

closed 2018-02-22 by unicornss

francoisdutoit, 2018-02-22:
Hi ,

I have looked at the above and also tried the below...

export LD_LIBRARY_PATH=/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH

But still getting an  libssl.so.1.1 in eclipse...

When building the demo project - at first I got the error : liblog4cplus-1.2.so.5: cannot open shared object file....after which - as suggested in installation instruction was to remove CMakeCache.txt and the CMake folder and run the install script again - which seem to have worked except now I'm getting the same error but for libssl.so.1.1. I did try repeating the steps but I still get the below error....

ERROR / KinesisVideo: 2018-02-22T20:09Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: libssl.so.1.1: cannot open shared object file: No such file or directory: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: libssl.so.1.1: cannot open shared object file: No such file or directory

How can I resolve this issue?

unicornss, 2018-02-22:
Can you please send output of the following:
1.  `ldd /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so`   . It will help to see if there is any missing links.
2.  `ls -l /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/dowloads/local/lib` . It will help us to see whether the library is present in the directory as expected after install-script run.  ` libssl.so.1.1: cannot open shared object file: No such file or directory`

If possible, I would suggest you run a clean build after` rm -rf /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads` and run the install-script again (which I hope uses the Flex 2.6.3 and automake 1.15.1 https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/34 )

PS: If you are familiar with Docker then we can refer to the Dockerfile that can build you the container for both CPP and Java SDK in Ubuntu 17.10.

francoisdutoit, 2018-02-23:
Hi , below is the output as asked for 

Also - after deleting downloads and building again - in eclipse I get the following 
ERROR / KinesisVideo: 2018-02-22T18:30Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or director

Then - after deleting CMakeCache.txt CMakeFiles and building again I get the same error
ERROR / KinesisVideo: 2018-02-22T20:09Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: libssl.so.1.1: cannot open shared object file: No such file or directory: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: libssl.so.1.1: cannot open shared object file: No such file or directory

//Output of ldd and ls -l

ldd /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so
	linux-vdso.so.1 =>  (0x00007ffc1c85a000)
	libproducer.so => /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libproducer.so (0x00007f2aab0cd000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2aaad47000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2aaab30000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2aaa750000)
	liblog4cplus-1.2.so.5 => /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/liblog4cplus-1.2.so.5 (0x00007f2aaa4d3000)
	libcrypto.so.1.1 => /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libcrypto.so.1.1 (0x00007f2aaa046000)
	libgtest.so => /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgtest.so (0x00007f2aa9d86000)
	libcurl.so.4 => /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libcurl.so.4 (0x00007f2aa9b18000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2aa97c2000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2aab93f000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2aa95a3000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2aa939f000)
	libssl.so.1.1 => not found

francois@francois-VirtualBox:~/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib$ ls -l 
total 71592
-rw-r--r-- 1 francois francois     235 Feb 23 05:43 charset.alias
drwxr-xr-x 2 francois francois    4096 Feb 23 05:29 engines-1.1
drwxr-xr-x 2 francois francois    4096 Feb 23 05:37 gettext
drwxr-xr-x 3 francois francois    4096 Feb 23 05:43 gio
drwxr-xr-x 3 francois francois    4096 Feb 23 05:43 glib-2.0
drwxr-xr-x 2 francois francois   16384 Feb 23 05:57 gstreamer-1.0
-rw-r--r-- 1 francois francois  111808 Feb 23 05:37 libasprintf.a
-rwxr-xr-x 1 francois francois    1054 Feb 23 05:37 libasprintf.la
lrwxrwxrwx 1 francois francois      20 Feb 23 05:37 libasprintf.so -> libasprintf.so.0.0.0
lrwxrwxrwx 1 francois francois      20 Feb 23 05:37 libasprintf.so.0 -> libasprintf.so.0.0.0
-rwxr-xr-x 1 francois francois   74736 Feb 23 05:37 libasprintf.so.0.0.0
-rw-r--r-- 1 francois francois 3522266 Feb 23 05:40 libblkid.a
-rwxr-xr-x 1 francois francois    1033 Feb 23 05:40 libblkid.la
lrwxrwxrwx 1 francois francois      17 Feb 23 05:40 libblkid.so -> libblkid.so.1.1.0
lrwxrwxrwx 1 francois francois      17 Feb 23 05:40 libblkid.so.1 -> libblkid.so.1.1.0
-rwxr-xr-x 1 francois francois 1633832 Feb 23 05:40 libblkid.so.1.1.0
-rw-r--r-- 1 francois francois   49496 Feb 23 05:39 libcharset.a
-rwxr-xr-x 1 francois francois    1047 Feb 23 05:39 libcharset.la
lrwxrwxrwx 1 francois francois      19 Feb 23 05:39 libcharset.so -> libcharset.so.1.0.0
lrwxrwxrwx 1 francois francois      19 Feb 23 05:39 libcharset.so.1 -> libcharset.so.1.0.0
-rwxr-xr-x 1 francois francois   35288 Feb 23 05:39 libcharset.so.1.0.0
-rw-r--r-- 1 francois francois 5070182 Feb 23 05:29 libcrypto.a
lrwxrwxrwx 1 francois francois      16 Feb 23 05:29 libcrypto.so -> libcrypto.so.1.1
-rwxr-xr-x 1 francois francois 2967384 Feb 23 05:29 libcrypto.so.1.1
-rw-r--r-- 1 francois francois  957030 Feb 23 05:31 libcurl.a
-rwxr-xr-x 1 francois francois    1193 Feb 23 05:31 libcurl.la
lrwxrwxrwx 1 francois francois      16 Feb 23 05:31 libcurl.so -> libcurl.so.4.5.0
lrwxrwxrwx 1 francois francois      16 Feb 23 05:31 libcurl.so.4 -> libcurl.so.4.5.0
-rwxr-xr-x 1 francois francois  510008 Feb 23 05:31 libcurl.so.4.5.0
drwxr-xr-x 3 francois francois    4096 Feb 23 05:37 libffi-3.2.1
-rw-r--r-- 1 francois francois   61544 Feb 23 05:37 libffi.a
[... cut ...]
-rw-r--r-- 1 francois francois 1644540 Feb 23 05:51 libyasm.a
-rw-r--r-- 1 francois francois  167170 Feb 23 05:38 libz.a
lrwxrwxrwx 1 francois francois      14 Feb 23 05:38 libz.so -> libz.so.1.2.11
lrwxrwxrwx 1 francois francois      14 Feb 23 05:38 libz.so.1 -> libz.so.1.2.11
-rwxr-xr-x 1 francois francois  129872 Feb 23 05:38 libz.so.1.2.11
drwxr-xr-x 2 francois francois    4096 Feb 23 05:50 pkgconfig
-rw-r--r-- 1 francois francois 1388592 Feb 23 05:39 preloadable_libiconv.so
-rw-r--r-- 1 francois francois  209400 Feb 23 05:37 preloadable_libintl.so

unicornss, 2018-02-23:
Hi @francoisdutoit  Thanks for the details.  The error message says that the 
` libssl.so.1.1: cannot open shared object file: No such file or directory `
However, that lib is available in the `/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib`
```
lrwxrwxrwx 1 francois francois 13 Feb 23 05:29 libssl.so -> libssl.so.1.1
-rwxr-xr-x 1 francois francois 528936 Feb 23 05:29 libssl.so.1.1
```
We will be looking into missing link issue.

By setting the LD_LIBRARY_PATH should resolve the issue.
 ( https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/17 , 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/7 )
 
Could you try, in Eclipse, Run configuration, set the Environment variable **LD_LIBRARY_PATH=/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH**

If you still see the error, can you try the following commands (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/14)  from the directory you have the 
amazon-kinesis-video-streams-producer-sdk-java downloaded. This should build the Java SDK and sets the path for LD_LIBRARY_PATH and start the demo sample application.
```
mvn package
# Create a temporary filename in /tmp directory
jar_files=$(mktemp)
# Create classpath string of dependencies from the local repository to a file
mvn -Dmdep.outputFile=$jar_files dependency:build-classpath
export LD_LIBRARY_PATH=/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib
classpath_values=$(cat $jar_files)
# Start the demo app
java -classpath target/kinesisvideo-java-demo-1.0-SNAPSHOT.jar:$classpath_values -Daws.accessKeyId=YOUR_ACCESS_KEY -Daws.secretKey=YOUR_SECRET_KEY -Djava.library.path=/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build com.amazonaws.kinesisvideo.demoapp.DemoAppMain

```

francoisdutoit, 2018-02-23:
Setting the environment variable worked :)
Thanks!