awslabs/amazon-kinesis-video-streams-producer-sdk-java #19: Fail to use custom mkv

SelniumTest, 2018-01-22:
hi , i run PutMediaDemo.java and replace cluster.mkv to my xxxx.mkv , but  i am getting below error message : 
onAckEvent AckEvent{ackEventType=ERROR, fragmentTimecode=null, fragmentNumber='null', errorCode=MORE_THAN_ONE_TRACK_FOUND, errorId=4005}

SelniumTest, 2018-01-23:
Hi zhiyua-git , 

I use single-MKV-track MKV and it's works , thanks.

closed 2018-01-23 by zhiyua-git

Rishabhadept, 2018-09-18:
Hi,

I am facing the same error. And I am not able to understand what does it mean when error says:  "More than one track" in MKV? I have tried different videos in MKV format (with audio, without audio, single frame throughout the video, recorded video) but this same error occurs. 

Can you please help me with what is "single MKV track" video?