awslabs/amazon-kinesis-video-streams-producer-sdk-java #20: Kinesis video stream output

SelniumTest, 2018-01-23:
Hi, 
I have some problem ,

1. I use SDK to send video to Kinesis video stream and what is the Kinesis video stream output and how to get the output ? 

2. How can I replace rekognition to other custom ML model with Kinesis video stream ?

zhiyua-git, 2018-01-23:
Hi SelniumTest,

For your questions:

1. You can use [Parser Library](https://github.com/aws/amazon-kinesis-video-streams-parser-library) to get either your original data out or SDK-generated MKV from your original data, based on your use case.

2. It depends on what kind of input your custom ML model accepts, you can get your original data out and operate on it.

Thanks,
-ZH

SelniumTest, 2018-01-24:
Hi zhiyua , 
So I need to use GetMedia calls to Amazon Kinesis Video first and  use a streaming Mkv Parser called StreamingMkvReader to read MkvElements from a stream ? 

another question , 

I use Rekognition Video with streaming video , a Rekognition Video stream processor to manage the analysis of the streaming video by calling CreateStreamProcessor and StartStreamProcessor , but it's only use for search faces in a collection and find match faces , i  cant find other Detecting such as Label Objects and Unsafe Content , its only support detect from S3?

zhiyua-git, 2018-01-24:
Hi SelniumTest,

Yes, you will need to call GetMedia and use StreamingMkvReader to parse it, there is an [example](https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/main/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoExample.java).

For your second question: Yes. At present, you can use Amazon Rekognition Video only to detect and recognize faces in streaming video carried by Kinesis Video Streams.

Thanks,
-ZH

closed 2018-01-24 by SelniumTest