awslabs/amazon-kinesis-video-streams-producer-sdk-java #21: rekognition output event records

SelniumTest, 2018-01-26:
Q:  What is the relationship between video and rekognition output event records length

Q: Is there any parameter in rekognition output handler event records content to represent video timestamp? 

We just can use FrameOffsetInSeconds and FragmentNumber to know the records content sequence, but not the timestamp of video ( I want to know FaceSearchResponse appear on what time of the video )

bdhandap, 2018-01-27:
From Rekognition team :

Information from the Rekognition results

1.       producerTimestamp: This should be the timestamp at which the mkv cluster was captured by the Webcam. This can be used with Method 1 below.

2.       fragmentNumber: This is the unique fragment number which identifies a mkv cluster in a Kinesis Video Stream. This can be used with Method 2 below.

3.       serverTimestamp: This is the timestamp at which the mkv cluster was received by Kinesis Video. This is provided for reference and can be useful for monitoring how much rekognition is lagging behind Kinesis Video.

4.       frameOffsetInSeconds: This is the relative offset of the frame to the mkv cluster. This is required in both Methods 1 and 2.

You can also refer Kinesis Video documentation for details on 1-3 above.

 

Following are the two methods to correlate the output on data-stream results with the input video stream.

 

Method 1: Using producer timestamp and frameOffset in seconds (More suitable for real-time streaming)

1.       “producerTimestamp” + “frameOffsetInSeconds” should give you the timestamp to which the rekognition results correspond

2.       Note: The fragmentTimeCodeType  should be set to “relative”  in putMedia request (See here)

 

Method 2: Using fragment number and frameOffset in seconds (More suitable for archived streams use-case)

1.       The putMedia calls have a callback which returns the “fragmentNumber” for the mkv cluster.

2.       The frameOffsetInSeconds identifies the frame within the fragment to which the rekognition results correspond.

3.       See also: Kinesis Video Streams Archived Media documentation for how you can get the actual video frames for a given fragment number.

bdhandap, 2018-01-27:
As this question is more related to Rekognition team, please contact rekognition-feedback@amazon.com for any further questions.

closed 2018-01-27 by bdhandap