awslabs/amazon-kinesis-video-streams-producer-sdk-java #22: Kinesis Video Streams data store

SelniumTest, 2018-01-29:
Hi ,

I have some questions ,

Amazon Kinesis Video Streams uses Amazon S3 as the underlying data store , but i did not find any info in S3, so. how can i find the underlying data store in S3 and the data format is ?

What is the kinesis video stream output ? I using parsing library to parses the merged output to print out the elements, only their offsets and size of the element in bytes but i can't find any actual video frames data .

The output format as follows:

Element Tags elementNumber 84 offset 916 element header size 8 element data size 235
Element Tag elementNumber 85 offset 924 element header size 6 element data size 229
Element SimpleTag elementNumber 86 offset 930 element header size 6 element data size 91
Element TagName elementNumber 87 offset 936 element header size 6 element data size 32
Tag Name :AWS_KINESISVIDEO_FRAGMENT_NUMBER
Element TagString elementNumber 88 offset 974 element header size 6 element data size 47
Element SimpleTag elementNumber 89 offset 1027 element header size 6 element data size 59
Element TagName elementNumber 90 offset 1033 element header size 6 element data size 33
Tag Name :AWS_KINESISVIDEO_SERVER_TIMESTAMP
Element TagString elementNumber 91 offset 1072 element header size 6 element data size 14
Element SimpleTag elementNumber 92 offset 1092 element header size 6 element data size 61
Element TagName elementNumber 93 offset 1098 element header size 6 element data size 35
Tag Name :AWS_KINESISVIDEO_PRODUCER_TIMESTAMP
Element TagString elementNumber 94 offset 1139 element header size 6 element data size 14
Element Cluster elementNumber 95 offset 1159 element header size 12 element data size 546583
Element CRC-32 elementNumber 96 offset 1171 element header size 2 element data size 4
Element Timecode elementNumber 97 offset 1177 element header size 2 element data size 1
Element SimpleBlock elementNumber 98 offset 1180 element header size 4 element data size 111839
Frame data (size): 111835 Frame(trackNumber=1, timeCode=0, keyFrame=true, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 99 offset 113023 element header size 3 element data size 5911
Frame data (size): 5907 Frame(trackNumber=1, timeCode=33, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 100 offset 118937 element header size 3 element data size 5215
Frame data (size): 5211 Frame(trackNumber=1, timeCode=67, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 101 offset 124155 element header size 3 element data size 4953
Frame data (size): 4949 Frame(trackNumber=1, timeCode=100, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 102 offset 129111 element header size 3 element data size 10084
Frame data (size): 10080 Frame(trackNumber=1, timeCode=133, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 103 offset 139198 element header size 3 element data size 14899
Frame data (size): 14895 Frame(trackNumber=1, timeCode=167, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 104 offset 154100 element header size 3 element data size 8184
Frame data (size): 8180 Frame(trackNumber=1, timeCode=200, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 105 offset 162287 element header size 3 element data size 4531
Frame data (size): 4527 Frame(trackNumber=1, timeCode=233, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 106 offset 166821 element header size 3 element data size 6178
Frame data (size): 6174 Frame(trackNumber=1, timeCode=267, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 107 offset 173002 element header size 3 element data size 11434
Frame data (size): 11430 Frame(trackNumber=1, timeCode=300, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 108 offset 184439 element header size 3 element data size 12073
Frame data (size): 12069 Frame(trackNumber=1, timeCode=333, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 109 offset 196515 element header size 3 element data size 7512
Frame data (size): 7508 Frame(trackNumber=1, timeCode=367, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 110 offset 204030 element header size 3 element data size 9619
Frame data (size): 9615 Frame(trackNumber=1, timeCode=400, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 111 offset 213652 element header size 3 element data size 14909
Frame data (size): 14905 Frame(trackNumber=1, timeCode=433, keyFrame=false, invisible=false, discardable=false, lacing=NO)
Element SimpleBlock elementNumber 112 offset 228564 element header size 3 element data size 7516
Frame data (size): 7512 Frame(trackNumber=1, timeCode=467, keyFrame=false, invisible=false, discardable=false, lacing=NO)
................................................................................................................................................................
................................................................................................................................................................

Can i use parsing libraries to merge multiple streams (ex: Three sections of 10 seconds video) into a single stream(30 seconds video)?

unicornss, 2018-01-29:
Based on the logs: ```Element SimpleBlock elementNumber 98 offset 1180 element header size 4 element data size 111839
Frame data (size): 111835 Frame(trackNumber=1, timeCode=0, keyFrame=true, invisible=false, discardable=false, lacing=NO)``` the Frame data is available and could you try ``` ByteBuffer dataBuffer = dataElement.getDataBuffer(); ``` You can also try decoding the  frame  (h264 encoded) using jcodec.
Thanks 
-SS

benalta, 2018-02-09:
Hi, 

You will not see data in S3 as Kinesis Video Streams as Kinesis Video Streams stores the data for you and you can access the video based on timestamp you provide in GetMedia API and provide timestamp, fragmentId and other metadata to capture collection of frames. In your consumer app, you can then create a video file and store on s3.
Would like to know more about your use case. Feel free to ping me at benaltar@amazon.com

unicornss, 2018-02-13:
@SelniumTest  A new example has been posted in the Kinesis Video Parser Library - to demonstrate how to retrieve frame data from the Kinesis Video Streams.  Details to similar question is also updated in   https://github.com/aws/amazon-kinesis-video-streams-parser-library/issues/7 

Thanks
SS

unicornss, 2018-03-12:
As the Kinesis Parser Library provides the samples for retrieving video streams using recommended  GetMedia API, closing this issue for now. 
Please re-open if required.

closed 2018-03-12 by unicornss