awslabs/amazon-kinesis-video-streams-producer-sdk-java #23: Failed to put a frame into the stream.

evi8899, 2018-02-01:
Hi everyone,

I just try to run DemoAppMain.java but get this error 
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
INFO / KinesisVideo: Initializing Kinesis Video client
DEBUG / KinesisVideo: calling describe stream: {StreamName: awsVideoRekogTest,}
DEBUG / KinesisVideo: describe stream result: {StreamInfo: {StreamName: awsVideoRekogxxxx,StreamARN: arn:aws:kinesisvideo:us-west-2:xxxxxxx:stream/awsVideoRekogxxxx/1517381810322,MediaType: video/h264,KmsKeyId: arn:aws:kms:us-west-2:xxxxx:alias/aws/kinesisvideo,Version: v4cQGOk7hO2k1dw23JT6,Status: ACTIVE,CreationTime: Wed Jan 31 14:56:50 CST 2018,DataRetentionInHours: 1}}
DEBUG / KinesisVideo: calling tag resource: {StreamARN: arn:aws:kinesisvideo:us-west-2:467417851175:stream/awsVideoRekogTest/1517381810322,Tags: {device=Test Device, stream=Test Stream}}
DEBUG / KinesisVideo: tag resource result: {}
DEBUG / KinesisVideo: calling get data endpoint: {StreamName: awsVideoRekogxxxx,APIName: PUT_MEDIA}
DEBUG / KinesisVideo: get data endpoint result: {DataEndpoint: https://xxxxx.kinesisvideo.us-west-2.amazonaws.com}
DEBUG / KinesisVideo: Stream awsVideoRekogxxxx is ready
DEBUG / KinesisVideo: PutFrame index: 0, pts: 15174701182350000, dts: 15174701182350000, duration: 200000, keyFrame: true, flags: 1
DEBUG / KinesisVideo: Kinesis Video client and stream metrics
	>> Overall storage size: 1073741824
	>> Available storage size: 1073741824
	>> Allocated storage size: 0
	>> Total view allocation size: 40056
	>> Total streams frame rate: 25
	>> Total streams transfer rate: 2000000
	>> Current view duration: 0
	>> Overall view duration: 0
	>> Current view size: 0
	>> Overall view size: 0
	>> Current frame rate: 25.0
	>> Current transfer rate: 2000000
DEBUG / KinesisVideo: Data availability notification. Size: 33709, Duration 200000
DEBUG / KinesisVideo: PutFrame index: 1, pts: 15174701182500000, dts: 15174701182500000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: Stream unblocked notification.
Exception in thread "pool-2-thread-1" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: **_Failed to put a frame into the stream._**
	at com.amazonaws.kinesisvideo.java.mediasource.file.ImageFileMediaSource.putFrame(ImageFileMediaSource.java:118)
	at com.amazonaws.kinesisvideo.java.mediasource.file.ImageFileMediaSource.access$3(ImageFileMediaSource.java:114)
	at com.amazonaws.kinesisvideo.java.mediasource.file.ImageFileMediaSource$1.onFrameDataAvailable(ImageFileMediaSource.java:105)
	at com.amazonaws.kinesisvideo.java.mediasource.file.ImageFrameSource.generateFrameAndNotifyListener(ImageFrameSource.java:75)
	at com.amazonaws.kinesisvideo.java.mediasource.file.ImageFrameSource.access$0(ImageFrameSource.java:69)
	at com.amazonaws.kinesisvideo.java.mediasource.file.ImageFrameSource$1.run(ImageFrameSource.java:64)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: **_Failed to put a frame into the stream._**

Please help me.
Thanks in advance.

zhiyua-git, 2018-02-01:
Hi evi8899,

Can you search for "Failed to put a frame with status code " in your log? That error code will be helpful for us to better understand the issue.

Thanks,
-ZH

evi8899, 2018-02-01:
Hi ZH,

I can't find "Failed to put a frame with status code " in my Eclipse console log. Can you be more specific which log should I check?

Thanks a lot.

zhiyua-git, 2018-02-01:
Hi evi8899,

We were having some issues on setting waiting time for sending frames and just sent out a fix for that. Can you please pull down the latest and check if you are still having this issue?

Thanks,
-ZH

evi8899, 2018-02-02:
Hi ZH, 

Thank you for the quick fix, now I didn't get  "Failed to put a frame" error any more.
But why my Kinesis videoPreview still blank?
[image]

Thank you.

zhiyua-git, 2018-02-02:
Hi evi8899,

How long did you wait for the preview? Are you checking the correct stream in the correct account and correct region? For default setting in demo app, stream name should be "my-stream", region is default to Oregon in your account. If your network throughput is low, you may see the preview video with a higher delay. 

And don't forget to click on the rightmost button to jump to current timestamp if you are getting console idle for a long time. 

Thanks,
-ZH

evi8899, 2018-02-02:
I wait more than 5 minutes but still no preview. 
Yes, I see in the console log it will created the stream if its not exist, even I used another stream that I have created before still no luck. 
I also clicking on the rightmost button to jump to the latest fragment but still nothing.

Thank you.

zhiyua-git, 2018-02-02:
Hi evi8899,

Can you search for "EventType" in your log to see if you get ACK which means your data sent was accepted by our service?

If you get the ACKs, they would be something like this: {"EventType":"PERSISTED","FragmentTimecode":0,"FragmentNumber":"91343852333183861656890514039540460834585194620"}

Thanks,
-ZH

evi8899, 2018-02-02:
Hi ZH,

I didn't get any ACKs and found this log: 
DEBUG / KinesisVideo: Data availability notification. Size: 32947159, Duration 400250000
DEBUG / KinesisVideo: PutFrame index: 10886, pts: 15175393621390000, dts: 15175393621390000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: Data availability notification. Size: 32946686, Duration 400240000
DEBUG / KinesisVideo: PutFrame index: 10887, pts: 15175393621790000, dts: 15175393621790000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: Data availability notification. Size: 32946344, Duration 400240000
DEBUG / KinesisVideo: PutFrame index: 10888, pts: 15175393622200000, dts: 15175393622200000, duration: 200000, keyFrame: false, flags: 0
: Reporting a dropped frame/fragment.
viewItemRemoved(): Reporting a dropped frame/fragment.
viewItemRemoved(): Reporting a dropped frame/fragment.
viewItemRemoved(): Reporting a dropped frame/fragment.
getKinesisVideoMetrics(): Get the memory metrics size.

What could be the problem?

Thanks

zhiyua-git, 2018-02-02:
Hi evi8899,

From the log, maybe the client was not able to establish the connection successfully, so the buffer was cumulating frame and finally got full and started dropping frames. Can you put a full log for 30-second running, it will help a lot on understanding the issue.

Thanks,
-ZH

evi8899, 2018-02-02:
Hi ZH,

This is the log you requested.
Thanks a lot
[debug_log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/1687976/debug_log.txt)

zhiyua-git, 2018-02-02:
Hi evi8899,

The log shows connection for your putMedia api call to our service was not established successfully. I just made some log improvement to get more information about SDK status. Can you pull down the latest, run it and paste the log? 

Also, did you have permission to call Kinesis Video Streams service? If you have AmazonKinesisVideoStreamsFullAccess, then it should be OK.

Thanks a lot.
-ZH

evi8899, 2018-02-05:
Hi ZH,

I didn't see the improvement log, but maybe you can help me to check it more details.
Yes, I have AmazonKinesisVideoStreamsFullAccess.
[debug_log_0205.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/1693679/debug_log_0205.txt)

Thanks a lot
Evi

evi8899, 2018-02-05:
Hi ZH,

I also found errors in Java projects
[image]

Thanks a lot for your help
Evi

zhiyua-git, 2018-02-05:
Hi evi8899,

Thanks for your log file. The log shows you are not able to send the PutMedia request out. Can you run "aws kinesisvideo describe-stream --stream-name stream --endpoint https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com" with [AWS CLI](https://aws.amazon.com/cli/) to check your connectivity? It should either return "An error occurred (403)" or "Could not connect to the endpoint URL".

For the errors in Java projects, if you are using Eclipse Java EE IDE Oxygen, this [post ](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/17)may help you.

If you are using Java 9, please switch to Java 8. Lombok has a compatible issue about Java 9.

Also, I am curious how you could run that DemoAppMain when you were having those compilation errors.

Thanks,
-ZH

evi8899, 2018-02-06:
Hi ZH, 

Running "aws kinesisvideo describe-stream --stream-name stream --endpoint https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com" with AWS CLI return "An error occurred (403)"
What does that mean?

I am  also curious how can I run DemoAppMain when I got those compilation errors. But installing lombok.jar did the trick and I can see the videoPreview. 

Thanks a lot for your help.
Evi

closed 2018-02-06 by evi8899

zhiyua-git, 2018-02-06:
Glad to hear you got it working! That 403 is just testing your connectivity with our service endpoint. It is kind of the easiest way to test it for now, if you get that 403 means you can connect to our service.

bernieatenovlab, 2018-02-12:
evi8899. Do you mind providing information on how you installed lombok.jar? I might be facing a similar problem and that's why I'm asking. Are you running on Windows, Mac or Ubuntu? Any help is greatly appreciated.

swetharnaik, 2018-02-23:
Hi,

I am getting this error at the beginnig of the code

putStreamResult(): Failed to put stream result event with status code 0x52000010
throwNativeException(): Throwing com/amazonaws/kinesisvideo/producer/ProducerException with message: Failed to put stream result event.

Also my frames are getting removed inbetween.

My java code is running. But when I go and check my video preview. Its just showing a black screen.

zhiyua-git, 2018-02-23:
Hi swetharnaik,

According to [error reference](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html), your error is some authorization error. 

Can you check if Kinesis Video Streams access is granted in your IAM policy? You can simply add AmazonKinesisVideoStreamsFullAccess to your account or you can refer to this [document](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iam.html) to set up your policy.

Let me know if it works.

Thanks,
-ZH

swetharnaik, 2018-02-23:
Yup. It worked. Just reran it again. Thanx a lot.

But that wasnt the issue. I had already given FullAccess to my IAM user.

evi8899, 2018-02-26:
Hi bernieatenovlab,

Sorry for the late reply, this [post](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/17) maybe helpful to install lombok jar. 
I'm running in Ubuntu.