awslabs/amazon-kinesis-video-streams-producer-sdk-java #25: Sending image to kinesis video stream.

raju249, 2018-02-12:
Hi, 

Is there a way to send image jpeg file to kinesis video stream and then use aws rekognition to analyse those files?

I can use the android smartphone camera to capture image and then send it over to kinesis video stream as soon as it is clicked.

Please help if there is any way.
I searched over documentation but couldn't find an example.
Thanks

zhiyua-git, 2018-02-12:
Hi raju249,

Have you looked in [Android SDK](https://github.com/aws/aws-sdk-android/tree/master/aws-android-sdk-kinesisvideo) yet? We have a sample app there sending frames captured from smartphone camera. If not yet, give it a try, feel free to contact if you need more help.

Thanks,
-ZH

raju249, 2018-02-13:
Thanks @zhiyua-git
Will take a look and try using it.
Is there any example available of using this SDK ?

zhiyua-git, 2018-02-13:
Hi raju249,

In Android project, we have the [Android demo app](https://github.com/awslabs/aws-sdk-android-samples/tree/master/AmazonKinesisVideoDemoApp) to stream frames from camera to our service. 

You can refer to our [document](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-android.html) to get more information about that.

Thanks,
-ZH

raju249, 2018-02-13:
Thanks @zhiyua-git

closed 2018-02-13 by zhiyua-git