awslabs/amazon-kinesis-video-streams-producer-sdk-java #26: Python Example of PutMedia API

nicklee1990, 2018-02-19:
I know the Python SDK is on the roadmap, but do you have any example code for using the PutMedia API with Python? I am trying to put something together, but there are few unknowns such as the format of the `x-amzn-producer-start-timestamp` header

Even just a few pointers would be helpful on how we can use the API with Python

bdhandap, 2018-02-20:
`x-amzn-producer-start-timestamp` is an optional start timestamp which you can use to set the relative producer timestamp offset for the PutMedia input. Say you have different input MKV files that are starting with cluster timecode '0' and you want the categorize each input MKV with different start timestamp, then you can do that by setting the `x-amzn-fragment-timecode-type` to RELATIVE and pass the different producer timestamp values to `x-amzn-producer-start-timestamp` for each PutMedia session. Later, you can use these producer timestamp values to search for a particular fragment in GetMedia API.

`payload` is the input data in MKV format which you need to send it as HTTP body in PutMedia request. The response which you get for PutMedia will contain a series of acknowledgements with  fragment timecode and fragment number. You can use the `requests` module to parse the chunked response of PutMedia API.

For more details about PutMedia API please refer : 
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_PutMedia.html

I don't have a Python example for now. I'll update this thread once I have it. I hope I answered your questions. Let me know if you need more clarification.

nicklee1990, 2018-02-20:
thanks for the assistance - very useful information! regarding the payload, what I meant was when generating a v4 signature for the request [the guide](https://docs.aws.amazon.com/general/latest/gr/sigv4-signed-request-examples.html#sig-v4-examples-post) states that the payload should be hashed to form part of the signature. So I was asking what do we do in this case? create a hash of the mkv payload?

I'll try work on a Python example as well. The 2 bits I'm struggling with currently are keeping open a duplex connection using the `requests` library, and generating the auth signature

bdhandap, 2018-02-23:
You dont have to create a hash for mkv payload as KinesisVideoStreams input can be real-time streaming input and signing the payload is impossible in this case. You have to send a special header `x-amz-content-sha256:UNSIGNED-PAYLOAD` to let KVS know that you are sending unsigned payload. Please refer this similar [java example](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/client/signing/KinesisVideoAWS4Signer.java#L88) which sets the header while signing the PutMedia request.

nicklee1990, 2018-02-25:
ok i'll take a look at that and try to work with the requests library. Looking forward to the Python SDK :)

prem48science, 2020-06-26:
@bdhandap according to [this guide ](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_PutMedia.html)
_"To be able to play the media on the console or via HLS, track 1 of each fragment should contain h.264 encoded video, the CodecID in the fragment metadata should be "V_MPEG/ISO/AVC", and the fragment metadata should include AVCC formatted h.264 codec private data. Optionally, track 2 of each fragment should contain AAC encoded audio, the CodecID in the fragment metadata should be "A_AAC", and the fragment metadata should include AAC codec private data."_

we are able to send data to kinesis video streams by inserting Mkv formated segment as payload. but as per the above guide to display it in browser or AWS console we need to send it in H264 formate or some meta data should be added(as I understand). 
If i send h264 formated segment the put media throws an error 4006. if I send Mkv formatted video it says PERSISTED but when i open the browser it shows an error _" type or codec is not defined"_

MushMal, 2020-06-26:
@prem48science here is the documentation of PutMedia API: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_PutMedia.html

The underlying streaming packaging format is MKV so whatever your elementary stream encoding is, it still needs to be packaged in MKV. In your case, you state that you are sending raw h264 frames which will fail MKV parser.

You also mentioned that when you package MKV it fails the playback. In order for the playback to work you will need elementary h264 or h265 stream packaged in AvCC NALus and wrapped in MKV where the MKV header should contain CPD (Codec Private Data). For H264 the CPD is SPS/PPS sequence. For H265, it's VPS/SPS/PPS sequence.

I understand that this is a lot of video technology, this is the reason we have producer SDKs to help.

daveisfera, 2022-01-17:
Any update on this from the Python perspective? Is the current recommendation to use the [C Producer](https://github.com/awslabs/amazon-kinesis-video-streams-producer-c) or [C++ Producer](https://github.com/awslabs/amazon-kinesis-video-streams-pic/)? If so, is one a "better" option than the other when working with Python code?