awslabs/amazon-kinesis-video-streams-producer-sdk-java #28: RTSP stream

francoisdutoit, 2018-02-25:
How can I use the PutMedia API to push an RTSP stream from an IP camera to Kinesis from a java application . I am able to run the demp app and it works great :) Was just hoping there could be an example of using a stream from an onvif cameras rtsp's stream . In my case I already have the RTSP stream url but have no idea as to how to push this up....
Any help or examples will be appreciated.
Thanks!!

vijaygj, 2018-02-25:
Hello,

Have you tried following example?

AWS_ACCESS_KEY_ID=AKIASAMPLEKEYID AWS_SECRET_ACCESS_KEY=MYSECRETACCESSKEY ./kinesis_video_gstreamer_sample_rtsp_app rtspurl  stream-name

From - https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp 

I understand you need support in Java from "from a java application". However, if you can confirm C++ works for you then we can find some options to do the same in Java.

francoisdutoit, 2018-02-25:
Ok Thanks- I will try the C++ app and then once I can confirm that it works from there we can look into java.

francoisdutoit, 2018-02-26:
Think I'm gonna need a bit of help - Not very big on C++....
But does the libKinesisVideoProducerJNI.so provide the same functionality?

francoisdutoit, 2018-02-27:
@unicornss 
Could you possible assist with the above?
Thanks!

unicornss, 2018-02-27:
As @vijaygj  has mentioned, if the objective is to send stream from your camera then its straightforward to run the executable ` ./kinesis_video_gstreamer_sample_rtsp_app` which will be inside the `kinesis-native-video-build` directory of the Producer CPP SDK once its built.
You can run/start streaming using:
```
AWS_ACCESS_KEY_ID= <MYACCESSKEYID>  AWS_SECRET_ACCESS_KEY=<MYSECRETKEY> ./kinesis_video_gstreamer_sample_rtsp_app  <rtspurl> <stream-name>
 ```
##### How to send video streams to Kinesis Video Streams using Producer Java SDK 
If you plan on using the Java, then you can refer to the 
* [PutMediaDemo.java](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/demo/com/amazonaws/kinesisvideo/demoapp/PutMediaDemo.java) 
* [DemoAppMain.java](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/demo/com/amazonaws/kinesisvideo/demoapp/DemoAppMain.java)
samples for sending the stream to Kinesis Video Streams.
##### How to  get h264 frames from RTSP Server
you have to implement/add the RTSP client to fetch the h264 frames from your camera RTSP endpoint, then use the above step to send video streams to the Kinesis.

Hope that helps.

francoisdutoit, 2018-02-28:
Thanks :) - that makes it a bit more clear. I will give it a go and let you know.

francoisdutoit, 2018-02-28:
@unicornss 
So I tried to run it with the below

AWS_ACCESS_KEY_ID=MYKEY  AWS_SECRET_ACCESS_KEY=MYSECRET ./kinesis_video_gstreamer_sample_rtsp_app  rtsp://192.168.0.108:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif MyOnvifStream

but getting the following error   "MyOnvifStream: command not found"

Any Ideas?

unicornss, 2018-02-28:
Looks like your camera RTSP URL has query params... you can try enclosing them in quotes as below.

```
AWS_ACCESS_KEY_ID=MYKEY AWS_SECRET_ACCESS_KEY=MYSECRET ./kinesis_video_gstreamer_sample_rtsp_app "rtsp://192.168.0.108:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif" MyOnvifStream
```

francoisdutoit, 2018-02-28:
That works thanks......

The latency is very bad though and when the camera picks up movement the quality in the AWS console get really bad and it is hard to make out what is going on....

Do I have to use the java media framework to read from rtsp? The demo's do not show how to 
do that?

bdhandap, 2018-02-28:
@francoisdutoit The video which you see in the console is not real-time and its shown after its persisted in Kinesis video system. In order to get the real-time video for latency sensitive applications, please use [GetMedia API](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_GetMedia.html). You can use the KinesisVideo parser library to parse the GetMedia fragments. Please refer [GetMedia example](https://github.com/aws/amazon-kinesis-video-streams-parser-library/tree/master/src/main/java/com/amazonaws/kinesisvideo/parser/examples) on how to use the API and parser library.

unicornss, 2018-02-28:
@francoisdutoit  
Would you please attach the screenshot of the video you are seeing. 
Also, _resolution of the camera_ and the _details of the computer (memory, cpu) _ you are running the RTSP demo app will help us understand the configuration of your setup.

francoisdutoit, 2018-02-28:
@unicornss 
I think the answer that @bdhandap gave sounds clear to  me and makes sense. ( To get a more "real-time" stream ) I will try that and give feedback.

Now I just need to figure out how to get the rtsp stream up to AWS using java.... hopefully one of my last requests :) 

PS: Just also want to say thank for all the help so far - really appreciated - especially the promptness and accuracy. Really impressed :1st_place_medal:

francoisdutoit, 2018-03-01:
@unicornss 
So in the java app...we create a Media source as below...
How do I create an onvif media source from the onvif url?

    private static MediaSource createImageFileMediaSource() {
        final ImageFileMediaSourceConfiguration configuration =
                new ImageFileMediaSourceConfiguration.Builder()
                        .fps(FPS_25)
                        .dir(IMAGE_DIR)
                        .filenameFormat(IMAGE_FILENAME_FORMAT)
                        .startFileIndex(START_FILE_INDEX)
                        .endFileIndex(END_FILE_INDEX)
                        .build();
        final ImageFileMediaSource mediaSource = new ImageFileMediaSource();
        mediaSource.configure(configuration);

        return mediaSource;
    }

MushMal, 2018-03-01:
@francoisdutoit could you please clarify your scenario for Java. Here is my assumption... You have a piece of code that connects to your camera and consumes the rtsp stream in java. It then splits the rtsp elementary stream and extracts an H264 encoded frame right? If so, you can use the Java examples and build an application that uses a custom media source.

The difficulty here is how the rtsp stream is being parsed.

If you prefer to just use the C++ demo application as is, you can launch this process from your Java and specify the parameters.

francoisdutoit, 2018-03-02:
@MushMal 

Thanks for the quick response :) 

I have thought of just calling the C++ demo app  ( as a last resort )  but would like to use java  as most of my code is already in java . At the moment my code only gets the RTSP url  and now I want to use this to create a media source.  The RTSP url allows you to specify the transfer protocol ( rtsp://192.168.0.164/axis-media/media.amp?Transport=multicast - Connect to an RTSP server running on 192.168.0.164 at the axis-media/media.amp location. The data will be streamed using the default compression (H264) and Multicast over UDP.)  Will I be bale to create a custom media source from this and if so - can you point me to an example of a custom media source?

Thanks!

MushMal, 2018-03-02:
@francoisdutoit the MediaSource and MediaSourceConfiguration are interfaces. Take a look at the implementations - there are some abstract classes like CameraMediaSource which you can use as a base. ImageMediaSource is a good example of an implementation.

As you parse the rtsp stream, just send the frames to the sink in your implementation of the MediaSource.

francoisdutoit, 2018-03-03:
Thanks. I'm gonna try to get something working with the information provided. I'm going to close the call for now - as at the moment I can push some streams up to kinesis with my current setup. Just need to play around a bit to see if I can get something going with RTSP in java :)

closed 2018-03-03 by francoisdutoit