awslabs/amazon-kinesis-video-streams-producer-sdk-java #29: Send video from a lambda to kinesis stream

naycont, 2018-03-01:
How can I use the PutMedia API into a lambda?

I tried to upload de source code  but it's to big for the lambda limits.

Any help?

Thanks.

MushMal, 2018-03-01:
@naycont could you please clarify your scenario. Are you trying to spin up a lambda to act as a producer? Generally, lambdas are for work that's "prompt" in nature - a short operation. Streaming, on the other hand, is a lengthy operation (given you have a long stream and not a few fragments at a time) and might not be suitable for lambda.

naycont, 2018-03-01:
Thanks for the answer.

I have a lambda that process S3 new object created event for convert videos fragments.  Then, I need to send them one by one to the stream, invoking another lambda.

MushMal, 2018-03-02:
Thanks @naycont that explains the situation a little better. Perhaps your lambda should "schedule" the upload and then a long-running EC2 instance would take the batched job and run a full-blown SDK to do the putting? It's hard to process long-running stream via lambda. However, if your S3 is already in the right format (MKV that KVS consumes) then you can simply use PutMedia API

naycont, 2018-03-02:
yep, my S3 files are in the MKV format.

What I try to do is a severless application. 

Thank you for your help,

vijaygj, 2018-03-02:
> I tried to upload de source code but it's to big for the lambda limits.

With lambda you can upload built artifacts. That should have way smaller footprint. Have you tried that?

Following is the example how to put OpenCV into lambda,

echo " --------- install OpenCV python module -----------"
cp /home/ec2-user/sources/opencv-3.1.0/lib/cv2.so /home/ec2-user/lambda-package/cv2/__init__.so
cp -L /home/ec2-user/sources/opencv-3.1.0/lib/*.so* /home/ec2-user/lambda-package/cv2/
cp -L /home/ec2-user/build/ffmpeg/lib/*.so* /home/ec2-user/lambda-package/cv2/
strip --strip-all /home/ec2-user/lambda-package/cv2/*
chrpath -r '$ORIGIN' /home/ec2-user/lambda-package/cv2/__init__.so
touch /home/ec2-user/lambda-package/cv2/__init__.py

You can zip this lambda-package and upload the artifacts to lambda so the overall binary footprint meets Lambda's need. Please let me know if this approach does not work out.

turb0kat, 2018-08-08:
resolving due to inactivity, feel free to re-open.

closed 2018-08-08 by turb0kat