awslabs/amazon-kinesis-video-streams-producer-sdk-java #31: Failed loading native library

romangrewal, 2018-03-07:
I am having the same issue as others in loading the native library.
Steps taken:
Using OSX,
ran the install-script from amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
ran the java-install-script from amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build

Using intelliJ IDEA, in idea.vmoptions set
-Djava.library.path=/absolute/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build

In https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/producer/jni/NativeLibraryLoader.java#L55
is set to 
System.load('/absolute/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build')

Then in intelliJ editor right clicked DemoAppMain and clicked run
it then outputs the following error:
/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/bin/java "-javaagent:/Applications/IntelliJ IDEA CE.app/Contents/lib/idea_rt.jar=49586:/Applications/IntelliJ IDEA CE.app/Contents/bin" -Dfile.encoding=UTF-8 -classpath /Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/charsets.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/deploy.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/cldrdata.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/dnsns.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/jaccess.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/jfxrt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/localedata.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/nashorn.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/sunec.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/sunjce_provider.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/sunpkcs11.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/zipfs.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/javaws.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jce.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jfr.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jfxswt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jsse.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/management-agent.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/plugin.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/resources.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/rt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/ant-javafx.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/dt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/javafx-mx.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/jconsole.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/packager.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/sa-jdi.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/tools.jar:/Users/romangrewal/t1cgacademy/sp14/t1-academy/kinesis/java/amazon-kinesis-video-streams-producer-sdk-java/target/classes:/Users/romangrewal/.m2/repository/com/amazonaws/aws-java-sdk-core/1.11.172/aws-java-sdk-core-1.11.172.jar:/Users/romangrewal/.m2/repository/commons-logging/commons-logging/1.1.3/commons-logging-1.1.3.jar:/Users/romangrewal/.m2/repository/org/apache/httpcomponents/httpclient/4.5.2/httpclient-4.5.2.jar:/Users/romangrewal/.m2/repository/org/apache/httpcomponents/httpcore/4.4.4/httpcore-4.4.4.jar:/Users/romangrewal/.m2/repository/commons-codec/commons-codec/1.9/commons-codec-1.9.jar:/Users/romangrewal/.m2/repository/software/amazon/ion/ion-java/1.0.2/ion-java-1.0.2.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/core/jackson-databind/2.6.7.1/jackson-databind-2.6.7.1.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/core/jackson-annotations/2.6.0/jackson-annotations-2.6.0.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/core/jackson-core/2.6.7/jackson-core-2.6.7.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/dataformat/jackson-dataformat-cbor/2.6.7/jackson-dataformat-cbor-2.6.7.jar:/Users/romangrewal/.m2/repository/joda-time/joda-time/2.8.1/joda-time-2.8.1.jar:/Users/romangrewal/.m2/repository/com/google/inject/guice/4.0/guice-4.0.jar:/Users/romangrewal/.m2/repository/javax/inject/javax.inject/1/javax.inject-1.jar:/Users/romangrewal/.m2/repository/aopalliance/aopalliance/1.0/aopalliance-1.0.jar:/Users/romangrewal/.m2/repository/com/google/code/findbugs/jsr305/3.0.2/jsr305-3.0.2.jar:/Users/romangrewal/.m2/repository/com/google/guava/guava/21.0/guava-21.0.jar:/Users/romangrewal/.m2/repository/com/google/code/findbugs/annotations/2.0.3/annotations-2.0.3.jar:/Users/romangrewal/.m2/repository/org/apache/commons/commons-lang3/3.4/commons-lang3-3.4.jar:/Users/romangrewal/.m2/repository/com/google/code/gson/gson/2.8.2/gson-2.8.2.jar:/Users/romangrewal/.m2/repository/javax/measure/jsr-275/0.9.1/jsr-275-0.9.1.jar:/Users/romangrewal/.m2/repository/com/google/inject/extensions/guice-multibindings/4.0/guice-multibindings-4.0.jar:/Users/romangrewal/.m2/repository/commons-io/commons-io/2.4/commons-io-2.4.jar:/Users/romangrewal/.m2/repository/com/amazonaws/aws-java-sdk-kinesisvideo/1.11.255/aws-java-sdk-kinesisvideo-1.11.255.jar:/Users/romangrewal/.m2/repository/io/netty/netty-codec-http/4.1.17.Final/netty-codec-http-4.1.17.Final.jar:/Users/romangrewal/.m2/repository/io/netty/netty-codec/4.1.17.Final/netty-codec-4.1.17.Final.jar:/Users/romangrewal/.m2/repository/io/netty/netty-handler/4.1.17.Final/netty-handler-4.1.17.Final.jar:/Users/romangrewal/.m2/repository/io/netty/netty-buffer/4.1.17.Final/netty-buffer-4.1.17.Final.jar:/Users/romangrewal/.m2/repository/io/netty/netty-common/4.1.17.Final/netty-common-4.1.17.Final.jar:/Users/romangrewal/.m2/repository/io/netty/netty-transport/4.1.17.Final/netty-transport-4.1.17.Final.jar:/Users/romangrewal/.m2/repository/io/netty/netty-resolver/4.1.17.Final/netty-resolver-4.1.17.Final.jar:/Users/romangrewal/.m2/repository/com/amazonaws/jmespath-java/1.11.255/jmespath-java-1.11.255.jar com.amazonaws.kinesisvideo.demoapp.DemoAppMain

objc[707]: Class JavaLaunchHelper is implemented in both /Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/bin/java (0x10ff0f4c0) and /Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/libinstrument.dylib (0x10ff9b4e0). One of the two will be used. Which one is undefined.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
ERROR / KinesisVideo: 2018-03-07T17:43Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: no KinesisVideoProducerJNI in java.library.path
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:45)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.initializeLibrary(NativeKinesisVideoProducerJni.java:1069)
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:220)
[... cut ...] 193)
	at com.amazonaws.kinesisvideo.client.NativeKinesisVideoClient.initialize(NativeKinesisVideoClient.java:112)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:107)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:80)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:31)

Process finished with exit code 1

unicornss, 2018-03-07:
can you tell us the files under the directory `/absolute/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build` by 
```
ls -l /absolute/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
```

romangrewal, 2018-03-08:
-rw-r--r--   1 romangrewal  staff    14093 Mar  7 15:56 CMakeCache.txt
drwxr-xr-x  19 romangrewal  staff      608 Mar  7 16:22 CMakeFiles
-rw-r--r--   1 romangrewal  staff    15756 Mar  7 15:13 CMakeLists.txt
-rw-r--r--   1 romangrewal  staff   390972 Mar  7 16:22 Makefile
-rw-r--r--   1 romangrewal  staff     1561 Mar  7 15:56 cmake_install.cmake
drwxr-xr-x  47 romangrewal  staff     1504 Mar  7 15:48 downloads
-r-xr-xr-x   1 romangrewal  staff    21391 Mar  7 15:13 install-script
-rwxr-xr-x   1 romangrewal  staff      845 Mar  7 15:13 java-install-script
-rwxr-xr-x   1 romangrewal  staff   121756 Mar  7 15:57 kinesis_video_gstreamer_sample_app
-rwxr-xr-x   1 romangrewal  staff   111692 Mar  7 15:57 kinesis_video_gstreamer_sample_rtsp_app
-rwxr-xr-x   1 romangrewal  staff   259852 Mar  7 15:57 libKinesisVideoProducerJNI.dylib
-rwxr-xr-x   1 romangrewal  staff  1373260 Mar  7 15:57 libproducer.dylib
-rwxr-xr-x   1 romangrewal  staff     2400 Mar  7 15:13 min-install-script
drwxr-xr-x  19 romangrewal  staff      608 Mar  7 15:13 samples
-rwxr-xr-x   1 romangrewal  staff  3427700 Mar  7 15:57 start
-rw-r--r--   1 romangrewal  staff   143289 Mar  7 16:02 test_h264_jpg.mkv
-rw-r--r--   1 romangrewal  staff   143305 Mar  7 16:02 test_jpeg.mkv

zhiyua-git, 2018-03-08:
Can you tell us the linkage on libKinesisVideoProducerJNI by 
```
otool -L /absolute/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.dylib
```

You can try to re-create libKinesisVideoProducerJNI.dylib meanwhile by 
```
rm -rf CMakeFiles CMakeCache.txt libKinesisVideoProducerJNI.dylib
./java-install-script
```

romangrewal, 2018-03-09:
otool -L libKinesisVideoProducerJNI.dylib
outputs:
libKinesisVideoProducerJNI.dylib:
	@rpath/libKinesisVideoProducerJNI.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.0.0)

recreating libKinesisVideoProducerJNI.dylib produced the same unsatisfied link error

zhiyua-git, 2018-03-09:
Everything looks correct. 

Can you set in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/producer/jni/NativeLibraryLoader.java#L55
to
System.load('/absolute/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.dylib') 
as a workaround now? 

You need to provide full path including file name inside direct library load.

romangrewal, 2018-03-13:
The workaround worked with the full path including the file name.

romangrewal, 2018-03-13:
Following the example here:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/10

I was able to get all the libraries built and stream from my webcam to kinesis. However, on the kinesis portal, it says:

[image]

Perhaps there is an issue with the codecPrivateData byte array?
Here:
https://stackoverflow.com/questions/32490988/smooth-streaming-codec-private-data
It says that the NAL_START_CODE for AVC1 should be 00,00,00,01 but I am seeing 01,42,00,20.
Changing it to 00,00,00,01 gives me the error:

[image]

MushMal, 2018-03-13:
@romangrewal I would like to provide some information before suggesting the steps. This is assuming the producer is streaming OK.

* Codec Private Data is codec specific bits. In case of H264 this is SPS and PPS information.
* Both frames and the CPD can be in Annex-B format or AvCC format - those are different for frames vs. the CPD.
* Console playback can only consume AvCC format frames and codec private data.
* Many encoders will produce AvCC format.

Now, with this in mind, we have a set of flags that control the adaptation behavior defined in StreamInfo.java::NalAdaptationFlags. You can use the combination of frame and CPD depending on your encoders output (NAL_ADAPTATION_ANNEXB_CPD_AND_FRAME_NALS).

Please try and let us know so we can eliminate the CPD.

romangrewal, 2018-03-13:
When
StreamInfo.NalAdaptationFlags.NAL_ADAPTATION_FLAG_NONE
is changed to
StreamInfo.NalAdaptationFlags.NAL_ADAPTATION_ANNEXB_CPD_AND_FRAME_NALS

the console outputs the error
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed to create a stream.
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:60)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed to create a stream.
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.createKinesisVideoStream(Native Method)
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.createStream(NativeKinesisVideoProducerJni.java:318)
	at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.createStreamSync(NativeKinesisVideoProducerJni.java:345)
	at com.amazonaws.kinesisvideo.client.NativeKinesisVideoClient.registerMediaSource(NativeKinesisVideoClient.java:121)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:53)

MushMal, 2018-03-13:
You mean the application fails with this error. This is odd - I don't see any error code that will tell me what's causing the stream object creation failure. 

* Can you capture all of the logs
* Could you capture and include the CPD bits so I can see what format those are in

romangrewal, 2018-03-13:
The entire log from the console:

/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/bin/java "-javaagent:/Applications/IntelliJ IDEA CE.app/Contents/lib/idea_rt.jar=51469:/Applications/IntelliJ IDEA CE.app/Contents/bin" -Dfile.encoding=UTF-8 -classpath /Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/charsets.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/deploy.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/cldrdata.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/dnsns.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/jaccess.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/jfxrt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/localedata.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/nashorn.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/sunec.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/sunjce_provider.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/sunpkcs11.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/ext/zipfs.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/javaws.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jce.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jfr.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jfxswt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/jsse.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/management-agent.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/plugin.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/resources.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/jre/lib/rt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/ant-javafx.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/dt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/javafx-mx.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/jconsole.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/packager.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/sa-jdi.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_144.jdk/Contents/Home/lib/tools.jar:/Users/romangrewal/t1cgacademy/sp15/t1-academy/kinesis/java/amazon-kinesis-video-streams-producer-sdk-java/target/classes:/Users/romangrewal/.m2/repository/org/jboss/netty/netty/3.2.0.Final/netty-3.2.0.Final.jar:/Users/romangrewal/.m2/repository/com/github/sarxos/webcam-capture/0.3.12/webcam-capture-0.3.12.jar:/Users/romangrewal/.m2/repository/com/nativelibs4java/bridj/0.7.0/bridj-0.7.0.jar:/Users/romangrewal/.m2/repository/org/slf4j/slf4j-api/1.7.2/slf4j-api-1.7.2.jar:/Users/romangrewal/.m2/repository/xuggle/xuggle-xuggler/5.4/xuggle-xuggler-5.4.jar:/Users/romangrewal/.m2/repository/com/amazonaws/aws-java-sdk-core/1.11.172/aws-java-sdk-core-1.11.172.jar:/Users/romangrewal/.m2/repository/commons-logging/commons-logging/1.1.3/commons-logging-1.1.3.jar:/Users/romangrewal/.m2/repository/org/apache/httpcomponents/httpclient/4.5.2/httpclient-4.5.2.jar:/Users/romangrewal/.m2/repository/org/apache/httpcomponents/httpcore/4.4.4/httpcore-4.4.4.jar:/Users/romangrewal/.m2/repository/commons-codec/commons-codec/1.9/commons-codec-1.9.jar:/Users/romangrewal/.m2/repository/software/amazon/ion/ion-java/1.0.2/ion-java-1.0.2.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/core/jackson-databind/2.6.7.1/jackson-databind-2.6.7.1.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/core/jackson-annotations/2.6.0/jackson-annotations-2.6.0.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/core/jackson-core/2.6.7/jackson-core-2.6.7.jar:/Users/romangrewal/.m2/repository/com/fasterxml/jackson/dataformat/jackson-dataformat-cbor/2.6.7/jackson-dataformat-cbor-2.6.7.jar:/Users/romangrewal/.m2/repository/joda-time/joda-time/2.8.1/joda-time-2.8.1.jar:/Users/romangrewal/.m2/repository/com/google/inject/guice/4.0/guice-4.0.jar:/Users/romangrewal/.m2/repository/javax/inject/javax.inject/1/javax.inject-1.jar:/Users/romangrewal/.m2/repository/aopalliance/aopalliance/1.0/aopalliance-1.0.jar:/Users/romangrewal/.m2/repository/com/google/code/findbugs/jsr305/3.0.2/jsr305-3.0.2.jar:/Users/romangrewal/.m2/repository/com/google/guava/guava/21.0/guava-21.0.jar:/Users/romangrewal/.m2/repository/com/google/code/findbugs/annotations/2.0.3/annotations-2.0.3.jar:/Users/romangrewal/.m2/repository/org/apache/commons/commons-lang3/3.4/commons-lang3-3.4.jar:/Users/romangrewal/.m2/repository/com/google/code/gson/gson/2.8.2/gson-2.8.2.jar:/Users/romangrewal/.m2/repository/javax/measure/jsr-275/0.9.1/jsr-275-0.9.1.jar:/Users/romangrewal/.m2/repository/com/google/inject/extensions/guice-multibindings/4.0/guice-multibindings-4.0.jar:/Users/romangrewal/.m2/repository/commons-io/commons-io/2.4/commons-io-2.4.jar:/Users/romangrewal/.m2/repository/com/amazonaws/aws-java-sdk-kinesisvideo/1.11.255/aws-java-sdk-kinesisvideo-1.11.255.jar:/Users/romangrewal/.m2/repository/io/netty/netty-codec-http/4.1.17.Final/netty-codec-http-4.1. [... cut ...]
Process finished with exit code 130 (interrupted by signal 2: SIGINT)

The CPD bits are defined in the code as:
byte[] codecPrivateData = { 0x01, 0x42, 0x00, 0x20, (byte) 0xff, (byte) 0xe1, 0x00, 0x23, 0x27, 0x42, 0x00, 0x20, (byte) 0x89, (byte) 0x8b, 0x60, 0x28, 0x02, (byte) 0xdd, (byte) 0x80, (byte) 0x9e, 0x00, 0x00, 0x4e, 0x20, 0x00, 0x0f, 0x42, 0x41, (byte) 0xc0, (byte) 0xc0, 0x01, 0x77, 0x00, 0x00, 0x5d, (byte) 0xc1, 0x7b, (byte) 0xdf, 0x07, (byte) 0xc2, 0x21, 0x1b, (byte) 0x80, 0x01, 0x00, 0x04, 0x28, (byte) 0xce, 0x1f, 0x20 };

MushMal, 2018-03-13:
`createKinesisVideoStream(): Failed to create a stream with status code 0x32000010`

This means that the CPD has invalid AnnexB format NALu to adapt.

Looking at the CPD you've provided I can see that it's already in the AVCC format so you don't need to adapt the CPD.

Please check the frames whether they are in Annex-B format and need to be adapted. You can simply check whether the frame starts with the Annex-B start code of 001 or 0001. If so, it needs to be adapted.

romangrewal, 2018-03-14:
The first 4 bytes of a video frame are 
0x00 0x00 0x00 0x01

When we pass in 
StreamInfo.NalAdaptationFlags.NAL_ADAPTATION_ANNEXB_NALS 
to withNalAdaptationFlags()
it reads and displays the first frame of the video in AWS kinesis but then displays
[image]

MushMal, 2018-03-14:
@romangrewal It seems that your CPD is AvCC format but the frames are in Annex-B, in which case you were right in selecting the NAL_ADAPTATION_ANNEXB_NALS flag.

What's strange is that the browsers decoder returns an error. Let me try to dig a little more info for you but can I ask you to do the following:

* Describe your end-to-end scenario - what are you trying to achieve and what's your media source. Perhaps we could reproduce it here.
* Could you please include the full logs from your run with this flag so we can see whether there is any issue on the producer side.
* It's also possible that somehow the frames get corrupted. Could you capture your stream and see if you can play it back using VLC or some other tool?

MushMal, 2018-03-14:
One more thing.. The error that's shown indicates the fragment size of 1.1Kb - I would expect the fragment size to be much larger for h264 video - a P or a B frame can be this small but not the entire fragment. Can you share more info?

zhiyua-git, 2018-04-26:
Close issue as library loading issue is resolved. Please open another thread for other issues.

closed 2018-04-26 by zhiyua-git

TechReporter, 2021-07-18:
UnsatisfiedLinkError: com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.getNativeLibraryVersion()Ljava/lang/String
I am following this workspace https://github.com/backdoorcodr/amazon-kinesis-video-streams-producer-sdk-java
Expectation is to run kinesis video stream using webcam in java.
Please help