awslabs/amazon-kinesis-video-streams-producer-sdk-java #32: PutMedia error (400 Bad Request) when default Locale has decimal comma

IlkkaAnttonen, 2018-03-23:
When running the example application I ran into an error where I only received 'HTTP 400 Bad Request' after the first PutMedia call. After hunting for quite a while I noticed that the culprit was the timestamp in the headers with the format "x-amzn-producer-start-timestamp: 1521762736,754". After modifying ito to "x-amzn-producer-start-timestamp: 1521762736.754" it worked.

From the source code (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/12f7c0f55b74fac0b7c95a7f3ad6310d8c7869f8/src/main/java/com/amazonaws/kinesisvideo/client/PutMediaClient.java#L56) I noticed that the timestamp is created without giving it a Locale which in my case causes it to use Finnish locale with a decimal comma. It can be fixed with changing the line to:

clientBuilder.header(PRODUCER_START_TIMESTAMP_HEADER,
String.format(Locale.US, "%.3f", mBuilder.mTimestamp / MILLI_TO_SEC));

Should I create a PR for this?

zhiyua-git, 2018-03-23:
Thanks for reaching out! If you can submit a PR, that will be great.

Also do you mind describing what are you going to do with Kinesis Video Streams, it will help us to provide better support.

Thanks,
-ZH

IlkkaAnttonen, 2018-03-26:
Hi, created a PR. I'm using the web camera on a Mac to upload video stream to Kinesis Video and hooking it up with different ML models for image recognition. I wanted to have a simple way to provide test material for the models and using webcam seemed the most obvious.

MushMal, 2018-03-26:
@IlkkaAnttonen thank you very much for the commit. Please go ahead and resolve this issue once the changes are merged

closed 2018-03-26 by IlkkaAnttonen