awslabs/amazon-kinesis-video-streams-producer-sdk-java #35: Docker Demo requires running install-script

smp, 2018-05-23:
Hello,

I ran through the docker demo application today and hit an issue whereby I was missing links to the requisite libraries. The solution was to run the install scripts in the following directory and accept an agreement to use them.

`/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build`

I'm happy to update the README documentation for the docker demo, but wanted to check that we couldn't automagically do this on behalf of the user.

MushMal, 2018-05-23:
@smp thanks a lot for the information. Please do proceed with a pull request. We will evaluate it asap.

smp, 2018-05-23:
done! https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/pull/36

MushMal, 2018-05-23:
Perfect! Thanks. I will go ahead and resolve this

closed 2018-05-23 by MushMal