awslabs/amazon-kinesis-video-streams-producer-sdk-java #38: Error code on MKV files last fragment for every file when we push using PUTMEDIA API

nikparmar, 2018-06-16:
Hello Guys,
We are using PUTMEDIA API code to push the data to Kinesis Video Stream, the demo file works fine and we are generating feeds from our webcam and storing the MKV files like 1 sec, 5 sec but the last fragment of every file is failing with issue:
`onAckEvent AckEvent{ackEventType=ERROR, fragmentTimecode=520, fragmentNumber='91343852333183132549821456210792172991539981790', errorCode=FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS, errorId=4004}`

What are we doing?
1. We are right now testing with random files(MKV) which we have generated
2. We are also trying to encode them into `base64` and inside decoding and then passing them to the code
I have attached file for reference:
[Uploading bigbuckbunnynew_0.mkv.zip…](File which we are using you can download and extract it)

zhiyua-git, 2018-06-16:
Hi @nikparmar,

The FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS is caused by b-frame from you input, we don't support b-frame now. Can you tune the encoder to stop generating b-frame and retry?

Let me know if it helps, thanks.

-ZH

nikparmar, 2018-06-16:
Hi @zhiyua-git we are using `ffmpeg` to convert the h264 to MKV do you know the suitable command for same?
Thanks

unicornss, 2018-06-16:
when generating the h264 encoder you may have to use the profile as basic  
could you try `-profile:v baseline`

nikparmar, 2018-06-18:
Thanks it worked!

closed 2018-06-18 by zhiyua-git