awslabs/amazon-kinesis-video-streams-producer-sdk-java #40: PUTMEDIA Api takes around 6-8 seconds in fragment insertion

nikparmar, 2018-06-18:
Hello Guys,
We are using `PutMedia` call to insert data into Kinesis using `JavaCode` but the time taken for a `6` second fragment or `1` is almost similar which is around ~`6-7 seconds`. Why does `PutMedia` takes so much time?

zhiyua-git, 2018-06-18:
PutMedia has around 5-second timeout for the connection. Before it get timeout, if you write to the [InputStream](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/demo/com/amazonaws/kinesisvideo/demoapp/PutMediaDemo.java#L117), current PutMedia would continue sending the data.

We suggest using [KinesisVideoClient](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/demo/com/amazonaws/kinesisvideo/demoapp/DemoAppMain.java) better than directly use PutMedia API, as the underlying implementation of KinesisVideoClient handles issues like reconnecting, AWS credential refresh, retry on some error case, etc..

nikparmar, 2018-06-19:
Hey @zhiyua-git when using the `KinesisVideoClient` we are not getting any acknowledgement from the client and the video is also not available on `Kinesis Console`

zhiyua-git, 2018-06-19:
@nikparmar Could you share the log using KinesisVideoClient here or to kinesis-video-support@amazon.com? We can help find out what is broken. Thank you.

nikparmar, 2018-06-19:
Error:
```
DEBUG / KinesisVideo: PutFrame index: 0, pts: 15294333374080000, dts: 15294333374080000, duration: 200000, keyFrame: true, flags: 1
DEBUG / KinesisVideo: Kinesis Video client and stream metrics
>> Overall storage size: 1073741824
>> Available storage size: 1073741824
>> Allocated storage size: 0
>> Total view allocation size: 40072
>> Total streams frame rate: 25
>> Total streams transfer rate: 2000000
>> Current view duration: 0
>> Overall view duration: 0
>> Current view size: 0
>> Overall view size: 0
>> Current frame rate: 25.0
>> Current transfer rate: 2000000
sleeping 52
-- handleRequest started @2018.06.19.18.35.38
-- handleRequest ended @2018.06.19.18.35.38
com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream.
at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.putKinesisVideoFrame(Native Method)
at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.putFrame(NativeKinesisVideoProducerJni.java:440)
at com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerStream.putFrame(NativeKinesisVideoProducerStream.java:259)
at com.amazonaws.kinesisvideo.mediasource.ProducerStreamSink.onFrame(ProducerStreamSink.java:35)
at com.amazonaws.kinesis.custom.S3FileMediaSource.putFrame(S3FileMediaSource.java:119)
at com.amazonaws.kinesis.custom.S3FileMediaSource.access$3(S3FileMediaSource.java:117)
at com.amazonaws.kinesis.custom.S3FileMediaSource$1.onFrameDataAvailable(S3FileMediaSource.java:103)
at com.amazonaws.kinesis.custom.S3FrameSource.generateFrameAndNotifyListener(S3FrameSource.java:92)
at com.amazonaws.kinesis.custom.S3FrameSource.access$0(S3FrameSource.java:81)
at com.amazonaws.kinesis.custom.S3FrameSource$1.run(S3FrameSource.java:76)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
at java.lang.Thread.run(Thread.java:748)
```

MushMal, 2020-04-19:
@nikparmar the snippet of log unfortunately doesn't contain anything that can pinpoint the issue. Could you please modify the logging level by changing DEFAULT_LOGGER_LOG_LEVEL to 2 in ClientInfo.java and re-run your scenario. This will enable verbose logging which would show the native side error.

MushMal, 2020-05-20:
Resolving due to staleness

closed 2020-05-20 by MushMal