awslabs/amazon-kinesis-video-streams-producer-sdk-java #42: live stream using java

jis1218, 2018-06-26:
Hello<br><br>

I am trying to do live stream from laptop camera using Java.<br><br>

With a help from issue#10 <a href="https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/10">(Link)</a>, It seemes to be successful at sending frame to kinesis video stream. Here is what I get from console.
<br><br>

DEBUG / KinesisVideo: Kinesis Video client and stream metrics<br>
	>> Overall storage size: 1073741824<br>
	>> Available storage size: 1073333536<br>
	>> Allocated storage size: 408288<br>
	>> Total view allocation size: 8072<br>
	>> Total streams frame rate: 6<br>
	>> Total streams transfer rate: 212372<br>
	>> Current view duration: 0<br>
	>> Overall view duration: 25890000<br>
	>> Current view size: 0<br>
	>> Overall view size: 407661<br>
	>> Current frame rate: 6.974047433144237<br>
	>> Current transfer rate: 212372<br>
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 21536, Duration 200000 <br>
DEBUG / KinesisVideo: Streamed 16384 bytes for stream insup with uploadHandle 0<br>
DEBUG / KinesisVideo: Streamed 5152 bytes for stream insup with uploadHandle 0<br>
putFrame() is working : <br>
DEBUG / KinesisVideo: PutFrame index: 20, pts: 15299987551760000, dts: 15299987551760000, duration: 200000, keyFrame: false, flags: 0<br>
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 21487, Duration 200000 <br>
DEBUG / KinesisVideo: Streamed 16384 bytes for stream insup with uploadHandle 0<br>
DEBUG / KinesisVideo: Streamed 5103 bytes for stream insup with uploadHandle 0<br>
DEBUG / KinesisVideo: Received ACK bits: 71<br>
{"EventType":"RECEIVED","FragmentTimecode":0,"FragmentNumber":"91343852333181853708243272841796361057028177684"}<br>
<br>
<br>
DEBUG / KinesisVideo: Received ACK bits: 75<br>
{"EventType":"BUFFERING","FragmentTimecode":2645,"FragmentNumber":"91343852333181853713195032998937882518142135007"}<br>
<br><br>
putFrame() is working : <br>
DEBUG / KinesisVideo: PutFrame index: 21, pts: 15299987552660000, dts: 15299987552660000, duration: 200000, keyFrame: false, flags: 0 <br>
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 21647, Duration 200000 <br>
DEBUG / KinesisVideo: Streamed 16384 bytes for stream insup with uploadHandle 0 <br>
DEBUG / KinesisVideo: Streamed 5263 bytes for stream insup with uploadHandle 0 <br>
putFrame() is working : <br>
DEBUG / KinesisVideo: PutFrame index: 22, pts: 15299987553140000, dts: 15299987553140000, duration: 200000, keyFrame: false, flags: 0<br>
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 21551, Duration 200000 <br>
DEBUG / KinesisVideo: Streamed 16384 bytes for stream insup with uploadHandle 0<br>
DEBUG / KinesisVideo: Streamed 5167 bytes for stream insup with uploadHandle 0<br>
DEBUG / KinesisVideo: Received ACK bits: 72<br>
{"EventType":"PERSISTED","FragmentTimecode":0,"FragmentNumber":"91343852333181853708243272841796361057028177684"} <br>
<br><br>

However, in aws console, nothing comes up. Here is a capture of the aws console

[image]

All codes are same as <a href="https://github.com/backdoorcodr/amazon-kinesis-video-streams-producer-sdk-java">https://github.com/backdoorcodr/amazon-kinesis-video-streams-producer-sdk-java</a> and configuration is like this.

<br><br>

    	final CameraMediaSourceConfiguration configuration =  <br>
    			new CameraMediaSourceConfiguration.Builder()<br>
    			.withFrameRate(FPS_19)<br>
    			.withRetentionPeriodInHours(24)<br>
    			.withCameraId(webcam.getName())<br>
    			.withIsEncoderHardwareAccelerated(false)<br>
    			.withEncodingMimeType("video/avc")<br>
    			.withNalAdaptationFlags<br>(StreamInfo.NalAdaptationFlags.NAL_ADAPTATION_ANNEXB_NALS)<br>
    			.withIsAbsoluteTimecode(false)<br>
    			.withEncodingBitRate(200000)<br>
    			.withHorizontalResolution(640)<br>
    			.withVerticalResolution(480)<br>
    			.withCodecPrivateData(codecPrivateData)<br>
    			.build();<br>

What will be the problem?

vijaygj, 2018-06-26:
If you got persisted ACK like below, that means KVS received the data,

> {"EventType":"PERSISTED","FragmentTimecode":0,"FragmentNumber":"91343852333181853708243272841796361057028177684"} 

To solve the next part, there could be some common patterns we have observed,

1. AWS Region where video is sent and region in the console are not the same.
2. Can you try to play form the "earliest" using the console button just to ensure it is because of some lag.

vijaygj, 2018-06-27:
>Bit rate is around 400-600kbps and Fragment duration is around 3-5s.
But I still cannot see anything on 'video preview'.

This is good progress, at least console is receiving some data. In the absence of it console displays following error message,

"No fragments found
No video fragments were found. Verify that video is currently being streamed and that the fragment timecodes are correct."

Can you please download the video using GetMedia API (AWS CLI can be used for the same) and confirm if VLC player or any other off the shelf player can play it? Trying to confirm if it is some encoding / decoding issue.

jis1218, 2018-07-04:
Thank you vijaygj!

Finally I was able to download the live stream video using GetMedia API, it took a lot of time to find the way of using the API though.

So it should be encoding/decoding issue?
Can I use 'rekognition' without solving the problem that 'Video preview' doesn't show anything? (Actually I want to solve the problem :) )

Thank you in advance!

vijaygj, 2018-07-04:
Thank you for your patience.

>Finally I was able to download the live stream video using GetMedia API, it took a lot of time to find the way of using the API though.

Can you please confirm if after downloading the video, you were able to play it with off-the-shelf player like VLC media player?

> Can I use 'rekognition' without solving the problem that 'Video preview' doesn't show anything? 

You should be able to use Rekognition even if there is any issue in the browser playback.

> (Actually I want to solve the problem :) )

Can you please send us the screen shot of your AWS console error, and if possible the downloaded video file? We can check further on what is going on. Please use kinesis-video-support@amazon.com to send us these details.

jis1218, 2018-07-05:
Thank you for your answer.

Downloaded video is played well in VLC player.

Accidently, I found out that 'Video preview' worked on Desktop with Ubuntu 18.04 and 16G RAM.
My previous environment which 'Video preview' didn't work was Laptop with Window10 and 8G RAM.
I don't think it is a problem with Webcam cause I used same one on both environment.

I'm not sure what is the problem and can you tell me any possible reason for me and someone who will possibly get help from this issue.

After that, I will close the issue.

I really appreciate your help.

vijaygj, 2018-07-05:
Thank you for the update. Can you please send the screen capture for Windows and Ubuntu and send us on the email provided in the previous update? We will try to repro the issue ourselves on the platform you mentioned. Can you please provide further details on which browser is being used?

jis1218, 2018-07-05:
Hello vijaygj!
I just send the capture file and video file.
On Window10, I used 'Microsoft Edge'.
I just tried on 'Chrome', Error said 'Fragment decoding error'.
Thank you!!

t1minator, 2018-12-28:
This is related. Live streaming from a net camera using a rtsp uri:
This is one of the demos with the CPP SDK, but cannot get a rtsp stream to push with the java sdk.
Has anyone solved this?
Thanks

Tim

mah385, 2019-08-23:
`gst-launch-1.0 rtspsrc location="YourRtspUri" short-header=TRUE protocols=tcp ! rtph264depay ! decodebin ! videorate ! videoscale ! vtenc_h264_hw allow-frame-reordering=FALSE max-keyframe-interval=25 bitrate=1024 realtime=TRUE ! video/x-h264,stream-format=avc,alignment=au,profile=baseline,width=640,height=480,framerate=1/25 ! multifilesink location=./frame-%03d.h264 index=1`

I installed GStreamer and tried executing the above command given in DemoAppMain.java as a part of comment but I am getting error as **[WARNING: erroneous pipeline: no element "vtenc_h264_hw"]**

can anyone help me.

MushMal, 2019-08-23:
@mah385 not sure if this is the right project, nor the right issue. However, it seems that you are failing to load hardware encoder -> vtenc_h264_hw. This would be available on OSx. If you are running on Linux, try some other encoder - perhaps a software encoder like x264

MushMal, 2020-04-19:
Resolving this issue as the original question was answered

closed 2020-04-19 by MushMal