awslabs/amazon-kinesis-video-streams-producer-sdk-java #46: No Ack in logs and no Fragments in kinesis video stream using Java producer

VijayalakshmV, 2018-08-29:
Hi ,

I have set up the code in Eclispe using Java 8 , have full access to Kinesis . On running the DemoApp .Ack (eventType) not displayed in logs and no Fragments message is displayed in  Kinesis video preview

Logs below
DEBUG / KinesisVideo: PutFrame index: 100, pts: 15355470940920000, dts: 15355470940920000, duration: 200000, keyFrame: true, flags: 1
DEBUG / KinesisVideo: Kinesis Video client and stream metrics
	>> Overall storage size: 1073741824
	>> Available storage size: 1073478412
	>> Allocated storage size: 263412
	>> Total view allocation size: 568041544
	>> Total streams frame rate: 22
	>> Total streams transfer rate: 2000000
	>> Current view duration: 44270000
	>> Overall view duration: 44270000
	>> Current view size: 259412
	>> Overall view size: 259412
	>> Current frame rate: 22.964761177665782
	>> Current transfer rate: 2000000
DEBUG / KinesisVideo: PutFrame index: 101, pts: 15355470941330000, dts: 15355470941330000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 102, pts: 15355470941800000, dts: 15355470941800000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 103, pts: 15355470942230000, dts: 15355470942230000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 104, pts: 15355470942650000, dts: 15355470942650000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 105, pts: 15355470943070000, dts: 15355470943070000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 106, pts: 15355470943480000, dts: 15355470943480000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 107, pts: 15355470943990000, dts: 15355470943990000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 108, pts: 15355470944400000, dts: 15355470944400000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: Received all data, close
ERROR / KinesisVideo: Kinesis Video service client returned an error com.amazonaws.kinesisvideo.service.exception.AccessDeniedException: Access is denied Reporting to Kinesis Video PIC.
INFO / KinesisVideo: putStreamResult uploadHandle 0 status 403
DEBUG / KinesisVideo: PutFrame index: 109, pts: 15355470944850000, dts: 15355470944850000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 110, pts: 15355470945260000, dts: 15355470945260000, duration: 200000, keyFrame: false, flags: 0
DEBUG / KinesisVideo: PutFrame index: 111, pts: 15355470945700000, dts: 15355470945700000, duration: 200000, keyFrame: false, flags: 0

zhiyua-git, 2018-08-29:
Hi,

From the log, you got com.amazonaws.kinesisvideo.service.exception.AccessDeniedException: Access is denied.

Can you double check if you followed the [steps](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/README.md#launching-demoapp-sample-application) to run with correct AccessKey/SecretKey?

Thanks,
ZH

sandeepyadav03, 2018-08-31:
With the latest Fix , All the Issues are resolved.

VijayalakshmV, 2018-08-31:
Hi All,

Thanks With the latest code , all the issues are resolved.

closed 2018-08-31 by VijayalakshmV

zhiyua-git, 2018-08-31:
Glad to hear it works for you! Thanks for letting us know.