awslabs/amazon-kinesis-video-streams-producer-sdk-java #51: KinesisVideoProducerJNI.dll: %1 is not a valid Win32 application

johnkm516, 2018-12-17:
I'm trying to run the DemoApp using the pre-built library provided, but I get the error in the title : KinesisVideoProducerJNI.dll: %1 is not a valid Win32 application. 

I've tried pulling https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp and building it, and after wasting more than half an hour building it it didn't even give me a KinesisVideoProducer.dll. 

It's extremely unclear from the README what exactly I'm supposed to do to get this working.

zhiyua-git, 2018-12-17:
Hi @johnkm516 ,

How did you get this error "KinesisVideoProducerJNI.dll: %1 is not a valid Win32 application."? The pre-built dll is for windows 64 bit.

To build dll file for 32 bit windows:
* If you are using msvc build, please cd to kinesis-video-native-build directory and run "windows-java-install.bat 32". You would find the KinesisVideoProducerJNI.dll in kinesis-video-native-build/Release/ after the build.
* If you are using mingw build, please open mingw32.exe and cd to kinesis-video-native-build directory, then run ./java-install-script. You would find the KinesisVideoProducerJNI.dll in kinesis-video-native-build/ after the build.

Let me know if it helps.

-ZH

johnkm516, 2018-12-17:
The error came up when running my application using the following JVM command : -Djava.library.path="pathtonativelibraryDLL"

> If you are using mingw build, please open mingw32.exe and cd to kinesis-video-native-build directory, then run ./java-install-script

This worked! Just a suggestion, the README here : https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msys2.md
should be updated with that information. I was using the "./msys2-install-script -a" command. 

Now I have a different error : 

`ERROR / KinesisVideo: 2018-12-18T03:58Z T28: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: C:\Users\John\Documents\workspace-sts-4.0.1.RELEASE\livestream\KinesisVideoProducerJNI.dll: Can't find dependent libraries: C:\Users\John\Documents\workspace-sts-4.0.1.RELEASE\livestream\KinesisVideoProducerJNI.dll: Can't find dependent libraries`

What dependent libraries am I missing and how do I add them?

zhiyua-git, 2018-12-18:
Hi @johnkm516 ,

Can you elaborate on what path were you using? With original dll, the command should be -Djava.library.path=src/main/resources/lib/windows. Can you give it a second try as you are using 64bits?

When my path is incorrect I would have error message like "ERROR / KinesisVideo: 2018-12-18T11:54Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: no KinesisVideoProducerJNI in java.library.path", but it wouldn't contain the path inside, can you confirm if you are inserting the absolute path inside the code? JNI library itself is kind of self contained and you shouldn't need to find dependent libraries.

Thanks,
-ZH

johnkm516, 2018-12-19:
Hi, 

I actually ended up using MediaLive instead but I do want to figure this out for future reference. The path I'm using leads directly to the kinesis-video-native-build folder, in 

`C:\msys64\home\John\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build`

The path is correct, and it indeed loads the .dll, and unlike the error message you posted, my error message above says, "can't find dependent libraries", not "no KinesisVideoProducerJNI in java.library.path". 

`JNI library itself is kind of self contained and you shouldn't need to find dependent libraries.`

That's what I assumed. This error is indeed very strange.

disa6302, 2020-04-01:
Stale issue. Closing this. Feel free to reopen in the future

closed 2020-04-01 by disa6302

TechReporter, 2021-07-18:
UnsatisfiedLinkError: com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.getNativeLibraryVersion()Ljava/lang/String
I am following this workspace https://github.com/backdoorcodr/amazon-kinesis-video-streams-producer-sdk-java
Expectation is to run kinesis video stream using webcam in java.
Please help

Lostefra, 2022-09-13:
I am commenting since I came across the error `java.lang.UnsatisfiedLinkError: C:\path\to\KinesisVideoProducerJNI.dll: Can't find dependent libraries` as 13th September 2022 on Windows 10. I cloned the repository today.

I managed to solve the error by picking an old version `KinesisVideoProducerJNI.dll`, the one from 9th July 2019.

Link:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/commits/026f890c7b8975e423c859062a807220e290bd23/src/main/resources/lib/windows/KinesisVideoProducerJNI.dll