awslabs/amazon-kinesis-video-streams-producer-sdk-java #52: could run the demo

liuxthu, 2019-01-04:
when trying to run the demo, i meet a request error, can you help me to diagnose it. the error log is shown as follows:

```"C:\Program Files\Java\jdk1.8.0_191\bin\java.exe" -Djava.library.path=F:\amazon-kinesis-video-streams-producer-sdk-java-master\lib -ea -Daws.accessKeyId=AKIAIEXK6N3XYQACQI6Q -Daws.secretKey=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 -Djava.library.path=F:\amazon-kinesis-video-streams-producer-sdk-java-master\src\main\resources\lib\windows\ "-javaagent:C:\Program Files\JetBrains\IntelliJ IDEA 2018.3\lib\idea_rt.jar=57594:C:\Program Files\JetBrains\IntelliJ IDEA 2018.3\bin" -Dfile.encoding=UTF-8 -classpath "C:\Program Files\Java\jdk1.8.0_191\jre\lib\charsets.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\deploy.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\access-bridge-64.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\cldrdata.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\dnsns.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\jaccess.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\jfxrt.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\localedata.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\nashorn.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\sunec.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\sunjce_provider.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\sunmscapi.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\sunpkcs11.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\ext\zipfs.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\javaws.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\jce.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\jfr.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\jfxswt.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\jsse.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\management-agent.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\plugin.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\resources.jar;C:\Program Files\Java\jdk1.8.0_191\jre\lib\rt.jar;F:\amazon-kinesis-video-streams-producer-sdk-java-master\target\classes;C:\Users\Administrator\.m2\repository\com\amazonaws\aws-java-sdk-core\1.11.172\aws-java-sdk-core-1.11.172.jar;C:\Users\Administrator\.m2\repository\commons-logging\commons-logging\1.1.3\commons-logging-1.1.3.jar;C:\Users\Administrator\.m2\repository\org\apache\httpcomponents\httpclient\4.5.2\httpclient-4.5.2.jar;C:\Users\Administrator\.m2\repository\org\apache\httpcomponents\httpcore\4.4.4\httpcore-4.4.4.jar;C:\Users\Administrator\.m2\repository\commons-codec\commons-codec\1.9\commons-codec-1.9.jar;C:\Users\Administrator\.m2\repository\software\amazon\ion\ion-java\1.0.2\ion-java-1.0.2.jar;C:\Users\Administrator\.m2\repository\com\fasterxml\jackson\core\jackson-databind\2.6.7.1\jackson-databind-2.6.7.1.jar;C:\Users\Administrator\.m2\repository\com\fasterxml\jackson\core\jackson-annotations\2.6.0\jackson-annotations-2.6.0.jar;C:\Users\Administrator\.m2\repository\com\fasterxml\jackson\core\jackson-core\2.6.7\jackson-core-2.6.7.jar;C:\Users\Administrator\.m2\repository\com\fasterxml\jackson\dataformat\jackson-dataformat-cbor\2.6.7\jackson-dataformat-cbor-2.6.7.jar;C:\Users\Administrator\.m2\repository\joda-time\joda-time\2.8.1\joda-time-2.8.1.jar;C:\Users\Administrator\.m2\repository\com\google\inject\guice\4.0\guice-4.0.jar;C:\Users\Administrator\.m2\repository\javax\inject\javax.inject\1\javax.inject-1.jar;C:\Users\Administrator\.m2\repository\aopalliance\aopalliance\1.0\aopalliance-1.0.jar;C:\Users\Administrator\.m2\repository\com\google\code\findbugs\jsr305\3.0.2\jsr305-3.0.2.jar;C:\Users\Administrator\.m2\repository\com\google\guava\guava\21.0\guava-21.0.jar;C:\Users\Administrator\.m2\repository\com\google\code\findbugs\annotations\2.0.3\annotations-2.0.3.jar;C:\Users\Administrator\.m2\repository\org\apache\commons\commons-lang3\3.4\commons-lang3-3.4.jar;C:\Users\Administrator\.m2\repository\com\google\code\gson\gson\2.8.2\gson-2.8.2.jar;C:\Users\Administrator\.m2\repository\javax\measure\jsr-275\0.9.1\jsr-275-0.9.1.jar;C:\Users\Administrator\.m2\repository\com\google\inject\extensions\guice-multibindings\4.0\guice-multibindings-4.0.jar;C:\Users\Administrator\.m2\repository\commons-io\commons-io\2.4\commons-io-2.4.jar;C:\Users\Administrator\.m2\repository\com\amazonaws\aws-java-sdk-kinesisvideo\1.11.255\aws-java-sdk-kinesisvideo-1.11.255.jar;C:\Users\Administrator\.m2\repository\io\netty\netty-codec-http\4.1.17.Final\netty-codec-http-4.1.17.Final.jar;C:\Users\Administrator\.m2\repository\io\netty\netty-codec\4.1.17.Final\netty-codec-4.1.17.Final.jar;C:\Users\Administrator\.m2\repository\io\netty\netty-handler\4.1.17.Final\netty-handler-4.1.17.Final.jar;C:\Users\Administrator\.m2\repository\io\netty\netty-buffer\4.1.17.Final\netty-buffer-4.1.17.Final.jar;C:\Users\Administrator\.m2\repository\io\netty\netty-common\4.1.17.Final\netty-common-4.1.17.Final.jar;C:\Users\Administrator\.m2\repository\io\netty\netty-transport\4.1.17.Final\netty-transport-4.1.17.Final.jar;C:\Users\Administrator\.m2\repository\io\netty\netty-resolver\4.1.17.Final\netty-resolver-4.1.17.Final.jar;C:\Users\Administrator\.m2\repository\com\amazonaws\jmespath-java\1.11.255\jmespath-java-1.11.255.jar" com.amazonaws.kinesisvideo.demoapp.DemoAppMain
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
INFO / KinesisVideo: Initializing Kinesis Video client
DEBUG / KinesisVideo: calling describe stream: {StreamName: face_test2,}
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: KinesisVideo producer stream creation time out StatusCode: 0xf
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:48)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: KinesisVideo producer stream creation time out StatusCode: 0xf
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerStream.awaitReady(NativeKinesisVideoProducerStream.java:469)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createStreamSync(NativeKinesisVideoProducerJni.java:355)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.registerMediaSource(NativeKinesisVideoClient.java:130)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:43)
ERROR / KinesisVideo: 2019-01-04T10:44Z T12: EXCEPTION: SdkClientException: Service call failed.: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/52.45.12.208] failed: connect timed out
ERROR / KinesisVideo: Kinesis Video service client returned an error com.amazonaws.SdkClientException: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/52.45.12.208] failed: connect timed out. Reporting to Kinesis Video PIC.
DEBUG / KinesisVideo: calling describe stream: {StreamName: face_test2,}
ERROR / KinesisVideo: 2019-01-04T10:44Z T12: EXCEPTION: SdkClientException: Service call failed.: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/52.45.12.208] failed: connect timed out
ERROR / KinesisVideo: Kinesis Video service client returned an error com.amazonaws.SdkClientException: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/52.45.12.208] failed: connect timed out. Reporting to Kinesis Video PIC.
DEBUG / KinesisVideo: calling describe stream: {StreamName: face_test2,}
ERROR / KinesisVideo: 2019-01-04T10:44Z T12: EXCEPTION: SdkClientException: Service call failed.: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/52.45.12.208] failed: connect timed out
ERROR / KinesisVideo: Kinesis Video service client returned an error com.amazonaws.SdkClientException: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east- [... cut ...]
ERROR / KinesisVideo: 2019-01-04T10:46Z T12: EXCEPTION: SdkClientException: Service call failed.: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/34.236.195.233] failed: connect timed out
ERROR / KinesisVideo: Kinesis Video service client returned an error com.amazonaws.SdkClientException: Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/34.236.195.233] failed: connect timed out. Reporting to Kinesis Video PIC.
`

zhiyua-git, 2019-03-01:
Hi @liuxthu ,

The log shows "Unable to execute HTTP request: Connect to kinesisvideo.us-east-1.amazonaws.com:443 [kinesisvideo.us-east-1.amazonaws.com/52.45.12.208] failed: connect timed out". Can you confirm that kinesis video endpoint is not blocked by your firewall or some connection issue?

Thanks,
-ZH

zhiyua-git, 2020-01-03:
Closing issue due to inactive. Please reopen if the issue appears on latest release.

closed 2020-01-03 by zhiyua-git

TechReporter, 2021-07-18:
UnsatisfiedLinkError: com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.getNativeLibraryVersion()Ljava/lang/String
I am following this workspace https://github.com/backdoorcodr/amazon-kinesis-video-streams-producer-sdk-java
Expectation is to run kinesis video stream using webcam in java.
Please help