awslabs/amazon-kinesis-video-streams-producer-sdk-java #55: Crash when lunching the app while already in session

avrum, 2019-02-05:
Hi, 
I'm keep getting crash exception when I'm launching the app, after closing it while I was already connected.
I managed to solve it by calling the "signout" every time I launch the app, but it's just a band aid and this issue should be fixed.

I'm running on Galaxy S9 Android 8.0.0

The complete Cat log prints:

> 02-05 08:41:16.959 18073-18108/com.amazon.myapp.demoapp D/KinesisVideoAndroidClient: Refreshing credentials
02-05 08:41:16.964 18073-18108/com.amazon.myapp.demoapp I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
    (HTTPLog)-Static: isSBSettingEnabled false
02-05 08:41:16.971 18073-18108/com.amazon.myapp.demoapp D/TcpOptimizer: TcpOptimizer-ON
02-05 08:41:18.102 18073-18073/com.amazon.myapp.demoapp E/KinesisVideoAndroidClient: 2019-02-05T08:41Z T2: EXCEPTION: ExecutionException: Awaiting for the credentials update threw an exception: com.amazonaws.services.cognitoidentity.model.NotAuthorizedException: Access to Identity 'us-east-1:cb8b84e8-3a24-4d40-a2c6-9468ed183ad1' is forbidden. (Service: AmazonCognitoIdentity; Status Code: 400; Error Code: NotAuthorizedException; Request ID: 0b14dc04-2911-11e9-98c3-8f23ade713f4)
02-05 08:41:18.117 18073-18073/com.amazon.myapp.demoapp A/zygote64: java_vm_ext.cc:504] JNI DETECTED ERROR IN APPLICATION: JNI GetObjectClass called with pending exception java.lang.RuntimeException: Certificate integration is not implemented
    java_vm_ext.cc:504]   at com.amazonaws.kinesisvideo.producer.AuthInfo com.amazonaws.kinesisvideo.auth.DefaultAuthCallbacks.getDeviceCertificate() (DefaultAuthCallbacks.java:84)
    java_vm_ext.cc:504]   at com.amazonaws.kinesisvideo.producer.AuthInfo com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.getDeviceCertificate() (NativeKinesisVideoProducerJni.java:604)
    java_vm_ext.cc:504]   at long com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createKinesisVideoClient(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoProducerJni.java:-2)
    java_vm_ext.cc:504]   at void com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.create(com.amazonaws.kinesisvideo.producer.DeviceInfo, java.lang.String) (NativeKinesisVideoProducerJni.java:251)
    java_vm_ext.cc:504]   at void com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(com.amazonaws.kinesisvideo.producer.DeviceInfo, java.lang.String) (NativeKinesisVideoProducerJni.java:263)
    java_vm_ext.cc:504]   at void com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoProducerJni.java:229)
    java_vm_ext.cc:504]   at com.amazonaws.kinesisvideo.internal.producer.KinesisVideoProducer com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initializeNewKinesisVideoProducer(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoClient.java:219)
    java_vm_ext.cc:504]   at void com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initialize(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoClient.java:136)
    java_vm_ext.cc:504]   at com.amazonaws.kinesisvideo.client.KinesisVideoClient com.amazonaws.mobileconnectors.kinesisvideo.client.KinesisVideoAndroidClientFactory.createKinesisVideoClient(android.content.Context, com.amazonaws.kinesisvideo.client.KinesisVideoClientConfiguration, com.amazonaws.kinesisvideo.producer.DeviceInfo, com.amazonaws.kinesisvideo.common.logging.Log, java.util.concurrent.ScheduledExecutorService) (KinesisVideoAndroidClientFactory.java:139)
    java_vm_ext.cc:504]   at com.amazonaws.kinesisvideo.client.KinesisVideoClient com.amazonaws.mobileconnectors.kinesisvideo.client.KinesisVideoAndroidClientFactory.createKinesisVideoClient(android.content.Context, com.amazonaws.regions.Regions, com.amazonaws.auth.AWSCredentialsProvider) (KinesisVideoAndroidClientFactory.java:114)
    java_vm_ext.cc:504]   at android.view.View com.amazon.myapp.demoapp.fragment.StreamConfigurationFragment.onCreateView(android.view.LayoutInflater, android.view.ViewGroup, android.os.Bundle) (StreamConfigurationFragment.java:76)
    java_vm_ext.cc:504]   at android.view.View android.support.v4.app.Fragment.performCreateView(android.view.LayoutInflater, android.view.ViewGroup, android.os.Bundle) (Fragment.java:2354)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.moveToState(android.support.v4.app.Fragment, int, int, int, boolean) (FragmentManager.java:1419)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(android.support.v4.app.Fragment) (FragmentManager.java:1740)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.moveToState(int, boolean) (FragmentManager.java:1809)
    java_vm_ext.cc:504]   at void android.support.v4.app.BackStackRecord.executeOps() (BackStackRecord.java:799)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.executeOps(java.util.ArrayList, java.util.ArrayList, int, int) (FragmentManager.java:2580)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.executeOpsTogether(java.util.ArrayList, java.util.ArrayList, int, int) (FragmentManager.java:2367)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(java.util.ArrayList, java.util.ArrayList) (FragmentManager.java:2322)
    java_vm_ext.cc:504]   at boolean android.support.v4.app.FragmentManagerImpl.execPendingActions() (FragmentManager.java:2229)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.dispatchStateChange(int) (FragmentManager.java:3221)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated() (FragmentManager.java:3171)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentController.dispatchActivityCreated() (FragmentController.java:192)
    java_vm_ext.cc:504]   at void android.support.v4.app.FragmentActivity.onStart() (FragmentActivity.java:560)
    java_vm_ext.cc:504]   at void android.support.v7.app.AppCompatActivity.onStart() (AppCompatActivity.java:177)
    java_vm_ext.cc:504]   at void android.app.Instrumentation.callActivityOnStart(android.app.Activity) (Instrumentation.java:1340)
    java_vm_ext.cc:504]   at void android.app.Activity.performStart() (Activity.java:7200)
    java_vm_ext.cc:504]   at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:2918)
    java_vm_ext.cc:504]   at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:3030)
    java_vm_ext.cc:504]   at void android.app.ActivityThread.-wrap11(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
    java_vm_ext.cc:504]   at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1696)
    java_vm_ext.cc:504]   at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:105)
    java_vm_ext.cc:504]   at void android.os.Looper.loop() (Looper.java:164)
    java_vm_ext.cc:504]   at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6938)
    java_vm_ext.cc:504]   at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
    java_vm_ext.cc:504]   at void com.android.internal.os.Zygote$MethodAndArgsCaller.run() (Zygote.java:327)
    java_vm_ext.cc:504]   at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:1374)
    java_vm_ext.cc:504] 
    java_vm_ext.cc:504]     in call to GetObjectClass
    java_vm_ext.cc:504]     from long com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createKinesisVideoClient(com.amazonaws.kinesisvideo.producer.DeviceInfo)
    java_vm_ext.cc:504] "main" prio=5 tid=1 Runnable
    java_vm_ext.cc:504]   | group="main" sCount=0 dsCount=0 flags=0 obj=0x7431afe0 self=0x7de60c5a00
    java_vm_ext.cc:504]   | sysTid=18073 nice=-10 cgrp=default sched=0/0 handle=0x7debdee9c8
    java_vm_ext.cc:504]   | state=R schedstat=( 639565272 48830812 609 ) utm=54 stm=9 core=6 HZ=100
[... cut ...]
    runtime.cc:524]   native: #60 pc 00000000005151f4  /system/lib64/libart.so (MterpInvokeVirtual+612)
    runtime.cc:524]   at android.os.Looper.loop(Looper.java:164)
    runtime.cc:524]   at android.app.ActivityThread.main(ActivityThread.java:6938)
    runtime.cc:524]   at java.lang.reflect.Method.invoke(Native method)
    runtime.cc:524]   at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
    runtime.cc:524] 
02-05 08:41:18.120 18073-18073/com.amazon.myapp.demoapp A/libc: Fatal signal 6 (SIGABRT), code -6 in tid 18073 (myapp.demoapp)

minbi, 2019-02-12:
Hi @avrum ,

We have pushed a fix to the sample to address this issue. Please do a `git pull --rebase` to see if it fixes the issue for you.

avrum, 2019-02-20:
@minbi Thanks I'll check it now

ZainabuGit, 2019-03-07:
I get the following error when i try to create the Vedio Client in Kenesis 

mKinesisVideoClient = KinesisVideoAndroidClientFactory.createKinesisVideoClient(
                    getActivity(),
                    KinesisVideoDemoApp.KINESIS_VIDEO_REGION,
                    KinesisVideoDemoApp.getCredentialsProvider());

E/KinesisVideoAndroidClient: 2019-03-07T14:03Z T2: EXCEPTION: TimeoutException: Awaiting for the credentials update threw an exception: null
A/zygote: java_vm_ext.cc:534] JNI DETECTED ERROR IN APPLICATION: JNI GetObjectClass called with pending exception java.lang.RuntimeException: Certificate integration is not implemented
A/zygote: java_vm_ext.cc:534]   at com.amazonaws.kinesisvideo.producer.AuthInfo com.amazonaws.kinesisvideo.auth.DefaultAuthCallbacks.getDeviceCertificate() (DefaultAuthCallbacks.java:84)
A/zygote: java_vm_ext.cc:534]   at com.amazonaws.kinesisvideo.producer.AuthInfo com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.getDeviceCertificate() (NativeKinesisVideoProducerJni.java:604)
A/zygote: java_vm_ext.cc:534]   at long com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createKinesisVideoClient(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoProducerJni.java:-2)
A/zygote: java_vm_ext.cc:534]   at void com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.create(com.amazonaws.kinesisvideo.producer.DeviceInfo, java.lang.String) (NativeKinesisVideoProducerJni.java:251)
A/zygote: java_vm_ext.cc:534]   at void com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(com.amazonaws.kinesisvideo.producer.DeviceInfo, java.lang.String) (NativeKinesisVideoProducerJni.java:263)
A/zygote: java_vm_ext.cc:534]   at void com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoProducerJni.java:229)
A/zygote: java_vm_ext.cc:534]   at com.amazonaws.kinesisvideo.internal.producer.KinesisVideoProducer com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initializeNewKinesisVideoProducer(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoClient.java:219)
A/zygote: java_vm_ext.cc:534]   at void com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initialize(com.amazonaws.kinesisvideo.producer.DeviceInfo) (NativeKinesisVideoClient.java:136)
A/zygote: java_vm_ext.cc:534]   at com.amazonaws.kinesisvideo.client.KinesisVideoClient com.amazonaws.mobileconnectors.kinesisvideo.client.KinesisVideoAndroidClientFactory.createKinesisVideoClient(android.content.Context, com.amazonaws.kinesisvideo.client.KinesisVideoClientConfiguration, com.amazonaws.kinesisvideo.producer.DeviceInfo, com.amazonaws.kinesisvideo.common.logging.Log, java.util.concurrent.ScheduledExecutorService) (KinesisVideoAndroidClientFactory.java:139)
A/zygote: java_vm_ext.cc:534]   at com.amazonaws.kinesisvideo.client.KinesisVideoClient com.amazonaws.mobileconnectors.kinesisvideo.client.KinesisVideoAndroidClientFactory.createKinesisVideoClient(android.content.Context, com.amazonaws.regions.Regions, com.amazonaws.auth.AWSCredentialsProvider) (KinesisVideoAndroidClientFactory.java:114)
A/zygote: java_vm_ext.cc:534]   at android.view.View com.amazonaws.kinesisvideo.demoapp.fragment.StreamConfigurationFragment.onCreateView(android.view.LayoutInflater, android.view.ViewGroup, android.os.Bundle) (StreamConfigurationFragment.java:70)

passionpatel, 2019-05-06:
Receiving the same error: JNI DETECTED ERROR IN APPLICATION: JNI GetObjectClass called with pending exception java.lang.RuntimeException: Certificate integration is not implemented.  Is fix implemented & available?

MushMal, 2019-05-06:
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-android.html highlights the steps configuring the Cognito. Once this is properly done, you can re-install the application or clear the application data and re-try.

vipulraut, 2019-06-27:
Facing the same issue,

tongvantruong, 2019-06-28:
Same issue. It worked yesterday. Today, the app just crashes

zhiyua-git, 2019-06-28:
Hi @vipulraut , @tongvantruong , can you +1 on this github issue? https://github.com/aws-amplify/aws-sdk-android/issues/956 The app is actually crashes in AWSMobileClient. 

Thanks.

zhiyua-git, 2019-06-28:
As a workaround to unblock you now, you can fix it by adding _auth.signOut();_ in [this line](https://github.com/awslabs/aws-sdk-android-samples/blob/master/AmazonKinesisVideoDemoApp/src/main/java/com/amazonaws/kinesisvideo/demoapp/activity/StartUpActivity.java#L22).

But you will need to sign in every time you start the app after this.

Hope this helps,
-ZH

tongvantruong, 2019-06-28:
I resolved my issue by changing the internet connection. because the internet speed is slow. I am not sure if it can help any others

faisalvv, 2020-01-15:
work for me.
change some code from StartupActivity.java

AsyncTask.execute(new Runnable() {
            @Override
            public void run() {
                if (auth.isSignedIn()) {
                    auth.getTokens(new Callback<Tokens>() {
                        @Override
                        public void onResult(Tokens result) {
                            ActivityUtils.startActivity(thisActivity, SimpleNavActivity.class);
                        }

                        @Override
                        public void onError(Exception e) {

                        }
                    });

                } else {
                    auth.showSignIn(thisActivity,
                            SignInUIOptions.builder()
                                    .nextActivity(SimpleNavActivity.class)
                                    .build(),
                            new Callback<UserStateDetails>() {
                                @Override
                                public void onResult(UserStateDetails result) {
                                    Log.d(TAG, "onResult: User signed-in " + result.getUserState());
                                }

                                @Override
                                public void onError(final Exception e) {
                                    runOnUiThread(new Runnable() {
                                        @Override
                                        public void run() {
                                            Log.e(TAG, "onError: User sign-in error", e);
                                            Toast.makeText(StartUpActivity.this, "User sign-in error: " + e.getMessage(), Toast.LENGTH_LONG).show();
                                        }
                                    });
                                }
                            });
                }
            }
        });

zhiyua-git, 2020-02-19:
Pull Request is ready for review: https://github.com/awslabs/aws-sdk-android-samples/pull/332

Thanks @faisalvv for the suggestion.

disa6302, 2020-04-01:
Closing this since it is fixed.

closed 2020-04-01 by disa6302

Ritesh-Kumar-4946, 2020-05-21:
Hello I am facing problem with Kinesis live streaming it working in debug version in android but when i generate released/debug .aar file and try to use it in some other project it is not working

Ritesh-Kumar-4946, 2020-05-21:
My session is not created in .aar build but its working in debug apk

seesiva, 2020-11-06:
Any further update on this issue, there is a significant delay happening due to the work around of sign-out.

seesiva, 2021-02-15:
Trying to pass the credential provider by using AWSMobileClient.getInstance() but getting null value exception whule passing but we are logged in
mKinesisVideoClient = KinesisVideoAndroidClientFactory.createKinesisVideoClient(
        requireContext(),
        KINESIS_VIDEO_REGION,
        AWSMobileClient.getInstance())

sirknightj, 2023-06-23:
You can try this:

```java
final CountDownLatch latch = new CountDownLatch(1);
new Thread(() -> {
    try {
        ((AWSMobileClient) KinesisVideoDemoApp.getCredentialsProvider()).getAWSCredentials();
    } catch (final Exception e) {
        Log.e(TAG, "Exception while fetching credentials", e);
    } finally {
        latch.countDown();
    }
}).start();

try {
    latch.await();
    mKinesisVideoClient = KinesisVideoAndroidClientFactory.createKinesisVideoClient(
            getActivity(),
            KinesisVideoDemoApp.KINESIS_VIDEO_REGION,
            KinesisVideoDemoApp.getCredentialsProvider());
} catch (final InterruptedException | KinesisVideoException e) {
    Log.e(TAG, "Failed to create Kinesis Video client", e);
}
```

kevin4dhd, 2023-06-23:
> You can try this:
>
> ```java
> final CountDownLatch latch = new CountDownLatch(1);
> new Thread(() -> {
>     try {
>         ((AWSMobileClient) [...]

That in KinesisVideoDemoApp.java or where?

sirknightj, 2023-06-24:
@kevin4dhd 

https://github.com/awslabs/aws-sdk-android-samples/blob/c43945662fe15669bd8ed3ae0afc96ece3d83170/AmazonKinesisVideoDemoApp/src/main/java/com/amazonaws/kinesisvideo/demoapp/fragment/StreamConfigurationFragment.java#L76

sirknightj, 2023-06-28:
https://github.com/aws-amplify/aws-sdk-android/issues/873