awslabs/amazon-kinesis-video-streams-producer-sdk-java #59: The requested stream is not found or not active. (Service: AmazonKinesisVideo; Status Code: 404; Error Code: ResourceNotFoundException;

zebpaypeswani, 2019-02-25:
Hello I am getting above error at below line when using putmedia demo app

```
   private static final String STREAM_NAME = "FaceMatching";
   final String dataEndpoint = frontendClient.getDataEndpoint(
                new GetDataEndpointRequest()
                        .withStreamName(STREAM_NAME)
                        .withAPIName("PUT_MEDIA")).getDataEndpoint();
```

I created the stream and it is active when I run the below code

```
 public class Rekoginition {
    public static void main(String[] args) {

        String streamProcessorName="FaceMatching";
        String kinesisVideoStreamArn="arn:aws:kinesisvideo:us-west-2:655136467581:stream/FaceMatching/1550831945886";
        String kinesisDataStreamArn="arn:aws:kinesis:us-west-2:655136467581:stream/FaceStream";
        String roleArn="arn:aws:iam::655136467581:role/face_matching";
        String collectionId="FaceCollection";
        Float matchThreshold=50F;

        try {
            StreamManager sm= new StreamManager(streamProcessorName,
                    kinesisVideoStreamArn,
                    kinesisDataStreamArn,
                    roleArn,
                    collectionId,
                    matchThreshold);
//            sm.createStreamProcessor();
            sm.startStreamProcessor();
//            sm.deleteStreamProcessor();
//            sm.deleteStreamProcessor();
//            sm.stopStreamProcessor();
            sm.listStreamProcessors();
            sm.describeStreamProcessor();
        }
        catch(Exception e){
            System.out.println(e.getMessage());
        }
    }
}
```

however not working as expected.

closed 2019-02-25 by zebpaypeswani

addodelgrossi, 2019-07-22:
@zebpaypeswani I encountered the same problem, how did you solve it?

zebpaypeswani, 2019-07-22:
@addodelgrossi  I used python lib.

biswajit123, 2020-04-21:
@zebpaypeswani Which Python Lib did you use?

tejuri, 2021-06-08:
Getting the same error. Trying to putmediademo example. It giving that the kvs is not found or not active.
I am running a docker container inside a windows machine.  The stream is active, i can see in the console

[image]