awslabs/amazon-kinesis-video-streams-producer-sdk-java #62: Green color in Kinesis live video streaming

kirilldev1010, 2019-05-18:
images.githubusercontent.com/44777334/57966949-c0179580-7926-11e9-936a-0d9de889edcf.png)

How can I resolve this issue?

MushMal, 2019-05-18:
@kirilldev1010 encoders use the "green screen" to indicate that there is a decoding problem. In your case it seems that some configuration is wrong between the encoder and decoder. Assuming the screen shot had been taken from the AWS Console playback then it's likely that your browsers MSE was not able to properly decode the H264 (assuming it's H264) stream. This generally means that while the stream itself is valid, the content of the stream is not semantically valid.

Recommend ensuring that the stream data is valid and the supplied CPD (Codec Private Data which is SPS/PPS for H264) is valid and corresponds to the encoder settings producing the video stream.

Please provide more details if this issue persists -
* Your end-to-end scenario
* Device that's producing the stream
* Settings for the encoder
* Stream settings
* Logs
* Any other information that can help to identify the problem.

kirilldev1010, 2019-05-18:
Thank for reply, MushMal. 
Yes, the screenshot is website in preview of Amazon Console page. 
How can I confirm easily that my stream is working?

MushMal, 2019-05-18:
Please provide more details if this issue persists -

Your end-to-end scenario
Device that's producing the stream
Settings for the encoder
Stream settings
Logs
Any other information that can help to identify the problem.

MushMal, 2019-05-18:
Knowing the media source, perhaps we can confirm that it produces a valid H264 video without streaming into KVS all together first of all. If it does, and if we have proper CPD and the stream is not interrupted (Please attach the logs) then it's possible that your browser doesn't properly support H264 level/profile that you are streaming.

kirilldev1010, 2019-05-18:
I am going to stream from Android UVC via Amazon Kinesis. 
I have implemented the source from this link. 
https://github.com/awslabs/aws-sdk-android-samples/tree/master/AmazonKinesisVideoDemoApp

kirilldev1010, 2019-05-18:
AndroidCameraMediaSourceConfiguration.builder()
                        .withCameraId(mCamerasDropdown.getSelectedItem().getCameraId())
                        .withEncodingMimeType(mMimeTypeDropdown.getSelectedItem().getMimeType())
                        .withHorizontalResolution(mResolutionDropdown.getSelectedItem().getWidth())
                        .withVerticalResolution(mResolutionDropdown.getSelectedItem().getHeight())
                        .withCameraFacing(mCamerasDropdown.getSelectedItem().getCameraFacing())
                        .withIsEncoderHardwareAccelerated(
                                mCamerasDropdown.getSelectedItem().isEndcoderHardwareAccelerated())
                        .withFrameRate(FRAMERATE_20)
                        .withRetentionPeriodInHours(RETENTION_PERIOD_48_HOURS)
                        .withEncodingBitRate(BITRATE_384_KBPS)
                        .withCameraOrientation(-mCamerasDropdown.getSelectedItem().getCameraOrientation())
                        .withNalAdaptationFlags(StreamInfo.NalAdaptationFlags.NAL_ADAPTATION_ANNEXB_CPD_AND_FRAME_NALS)
                        .withIsAbsoluteTimecode(false));

MushMal, 2019-05-18:
Seems to be OK. Could you please include the device make and model? If you are using the demo application, could you ensure that the encoder drop down is set to H264?

Also, do you see the green screen when you are streaming different resolutions?

kirilldev1010, 2019-05-18:
Yes, when I am streaming different resolutions, it is still same. 
I am new in this field. 
Please let me know how I can ensure that it is set to H264. 
I confirmed Amazon Console says it is H264 video/avc in admin panel's preview

MushMal, 2019-05-18:
Thanks. This is rather strange that the console parses the CPD and the frames OK. It seems that the stream correctly sends the data.

Could I ask you to do the following:

* In the demo application, there should be a drop down listing the codecs. Could you ensure H264 is selected.
* Could you try another browser?
* Please include the make and the model of the device - perhaps it has an incompatible encoder (highly doubtful).

kirilldev1010, 2019-05-18:
Yes, I am sure it is H264. 
I didn't try another browser because I don't know which can resolve this issue. 
Device is Umidigi One Max.

MushMal, 2019-05-18:
Looking at the specs it seems that it does support a hardware accelerated H264 encoder: https://www.umidigi.com/page-umidigi_one_max_specification.html

Would you please try to use another browser perhaps?

kirilldev1010, 2019-05-18:
Just checked in Safari, but same issue. 
Seems like my device doesn't support H264 encoder as your words

MushMal, 2019-05-18:
I am at odds really. Do you have another Android device to try? I have never encountered this device before. I will try to dig into it and reply as soon as I have more info

kirilldev1010, 2019-05-18:
Yes, I have got another. 
Have you developed an app for Android USB Camera?

kirilldev1010, 2019-05-18:
I found this in Umidigi One Max specification. 
Supports H.264 (Baseline / Main / High profile), MPEG4 (Simple profile / ASP) and other video formats,
Supports audio formats such as PCM, AAC / AAC + / eAAC +, MP3, AMR - NB and WB, APE, WAV

MushMal, 2019-05-21:
Hi Kirill,

I have not been able to find more information about the particular device. It seems that the encoder uses a different color space configuration than the ones the SDK uses by default:

https://github.com/aws-amplify/aws-sdk-android/blob/master/aws-android-sdk-kinesisvideo/src/main/java/com/amazonaws/mobileconnectors/kinesisvideo/encoding/EncoderFactory.java#L51

Unfortunately, I don't have the device so here is what we can suggest:

1. git clone https://github.com/aws-amplify/aws-sdk-android
2. Open build.gradle in aws-sdk-android as project in AndroidStudio
3. Modify line https://github.com/aws-amplify/aws-sdk-android/blob/master/aws-android-sdk-kinesisvideo/src/main/java/com/amazonaws/mobileconnectors/kinesisvideo/encoding/EncoderFactory.java#L51 to COLOR_FormatYUV420Flexible (API 21)
4. Select aws-sdk-android-kinesisvideo in the Android section of project view, then Build -> Make Module 'aws-sdk-android-kinesisvideo'
5. git clone https://github.com/awslabs/aws-sdk-android-samples
6. cp aws-sdk-android/aws-android-sdk-kinesisvideo/build/outputs/aar/aws-android-sdk-kinesisvideo-debug.aar aws-sdk-android-samples/AmazonKinesisVideoDemoApp/libs/
7. Open build.gradle in aws-sdk-android-samples/AmazonKinesisVideoDemoApp/ as project in AndroidStudio
8. Modify line 54 from "implementation ("com.amazonaws:aws-android-sdk-kinesisvideo:$aws_version@aar") { transitive = true }" to "implementation fileTree(dir: 'libs', include: ['*.jar', '*.aar'])"
9. "Gradle files have changed ..." would show up, click Sync Now and then Build -> Make Project.

Sorry for the lengthy instruction set but at this stage this might be the shortest path.

kirilldev1010, 2019-05-21:
Thank you MushMal, 

I have updated the code as your explanation, and got this. 

https://www.dropbox.com/s/fc2f10f0q6p430v/Screen%20Shot%202019-05-21%20at%2012.06.25%20PM.png?dl=0

MushMal, 2019-05-21:
Not sure what's going on. It seems that the stride is broken. Kirill, could you provide the following:

* Have you tried with another Android device? If so, which one?
* Could you please send your modifications to the code to kinesis-video-support@amazon.com

kirilldev1010, 2019-05-21:
OK, Let me send it to the email. 
Just followed Amazon documentation and your instructions. 

Please let me know byte data streaming if you have any experience. 

Thank you

disa6302, 2020-04-01:
Closing this for now. Feel free to re-open in the future

closed 2020-04-01 by disa6302

makelitan, 2020-04-16:
Hello everyone.

The target hardware is an iMX8M mini development kit with a camera module ([link](https://www.nxp.com/part/MINISASTOCSI#/)) running on Android 10.

The software is https://github.com/awslabs/aws-sdk-android-samples/tree/master/AmazonKinesisVideoDemoApp.

AWS Media playback shows a green screen when streaming with the kit.

Streaming with the Android API 23 smartphone shows correctly.

Is there anything that I can do to stream with the development kit?

Best regards,
Mihai Gîsca

MushMal, 2020-04-16:
@makelitan the green screen issues are very hard to debug as they are dependent on the devices you have and the browser/platform you are using. The KVS end-to-end seems to work fine in this case in order to get the frames down to your browser (Producer SDK -> backend -> console playback -> your browser). The issue is to ensure the H264 that's produced by the device has the same capabilities of your browser, including color space, H264 profiles, resolutions, etc..

makelitan, 2020-04-16:
Hello @MushMal,

Thank you for your feedback.
I have no knowledge about any of these. The green screen is provided by both Chrome and Microsoft Edge.

Are there any common issues that I can try to check and adjust in order to try to get this working?

MushMal, 2020-04-16:
@makelitan I am not familiar with this hardware so I can't tell how you can configure. However, there is some information attached above for the previous customer to try to set the color plane. Also, just in case, try to run the demo app on an Android phone and see if you can get proper playback in your browser.

NOTE: The browser is using underlying platforms hardware acceleration capabilities most of the time to decode and render video.

makelitan, 2020-04-17:
Good day,

I am struggling with following above steps on Windows 10 OS Android 3.6.2

Build fails at step 4 with the message:
...
:aws-android-sdk-core:getApi23 FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':aws-android-sdk-core:getApi23'.
> A problem occurred starting process 'command 'sh''

Any ideas?

Best regards,
Mihai Gîsca

MushMal, 2020-04-17:
@makelitan apologies for this whole ordeal. The steps were put together as a quick fix and if I guess, it seems that there is a custom step in gradle command for building Android that kicks off a shell and uses sh which is not Windows specific. You can attempt to fix the gradle script by making it Windows specific or try to use a Linux machine.

We are working on making the source-code distribution of the core resources available in which will make these steps much easier but until then it's a little bit of a struggle.

makelitan, 2020-04-17:
@MushMal I see. I guess it is the

task getApi23(type:Exec) {
    print 'Creating a symlink for Android API Level 23.'
    workingDir '.'
    commandLine 'sh', '-c', "rm -rf libs && mkdir libs && ln -s $System.env.ANDROID_HOME/platforms/android-23/android.jar libs/android-23.jar"
}

piece of code in build.gradle that raises this error.

I will see what I can do, thank you for sharing. If there is any code for Windows cmd I could use, please, provide it here.

Best regards,
Mihai Gîsca

makelitan, 2020-04-17:
Could not make it work.

Changed the build.gradle as follows:

task getApi23(type:Exec) {
    print 'Creating a symlink for Android API Level 23.'
    workingDir '.'
    commandLine 'cmd', "rmdir \"libs\" /s /q 2>nul && mkdir \"libs\" && mklink /d \"C:\\Android\\android-sdk\\platforms\\android-23\\android.jar\" \"libs\\android-23.jar\""
}

Stopped at this errors:

warning: [options] bootstrap class path not set in conjunction with -source 1.7
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider18.java:20: error: package android.security does not exist
import android.security.KeyPairGeneratorSpec;
                       ^
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider23.java:18: error: package android.security.keystore does not exist
import android.security.keystore.KeyGenParameterSpec;
                                ^
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider23.java:19: error: package android.security.keystore does not exist
import android.security.keystore.KeyProperties;
                                ^
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider18.java:142: error: cannot find symbol
            KeyPairGeneratorSpec spec = new KeyPairGeneratorSpec.Builder(context)
            ^
  symbol:   class KeyPairGeneratorSpec
  location: class KeyProvider18
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider18.java:142: error: package KeyPairGeneratorSpec does not exist
            KeyPairGeneratorSpec spec = new KeyPairGeneratorSpec.Builder(context)
                                                                ^
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider23.java:94: error: cannot find symbol
                                .setEncryptionPaddings(KeyProperties.ENCRYPTION_PADDING_NONE)
                                                       ^
  symbol:   variable KeyProperties
  location: class KeyProvider23
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider23.java:93: error: cannot find symbol
                                .setBlockModes(KeyProperties.BLOCK_MODE_GCM)
                                               ^
  symbol:   variable KeyProperties
  location: class KeyProvider23
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider23.java:91: error: package KeyGenParameterSpec does not exist
                        new KeyGenParameterSpec.Builder(keyAlias,
                                               ^
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider23.java:92: error: cannot find symbol
                                KeyProperties.PURPOSE_ENCRYPT | KeyProperties.PURPOSE_DECRYPT)
                                ^
  symbol:   variable KeyProperties
  location: class KeyProvider23
C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\KeyProvider23.java:92: error: cannot find symbol
                                KeyProperties.PURPOSE_ENCRYPT | KeyProperties.PURPOSE_DECRYPT)
                                                                ^
  symbol:   variable KeyProperties
  location: class KeyProvider23
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\aws-sdk-android\aws-android-sdk-core\src\main\java\com\amazonaws\internal\keyvaluestore\AWSKeyValueStore.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
10 errors
1 warning
C:\aws-sdk-android\aws-android-sdk-core>:aws-android-sdk-core:compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':aws-android-sdk-core:compileJava'.
> Compilation failed; see the compiler error output for details.

Everything I do is new to me. Any info on this topic are appreciated.

Best regards,
Mihai Gîsca

tendynewbies, 2020-05-27:
I am also facing the same issue on my devices.
320*240 - h264 works fine
Any other resolution - h264 gives issue on both of my Android devices.

Any resolution?