awslabs/amazon-kinesis-video-streams-producer-sdk-java #65: Fragment Decoding Errror in live feed.

Shvm, 2019-07-08:
Hello
I am trying to do live stream from laptop camera using Java.
With a help from issue#10 (Link), It seemes to be successful at sending frame to kinesis video stream. Here is what I get from console.

[image]

Please can someone help me or point me in right direction to solve this.

MushMal, 2019-07-08:
Decoding errors usually are an indicator of an invalid h264 stream. It seems that logically your stream is being parsed at the container level and persisted. However, the h264 decoding fails on the browser during the playback. Could you ensure your stream source is valid h264 stream and the key-frames are appropriately set?

Shvm, 2019-07-09:
@MushMal The logs doesn't say much about what is expected and what is given. is there any way I can get that info?

MushMal, 2019-07-09:
@Shvm what I was trying to say is that while you can stream the data OK, it might not be semantically valid h264. You might want to check the media source on the device - whether it produces valid h264. 

Can you describe what you are trying to do? Please provide some details on the device/SDK you are using.

Shvm, 2019-07-10:
@MushMal I want to do facial recognition using Amazon Rekognition, for that what I am trying to achieve is feed my webcam streaming to Kinesis video stream for capturing the video from webcam using java  I am using this https://github.com/sarxos/webcam-capture and I am encoding the frame using jcodec H264.

This is the pull request https://github.com/Shvm/amazon-kinesis-video-streams-producer-sdk-java/pull/1/files inspired from [issue#10](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/10).

Since dependencies like xuggler and geotoolkit are not supported any more. I have added external jar for xuggler, and remove the code dependency of geotoolkit.

MushMal, 2019-07-11:
Hi there,

Sorry for the delay. I am not super familiar with jCodec codebase. Here are some suggestions:

* Are you specifying the correct CPD (Codec Private Data) that's returned from the encoder itself and not the one that's provided as a sample.
* Are you specifying the proper encoder flags for an I-frame vs P-frames?
* You can try to specify different NAL adaptation flags to see if your configuration already produces AvCC stream.
* Try to update to the latest SDK which has auto-CPD extraction feature. In this case, you don't need to specify the CPD - it will be extracted from the elementary stream iff the stream is H264, it's in Annex-B format and adaptation flags specify Annex-B -> AvCC adaptation of the frames.

disa6302, 2020-04-01:
Closing this for now. Feel free to open a new issue in the future.

closed 2020-04-01 by disa6302