awslabs/amazon-kinesis-video-streams-producer-sdk-java #68: C++ release 2.0.1 causes erroneous retention period mismatch warnings

bkempe, 2019-08-14:
[JavaKinesisVideoServiceClient#toStreamDescription](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/8d7faf74297e9bb5b5809b792b1656ee66d6998a/src/main/java/com/amazonaws/kinesisvideo/java/service/JavaKinesisVideoServiceClient.java#L454) does not use the `StreamDescription` constructor with the `retention` parameter, therefore [the new validation check in release 2.0.1 of the Amazon Kinesis Video Streams Producer SDK C/C++](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/releases) fails, leading to:

```Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo```
logs.

zhiyua-git, 2019-08-14:
Hi @bkempe ,

Thanks for notifying us. We will fix that in next java release. Also pull-requests are welcome.

Thanks,
-ZH

zhiyua-git, 2020-02-19:
This is fixed in Release 1.9.5. Closing the issue.

closed 2020-02-19 by zhiyua-git