awslabs/amazon-kinesis-video-streams-producer-sdk-java #70: how to connect RTSP url in Kinesis video stream

hameedriswan7, 2019-08-23:
// CHECKSTYLE:SUPPRESS:LineLength
    // Need to get key frame configured properly so the output can be decoded. h264 files can be decoded using gstreamer plugin
    // gst-launch-1.0 rtspsrc location="YourRtspUri" short-header=TRUE protocols=tcp ! rtph264depay ! decodebin ! videorate ! videoscale ! vtenc_h264_hw allow-frame-reordering=FALSE max-keyframe-interval=25 bitrate=1024 realtime=TRUE ! video/x-h264,stream-format=avc,alignment=au,profile=baseline,width=640,height=480,framerate=1/25 ! multifilesink location=./frame-%03d.h264 index=1
   //private static final String IMAGE_FILENAME_FORMAT = "frame-%03d.h264";.
How to Implement correctly 

private static final String STREAM_NAME = "my-trail";
    private static final int FPS_25 = 25;
    private static final String IMAGE_DIR = "src/main/resources/data/h264/";
    private static final String IMAGE_FILENAME_FORMAT = "frame-%03d.h264";
    private static final int START_FILE_INDEX = 1;
    private static final int END_FILE_INDEX = 444;
    private final static String rtspUri="rtsp://192.168.1.194:554";
    private static final Duration RETENTION_PERIOD = Duration.ofHours(1);

  final MediaSource bytesMediaSource =  new GStreamerRtspMediaSource(
                    new URI(rtspUri),
                    RETENTION_PERIOD,
                    new Tag[] { new Tag("Produced-By", "Kinesis-Video-GStreamer-Demo") });

zhiyua-git, 2019-08-23:
With gst1-java-core 1.0.0, it seems Gst.parseBinFromDescription() with the gst-launch-1.0 pipeline  would not work for RTSP, you will need to create source, filter, depay, sink and connect source to depay and pad with Element.PAD_ADDED. 

There is no public sample available now but it is possible to use gst-java-core to send from RTSP url to AWS KVS.

unicornss, 2019-11-06:
Closing the issue; please reopen or create a new one if needed.

closed 2020-04-19 by MushMal