awslabs/amazon-kinesis-video-streams-producer-sdk-java #71: EXCEPTION: UnsatisfiedLinkError

AceKillers, 2019-11-04:
ERROR / KinesisVideo: 2019-11-04T16:24Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: [<src\main\resources\lib\windows>]: no KinesisVideoProducerJNI in java.library.path: [<src\main\resources\lib\windows>]

MushMal, 2019-11-04:
@AceKillers we are aware that the Java native library loader can print out an error when it re-tries various methods for loading the native library. In reality this is a debug message and the wording will be fixed in the upcoming release. Could you please include more info whether it finally succeeded loading your library?

AceKillers, 2019-11-04:
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:59)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.initializeLibrary(NativeKinesisVideoProducerJni.java:1216)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:229)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(NativeKinesisVideoProducerJni.java:247)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(NativeKinesisVideoProducerJni.java:213)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initializeNewKinesisVideoProducer(NativeKinesisVideoClient.java:223)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initialize(NativeKinesisVideoClient.java:119)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:107)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:80)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:42)
@MushMal

MushMal, 2019-11-04:
This is indeed the case where the native library loader couldn't find and load the native library.

Are you building the native library? Some of the steps are highlighted here

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/README.md#launching-demoapp-sample-application

AceKillers, 2019-11-04:
I used Intellij IDEA to configure the JVM parameters step by step, and this error occurred at runtime. The library path is `-Djava.library.path=<src\main\resources\lib\windows>` . Is there a problem with the parameter configuration?

AceKillers, 2019-11-04:
I changed the parameters a little, and there was a new problem
`Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: KinesisVideo producer stream creation time out StatusCode: 0xf
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:59)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: KinesisVideo producer stream creation time out StatusCode: 0xf
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerStream.awaitReady(NativeKinesisVideoProducerStream.java:501)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createStreamSync(NativeKinesisVideoProducerJni.java:356)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.registerMediaSource(NativeKinesisVideoClient.java:131)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:54)`

MushMal, 2019-11-04:
Seems you are able to load the native library now. Could you please make sure you have the credentials properly setup per ReadMe?

Could you please include any output from the native code?

AceKillers, 2019-11-04:
`DEBUG / KinesisVideo: calling describe stream: {StreamName: my-stream,}
DEBUG / KinesisVideo: describe stream result: {StreamInfo: {DeviceName: java-demo-application,StreamName: my-stream,StreamARN: arn:aws:kinesisvideo:us-east-1:887378151685:stream/my-stream/1560529859858,MediaType: video/h264,KmsKeyId: arn:aws:kms:us-east-1:887378151685:alias/aws/kinesisvideo,Version: eymbKHm0TOZP4iGByfEG,Status: ACTIVE,CreationTime: Sat Jun 15 00:30:59 CST 2019,DataRetentionInHours: 1}}
DEBUG / KinesisVideo: calling tag resource: {StreamARN: arn:aws:kinesisvideo:us-east-1:887378151685:stream/my-stream/1560529859858,Tags: {device=Test Device, stream=Test Stream}}
DEBUG / KinesisVideo: Stream my-stream is closed
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: KinesisVideo producer stream creation time out StatusCode: 0xf
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:59)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: KinesisVideo producer stream creation time out StatusCode: 0xf
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerStream.awaitReady(NativeKinesisVideoProducerStream.java:501)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createStreamSync(NativeKinesisVideoProducerJni.java:356)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.registerMediaSource(NativeKinesisVideoClient.java:131)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:54)
DEBUG / KinesisVideo: tag resource result: {}`

AceKillers, 2019-11-04:
`        try {
            // create Kinesis Video high level client
            final KinesisVideoClient kinesisVideoClient = KinesisVideoJavaClientFactory
                    .createKinesisVideoClient(
                            Regions.US_EAST_1,
                            AuthHelper.getSystemPropertiesCredentialsProvider());

            // create a media source. this class produces the data and pushes it into
            // Kinesis Video Producer lower level components
            final MediaSource mediaSource = createImageFileMediaSource();

            // Audio/Video sample is available for playback on HLS (Http Live Streaming)
            //final MediaSource mediaSource = createFileMediaSource();

            // register media source with Kinesis Video Client
            kinesisVideoClient.registerMediaSource(mediaSource);

            // start streaming
            mediaSource.start();
        } catch (final KinesisVideoException e) {
            throw new RuntimeException(e);
        }`

AceKillers, 2019-11-04:
`    public static void main(final String[] args) {
        try {
            // create Kinesis Video high level client
            final KinesisVideoClient kinesisVideoClient = KinesisVideoJavaClientFactory
                    .createKinesisVideoClient(
                            Regions.US_EAST_1,
                            AuthHelper.getSystemPropertiesCredentialsProvider());

            // create a media source. this class produces the data and pushes it into
            // Kinesis Video Producer lower level components
            final MediaSource mediaSource = createImageFileMediaSource();

            // Audio/Video sample is available for playback on HLS (Http Live Streaming)
            //final MediaSource mediaSource = createFileMediaSource();

            // register media source with Kinesis Video Client
            kinesisVideoClient.registerMediaSource(mediaSource);

            // start streaming
            mediaSource.start();
        } catch (final KinesisVideoException e) {
            throw new RuntimeException(e);
        }
    }`

zhiyua-git, 2019-11-04:
Hi @AceKillers ,

The default timeout is 15 seconds for create stream. If US_EAST_1 is not your nearby region, you can change it to a different region to get lower latency. 
If you want to increase the timeout, you can modify this [line](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/885886630533261080dea80af4a221d91fd94a89/src/main/java/com/amazonaws/kinesisvideo/internal/producer/KinesisVideoProducerStream.java#L21).

unicornss, 2019-11-05:
Hi @AceKillers 

Were you able to try with the suggestions from @zhiyua-git ?

unicornss, 2019-11-06:
Closing the issue as the question has been answered. Please reopen or create new one if needed.

closed 2020-01-09 by zhiyua-git