awslabs/amazon-kinesis-video-streams-producer-sdk-java #72: KinesisVideo producer stream stopping time out StatusCode: 0xf

changlinli, 2019-12-03:
I'm non-deterministically and occasionally (on the order of once maybe every 10-20 times) getting the following error when trying to stop a producer with `unregisterMediaSource`.

```
ProducerException: Stopping stream threw an exception. Force stopping the input stream.: KinesisVideo producer stream stopping time out StatusCode: 0xf
```

Version Used: 1.9.4

Is this expected behavior (i.e. just general expected flakiness on the part of KVS)?

MushMal, 2019-12-03:
This comes from the native PIC:

#define STATUS_OPERATION_TIMED_OUT                  STATUS_BASE + 0x0000000f

When an unregisterMediaSource is called stopStreamSync command is issued to attempt to finish up streaming out the accumulated buffer and get the last ACK from the backend before terminating. In your case it seems that it can't do that within the default timeout.

Check your network throughput and ensure you have adequate bandwidth. Resolving for now

closed 2019-12-03 by MushMal

changlinli, 2019-12-03:
Ah good to know, thanks! Is there any way to increase the timeout?

zhiyua-git, 2019-12-03:
Hi @changlinli ,

You can modify this https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/internal/service/AckConsumer.java#L19 to bump it up.