awslabs/amazon-kinesis-video-streams-producer-sdk-java #75: Retrieving Kinesis video in chunks (e.g. 1hr )

ahmedha, 2020-01-08:
Hello,

I am able to retrieve the video from the Kinesis video stream, staring from the latest.  I used the code in the file - **KinesisVideoExample.java** in the SDK as a reference.  

Here is what I would like to do.  I have a thread that wakes up on the hour (start of hour), and performs the following:
1. Thread wakes up, say at time T.
2. Read an hour worth of video stream from Kinesis.
3. Store the video into a file.  Save the last "frame read" (some sort of offset or opaque marker which can be passed to GetMeadi() in a later call).   (Similar to record offset/commit offset in partition in Kafka).
4. Thread goes back to sleep.
5. Thread wakes up T + 1 hr  (roughly).
6. Starts reading video frame from the saved offset in stream.
7. Repeat at step 3.

I have reviewed the following open issue for guidance:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/47

Is there any way I can call GetMedia providing it a staring offset, and possibly a time unit amount of video to read?  And get an offset/market/opaque object as to next video segment to start reading from next?

Thank you,
Ahmed.

MushMal, 2020-04-19:
GetMedia API https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_GetMedia.html

has various selectors and modes. One of them is the last retrieved fragment number. Perhaps this is what you can submit as a checkpoint.

MushMal, 2020-05-20:
Resolving due to being answered

closed 2020-05-20 by MushMal