awslabs/amazon-kinesis-video-streams-producer-sdk-java #77: Build Docker image

sajadmaster, 2020-01-20:
Please update readme.md file in the following location:
amazon-kinesis-video-streams-producer-sdk-java/dockerscripts/readme.md 

Please mention that the ubuntu version specified in the first line of Dockerfile, MUST match with your ubuntu version in the following line of code:
FROM ubuntu:17.10

My ubuntu version is 18.01 and I got error while building the dock image. I solved the error by changing the ubuntu version to 18.01:
FROM ubuntu:18.01

zhiyua-git, 2020-01-27:
Hi @sajadmaster ,

Can you describe what error you were hitting while building the docker image? It shouldn't matter if it is the same OS version between container and host for docker. Were you copying out the library and run it on host? Thanks.

zhiyua-git, 2020-01-27:
Verified issue was cause by discontinued support for ubuntu 17.10 on docker, change merged into master branch. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/pull/79

closed 2020-01-27 by zhiyua-git