awslabs/amazon-kinesis-video-streams-producer-sdk-java #81: KinesisVideoProducerJNI UnsatisfiedLinkError

ramneetsingh, 2020-02-04:
Hi,
I am facing issue with
java.lang.UnsatisfiedLinkError: com.amazonaws.kinesisvideo.producer.jni.NativeKinesisVideoProducerJni.getNativeLibraryVersion()Ljava/lang/String;

I have referred to 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/16
but it does not solve the error

I've tried to set the native library path and LD_LIBRARY_PATH through command line/VM arguments, but nothing seems to work. Any help will be appreciated.

Below is the screenshot from the error
[image]

zhiyua-git, 2020-02-04:
Hi @ramneetsingh ,

Can you confirm what version of java SDK are you using? The libKinesisVideoProducer library loading would fail if the version doesn't match. Thanks.

disa6302, 2020-04-02:
Stale issue. Closing it. Feel free to reopen in the future.

closed 2020-04-02 by disa6302

BodapatlaKiranReddy, 2021-01-21:
@ramneetsingh @disa6302 

i am facing similar issue, any help appreciated.
i am using JDK 7 and provided KinesisVideoProducerJNI.dll library path through VM arguments.

Below is the screenshot of the error
[image]

aqiao, 2023-08-13:
@BodapatlaKiranReddy ,
I encountered the same issue. In my case ,since i modified producer-cpp-sdk for debug, after i replace the modified dylib with original one, error disappeared. Hope it will give your some hints