awslabs/amazon-kinesis-video-streams-producer-sdk-java #83: Where to configure connection to a usb camera?

suhailmoideenv, 2020-02-09:
Hi all,
Am trying to connect a usb camera from a Mac and stream to kinesis using java producer library.
Tried running the demo project and it got successfully connected to the kinesis but with sample video.
How or where do I configure my usb camera details for streaming data from cam to kinesis?
What is the use of kinesisvideoproducerJNI native library?

Highly appreciate your answers

zhiyua-git, 2020-02-09:
Hi @suhailmoideenv ,

We don't have a media pipeline integration in java client now. You will need to write a media pipeline integration with your USB camera as MediaSource like [ImageFileMediaSource](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/java/mediasource/file/ImageFileMediaSource.java#L44) and [get frames from media pipeline into KVS producer SDK](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/java/mediasource/file/ImageFrameSource.java#L89). For this path, you can use gstreamer java for the media pipeline integration, there are some [samples](https://github.com/gstreamer-java/gst1-java-examples/blob/master/src/main/java/org/freedesktop/gstreamer/examples/CameraTest.java).

Alternatively, if you just want to stream data from your USB camera into KVS, you can build C/C++ producer on your Mac following this [guide](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#alternate-option-to-build-the-sdk-and-sample-applications-using-libraries-installed-by-homebrew) and run kinesis_video_gstreamer_sample_app to do that. Thanks.

MushMal, 2020-02-10:
@suhailmoideenv, I am resolving this issue as a "Question". Please open another if you need information/help with the SDK

closed 2020-02-10 by MushMal

BodapatlaKiranReddy, 2021-02-10:
@suhailmoideenv @
Hi,
The same question i  have now, i know it's quite along time but wanted to know if  you find any progress on the suggestion given by MushMal.

I would really appreciate if u can help me here.