awslabs/amazon-kinesis-video-streams-producer-sdk-java #84: No media found Verify that you are streaming to us-east-1 and the fragment timecodes are correct, or download the Kinesis Video Streams Producer SDK to set up your device.

naxir, 2020-02-14:
[image]

Any help please!

disa6302, 2020-02-14:
Hello @naxir ,

A couple of questions:

1. Are you currently running one of the applications in the producer SDK?

2. Can you post the logs here so that we can see what might be going on?

3. Which platform are you running the application in?

Thanks,
Divya

naxir, 2020-02-14:
Hi @disa6302 , 
https://github.com/awslabs/aws-sdk-android-samples 
I am following these sample code and using **AmazonKinesisVideoDemoApp** i have configured it successfully on my android mobile and i have created account and record a stream 
but facing above error as i mentioned

naxir, 2020-02-14:
[image]

you can check it the result

naxir, 2020-02-14:
@disa6302 
check it also i have created video stream 
[image]

when i am going to click it , in media playback section i am facing above error

MushMal, 2020-02-14:
@naxir Would you be able to get the logs from the producer side and attach?

naxir, 2020-02-17:
[image]

@MushMal  Please have a look

MushMal, 2020-02-17:
@naxir I don't see any logs attached. Could you please attach the logs from the producer device to ensure that you are able to stream in the video bits. I see that you are using Android demo app. If you could collect the logcat logs from the start of the application and into a few minutes into a streaming. This way we could see whether you are successfully streaming in before we see whats going on with the consumer (HLS/MPEG-DASH) side which in case of successful streaming might be the time selection.

naxir, 2020-02-18:
[image]

@MushMal  Please find this attached document

naxir, 2020-02-18:
@MushMal   & @disa6302  Thanks for helping 

I have fixed it Finally!!

closed 2020-02-18 by naxir

MushMal, 2020-02-18:
@naxir could you please update the thread on your findings? Looking at the small piece of the picture that contains a snippet of the log doesn't really reveal anything.

Joseph-TSS, 2021-03-27:
@naxir how did you finish please explain?

PhilipSlisenko, 2022-02-19:
I was following this guide: https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/producer-cpp/docker-rtsp and also faced this issue. In my case, I was indeed not streaming to the proper region as the error message suggested. Producer by default created a stream in us-west-2 and was streaming there (so you might want to delete that stream). I needed to add AWS_DEFAULT_REGION=<region-name> env var when starting producer to make it stream to the proper region.