awslabs/amazon-kinesis-video-streams-producer-sdk-java #85: save streams into as png format local disk

varun-tangoit, 2020-02-18:
Hi,

I try to save each frame into image png format, while i can able get live kinesis video viewer, but i don't know how to save images locally. I did piece code in FrameRendererVisitor module, but frames not save locally. Please help me out from this issue.
After that
AWTUtil.toBufferedImage(rgb, renderImage);
 try {
                    	System.out.println("writing into local disk");
                        ImageIO.write(renderImage, "png", new File(String.format("frame-capture-%s.png", UUID.randomUUID())));
                     } catch (IOException e) {
                        log.warn("Couldn't convert to a PNG", e);
                    }

MushMal, 2020-02-18:
@varun-tangoit what's the error/exception you are getting? Mind you, when you save the frame, it will be in the format that the original frame was put in. In case of H264 video, it will be an H264 frame and not a PNG. You will need to decode it first. 

Please provide more specific questions

MushMal, 2020-04-19:
Resolving as stale/answered.

closed 2020-04-19 by MushMal