awslabs/amazon-kinesis-video-streams-producer-sdk-java #87: I have successfully run AmazonKinesisVideoDemoApp and from mobile live video is showing on console preview , How Can i used this AmazonKinesisVideoDemoApp instead of mobile from IP Camera

naxir, 2020-02-18:
Hi , 
 I have run AmazonKinesisVideoDemoApp and now video is streaming live on console Preview 

What is the method to connect with Camera ??

Please guide me thanks 

@MushMal  any Help ?

MushMal, 2020-02-18:
@naxir I assume you are referring to the generic Java application? The purpose of the application is to simply demo the image file source as the media source. You could use any media source. Unfortunately, there is no generic camera and encoder media sources available for generic platform for Java. You could try looking into Java GStreamer or check out these: https://stackoverflow.com/questions/11487251/best-way-to-access-web-camera-in-java

naxir, 2020-02-19:
@MushMal  I am using this application of android  **AmazonKinesisVideoDemoApp**
I just want to connect with Camera by using this application

naxir, 2020-02-19:
There are three options 
  ```
 switch (cameraMediaSource.getCameraFacing()) {
            case FACING_BACK:
                return context.getString(R.string.camera_back_facing);
            case FACING_FRONT:
                return context.getString(R.string.camera_front_facing);
            case FACING_EXTERNAL:
                return context.getString(R.string.camera_external);
            default:
                return context.getString(R.string.camera, itemIndex);
        }
```
I assume this case is for our external camera's  **case FACING_EXTERNAL:**
if then how may i connect with the camera by using IP ?
@MushMal

MushMal, 2020-02-19:
Thanks for the clarification. Yes, the FACING_EXTERNAL is the wrapper for the new API LENSE_FACING_EXTERNAL enumeration that's part of Camera2 API.

https://source.android.com/devices/camera/external-usb-cameras

Unfortunately, we don't have detailed integration guide for external USB cameras. 

This said, it also seems that you are attempting to stream from an external IP camera? There is no ready platform level solution for Android I believe. You should see if there are 3rd party libraries that can provide some functionality to get the stream from your IP camera, provided you know what protocol they support. Once you have the frame information, you could create your own Media Source which you could put the frame into.

naxir, 2020-02-20:
Yes @MushMal I am attempting to stream from an external Camera IP address

MushMal, 2020-02-20:
Perhaps it's a better question to ask in an Android forum?

naxir, 2020-02-24:
Yes @MushMal

closed 2020-02-24 by naxir