awslabs/amazon-kinesis-video-streams-producer-sdk-java #89: DEBUG / KinesisVideo: Received end-of-stream for ACKs. DEBUG / KinesisVideo: Finished reading ACKs stream DEBUG / KinesisVideo: Received all data, close INFO / KinesisVideo: Complete callback triggered for sampleKVS4 with statuscode 200

hameedriswan7, 2020-03-05:
The above exception i have faced in while using Putmedia for sending RTSP camera to aws. in beginning it ran properly in the middle frame not sending and log moves to Callback method in Default class and ACK. Can i know reason for why callback method calling.what are the possibilities. 

below here is complete log.

DEBUG / KinesisVideo: PutFrame index: 1026, pts: 1126753754, dts: 1126753754, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 820, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 820 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 820 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1027, pts: 1127755066, dts: 1127755066, duration: 200000, keyFrame: true, track: 1
DEBUG / KinesisVideo: Kinesis Video client and stream metrics
>> Overall storage size: 1073741824
>> Available storage size: 1073610500
>> Allocated storage size: 131324
>> Total view allocation size: 172872
>> Total streams frame rate: 9
>> Total streams transfer rate: 29766
>> Current view duration: 0
>> Overall view duration: 31240000
>> Current view size: 0
>> Overall view size: 129756
>> Current frame rate: 9.961970510498777
>> Current transfer rate: 29766
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 100539, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 100539 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 100539 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: PutFrame index: 1028, pts: 1128756379, dts: 1128756379, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 1084, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 1084 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 1084 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: Received ACK bits: 76
{"EventType":"RECEIVED","FragmentTimecode":108900,"FragmentNumber":"91343852333181491972260273353598806020266865294"}

DEBUG / KinesisVideo: Received ACK bits: 77
{"EventType":"BUFFERING","FragmentTimecode":112104,"FragmentNumber":"91343852333181491977212033510740328002211633275"}

DEBUG / KinesisVideo: PutFrame index: 1029, pts: 1129757687, dts: 1129757687, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 920, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 920 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 920 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1030, pts: 1130758995, dts: 1130758995, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 846, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 846 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 846 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1031, pts: 1131760303, dts: 1131760303, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 768, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 768 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 768 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1032, pts: 1132761612, dts: 1132761612, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 687, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 687 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 687 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1033, pts: 1133762921, dts: 1133762921, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 754, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 754 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 754 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: Received ACK bits: 77
{"EventType":"PERSISTED","FragmentTimecode":108900,"FragmentNumber":"91343852333181491972260273353598806020266865294"}

DEBUG / KinesisVideo: PutFrame index: 1034, pts: 1134764230, dts: 1134764230, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 772, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 772 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 772 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1035, pts: 1135765520, dts: 1135765520, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 764, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 764 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 764 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1036, pts: 1136766811, dts: 1136766811, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 745, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 745 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 745 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1037, pts: 1137768102, dts: 1137768102, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 844, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 844 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 844 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting
DEBUG / KinesisVideo: PutFrame index: 1038, pts: 1138769394, dts: 1138769394, duration: 200000, keyFrame: false, track: 1
DEBUG / KinesisVideo: Data availability notification. Upload handle: 0, Size: 865, Duration 200000
DEBUG / KinesisVideo: getStreamData fill 865 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: Streamed 865 bytes for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0
DEBUG / KinesisVideo: no data for stream KVSJ2ejq3g9zdAMbQC with uploadHandle 0, waiting

DEBUG / KinesisVideo: Received end-of-stream for ACKs.
DEBUG / KinesisVideo: Finished reading ACKs stream
DEBUG / KinesisVideo: Received all data, close
INFO / KinesisVideo: Complete callback triggered for KVSJ2ejq3g9zdAMbQC with statuscode 200

zhiyua-git, 2020-03-05:
Hi @hameedriswan7 ,

DEBUG / KinesisVideo: Received end-of-stream for ACKs.
indicates you received http 0 from KVS service, it is possible that your media pipeline stop generating frames for some reason and then backend will disconnect after it get idle timeout after 30 seconds. If you have frames keep coming, you should see DEBUG / KinesisVideo: PutFrame showing up continuously. Thanks.

hameedriswan7, 2020-03-06:
@zhiyua-git  okey thanks. it possible to hold the process start again if frames comes to prevent move to  end-of-stream state. i mean waiting for stream for restart.

hameedriswan7, 2020-03-09:
is there ant way to reproduce it? @zhiyua-git  @MushMal

zhiyua-git, 2020-03-09:
Hi @hameedriswan7 ,

Is there any way for you to know when the data will be stop flowing in? If you get a signal from your media pipeline that you know it is paused/stopped/restarting, you can call unregisterMediaSource() when you get the signal and registerMediaSource() when your pipeline restarts. Thanks.

hameedriswan7, 2020-03-10:
@zhiyua-git  yes thanks. i will check it.

MushMal, 2020-03-23:
Any further update on this issue?

MushMal, 2020-04-19:
Resolving as stale. Please feel free to reopen

closed 2020-04-19 by MushMal