awslabs/amazon-kinesis-video-streams-producer-sdk-java #90: The security token included in the request is invalid

java-Xs, 2020-03-06:
I run a demo which it is kinesis video stream, i create a user ,i get aaws_secret_access_key and aws_access_key_id, i use them to run this demo,i get an error,the log like this:

 nesisVideoException: Service call failed.: The security token included in the request is invalid. (Service: AmazonKinesisVideo; Status Code: 403; Error Code: UnrecognizedClientException; Request ID: e6810e7a-0f0f-4d5f-ade5-3fbef06b02d2)
ERROR / KinesisVideo: Kinesis Video service client returned an error com.amazonaws.services.kinesisvideo.model.AmazonKinesisVideoException: The security token included in the request is invalid. (Service: AmazonKinesisVideo; Status Code: 403; Error Code: UnrecognizedClientException; Request ID: e6810e7a-0f0f-4d5f-ade5-3fbef06b02d2). Reporting to Kinesis Video PIC.
DEBUG / KinesisVideo: calling describe stream: {StreamName: my-stream,}

zhiyua-git, 2020-03-06:
Hi @java-Xs ,

Did you follow the [steps](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java#launching-demoapp-sample-application) to set the access key/secret key? Please note if your credentials is temporary credentials with session token, you will need to provide that too. Thanks.

java-Xs, 2020-03-20:
hi @zhiyua-git ,access key/secret key  put the system path,and add it the file of bashrc ,then run java demo,an error appearance; I run a c++ demo, c++ application is good.i dont kown what they are different.

closed 2020-03-20 by java-Xs