awslabs/amazon-kinesis-video-streams-producer-sdk-java #91: Not able to run DemoApp

divyaJainPhilips, 2020-04-18:
getting below error while trying to run DemoApp
OS: ubuntu
ERROR / KinesisVideo: 2020-04-18T11:11Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: no KinesisVideoProducerJNI in java.library.path
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:59)

O/P of ldd command
	linux-vdso.so.1 (0x00007ffdfde96000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f201444a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2014232000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2013e41000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2013aa3000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2014a25000)

Tried to build Native library..build for native library also failed with below error
/home/ubuntu/DIVYA/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus/missing: line 81: aclocal-1.16: command not found
WARNING: 'aclocal-1.16' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
Makefile:934: recipe for target 'aclocal.m4' failed
CMakeFiles/project_log4cplus.dir/build.make:111: recipe for target 'build/src/project_log4cplus-stamp/project_log4cplus-build' failed
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/project_log4cplus.dir/all' failed
Makefile:83: recipe for target 'all' failed
make[3]: *** [aclocal.m4] Error 127
make[2]: *** [build/src/project_log4cplus-stamp/project_log4cplus-build] Error 2
make[1]: *** [CMakeFiles/project_log4cplus.dir/all] Error 2
make: *** [all] Error 2
CMake Error at CMake/Utilities.cmake:55 (message):
  CMake step for liblog4cplus failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:77 (build_dependency)

-- Configuring incomplete, errors occurred!

Can somebody please suggest how to fix this

MushMal, 2020-04-19:
@divyaJainPhilips the Readme has some information on how to get the demo to load the native JNI which is pre-built with the SDK under https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/tree/master/src/main/resources/lib

-Daws.accessKeyId=<YourAwsAccessKey> -Daws.secretKey=<YourAwsSecretKey> -Djava.library.path=<NativeLibraryPath>

There are a few platform specific binaries. Could you try those please?

divyaJainPhilips, 2020-04-19:
Already tried everything mentioned in the README, but still not able to run the app

MushMal, 2020-04-19:
You haven’t mentioned which platform you are using. Could you please include the full command line with the full path you are using including the current directory from which you are running the command. Please include the entire output from the execution of the command line.

divyaJainPhilips, 2020-04-20:
**Path**: /Documents/test/research/amazon-kinesis-video-streams-producer-sdk-java-master/
**command**: java -jar  target/amazon-kinesis-video-streams-producer-sdk-java-1.9.5-jar-with-dependencies.jar -Daws.accessKeyId=XXXX -Daws.secretKey=XXXX -Djava.library.path=src/main/resources/lib/mac/libKinesisVideoProducerJNI.dylib com.amazonaws.kinesisvideo.demoapp.DemoAppMain
**Platform**:JAVA
**OUTPUT**
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
ERROR / KinesisVideo: 2020-04-20T10:05Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: no KinesisVideoProducerJNI in java.library.path
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:59)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.initializeLibrary(NativeKinesisVideoProducerJni.java:1170)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:223)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(NativeKinesisVideoProducerJni.java:241)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(NativeKinesisVideoProducerJni.java:207)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initializeNewKinesisVideoProducer(NativeKinesisVideoClient.java:238)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initialize(NativeKinesisVideoClient.java:119)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:113)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:86)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:42)

MushMal, 2020-04-20:
By platform I meant the OS platform. I assume it's Linux as you are attempting to load the .so extension from src/main/resources/lib/mac/libKinesisVideoProducerJNI.so

Could you specify the fully qualified path to the so file?

divyaJainPhilips, 2020-04-20:
I tried both on MACOs mojave and ubuntu
Path of so file for ubuntu amazon-kinesis-video-streams-producer-sdk-java-master/src/main/resources/lib/ubuntu/libKinesisVideoProducerJNI.so

path for MAC amazon-kinesis-video-streams-producer-sdk-java-master/src/main/resources/lib/mac/libKinesisVideoProducerJNI. dylib

closed 2020-04-21 by divyaJainPhilips

jelling, 2020-06-04:
@divyaJainPhilips the problem is that you have the filename in the path, just put the path, meaning `-Djava.library.path=src/main/resources/lib/mac`

Amazon, you really need to improve the Readme.

MushMal, 2020-06-04:
We hear you. Sorry about this. BTW, this is an open source project and we welcome any contribution to the codebase! If this is something you can help with please don't hesitate to cut a PR.

jelling, 2020-06-04:
Well when people do submit PRs to fix the docs ([like this one from February](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/pull/82)) your team doesn't even merge them. Mind you, without that change the Readme instructions do not work.

I'm frustrated because I had the same issues with the documentation for the Kinesis producer library. So there's definitely a pattern and punting the fix to your customers (and not even merging their fixes) is a terrible developer experience.

MushMal, 2020-06-05:
This is a big gap indeed. We want to make sure that the changes undergo reviews and get LGTMs before merging. Let us try to revive this PR - seems to have gotten stale. Thanks for bringing to our attention.