awslabs/amazon-kinesis-video-streams-producer-sdk-java #92: unable to Add Meta Data in Video Stream Android

amitsharmakks, 2020-04-26:
I integrated Android Video Stream in our project to add meta Data with video Stream .

I want to add metaData in video Stream Please share code to add meta data in video stream android  .

Please guide us

I am using below reference to integrate kinesis Video Stream.

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-android-writecode.html

    private KinesisVideoClient mKinesisVideoClient;

  mKinesisVideoClient  = KinesisVideoAndroidClientFactory.createKinesisVideoClient(
                    getActivity(),
                    App.KINESIS_VIDEO_REGION,
                    getCredentialsProvider());

   mCameraMediaSource = (AndroidCameraMediaSource) mKinesisVideoClient
                    .createMediaSource(mStreamName, mConfiguration);

            mCameraMediaSource.setPreviewSurfaces(new Surface(previewTexture));

            mCameraMediaSource.start();

 client = new AmazonRekognitionClient(credentials,clientConfiguration);

                    CreateStreamProcessorRequest createStreamProcessorRequest = new CreateStreamProcessorRequest();
                    createStreamProcessorRequest.setRoleArn("");

                    KinesisVideoStream kinesisVideoStream = new KinesisVideoStream();
                    kinesisVideoStream.setArn("");

                    StreamProcessorInput streamProcessorInput = new StreamProcessorInput();
                    streamProcessorInput.setKinesisVideoStream(kinesisVideoStream);

                    createStreamProcessorRequest.setInput(streamProcessorInput);

                    StreamProcessorOutput streamProcessorOutput = new StreamProcessorOutput();

                    KinesisDataStream kinesisDataStream = new KinesisDataStream();
                    kinesisDataStream.setArn("");

                    streamProcessorOutput.setKinesisDataStream(kinesisDataStream);

                    createStreamProcessorRequest.setOutput(streamProcessorOutput);
                    createStreamProcessorRequest.setName("");

                    StreamProcessorSettings streamProcessorSettings = new StreamProcessorSettings();

                    FaceSearchSettings faceSearchSettings = new FaceSearchSettings();

                    faceSearchSettings.setCollectionId("");
                    faceSearchSettings.setFaceMatchThreshold((float) 90);

 streamProcessorSettings.setFaceSearch(faceSearchSettings);

                    createStreamProcessorRequest.setSettings(streamProcessorSettings);

// for create Stream
                    CreateStreamProcessorResult streamProcessor = client.createStreamProcessor(createStreamProcessorRequest);

//Start Stream
        StartStreamProcessorResult startStreamProcessorResult = client.startStreamProcessor(new StartStreamProcessorRequest().withName(""));

MushMal, 2020-04-28:
MediaSource has a method for getting MediaSourceSink object. You can use this object and produce metadata.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/d490f31c692b1b2e989c58554ef00890ee62b515/src/main/java/com/amazonaws/kinesisvideo/internal/client/mediasource/MediaSourceSink.java#L25

amitsharmakks, 2020-04-29:
I am adding below code to add meta data but not working , unable to get this meta data on 
CloudWatch > Log Groups > Streams for /aws/lambda/videokinesis

Android code :

    private KinesisVideoClient mKinesisVideoClient;
    private AndroidCameraMediaSource mCameraMediaSource;

            mKinesisVideoClient = KinesisVideoAndroidClientFactory.createKinesisVideoClient(
                    getActivity(),
                    App.KINESIS_VIDEO_REGION,
                    getCredentialsProvider());

            mCameraMediaSource = (AndroidCameraMediaSource) mKinesisVideoClient
                    .createMediaSource(mStreamName, mConfiguration);

            mCameraMediaSource.setPreviewSurfaces(new Surface(previewTexture));

            MediaSourceSink mediaSourceSink = mCameraMediaSource.getMediaSourceSink();

            mediaSourceSink.onFragmentMetadata("AmitKey","AmitValue",true);

            mCameraMediaSource.initialize(mediaSourceSink);

MushMal, 2020-04-29:
It's not clear who produces this metadata. Do you have a "consumer" application that uses GetMedia to retrieve the stream and parse it and output some data? This is the application that can produce into CW.

amitsharmakks, 2020-04-29:
I am using below application reference for video streaming :
https://github.com/awslabs/aws-sdk-android-samples/tree/master/AmazonKinesisVideoDemoApp

MushMal, 2020-04-29:
I see, cool. Have you familiarized yourself with the sample application? You need to modify it to suit your need. Once you are familiar with it and have made modifications, you can post questions so some of the community members could help.

From your original question, it seems that you have the answer. If your original question is answered, could you please go ahead and close this issue and open a new one specific to the problem you are seeing.

amitsharmakks, 2020-04-29:
I integrated kinesis video streaming module in my app . this module is working fine it's streaming 
video to console . but i need to add metadata then i added above code like

        MediaSourceSink mediaSourceSink = mCameraMediaSource.getMediaSourceSink();

        mediaSourceSink.onFragmentMetadata("AmitKey","AmitValue",true);

        mCameraMediaSource.initialize(mediaSourceSink);

it's correct or not .  If no then you can share me code to send metadata .

MushMal, 2020-04-29:
You need to add the metadata after the media source is initialized

amitsharmakks, 2020-04-29:
Like below code :
 MediaSourceSink mediaSourceSink = mCameraMediaSource.getMediaSourceSink();
    mCameraMediaSource.initialize(mediaSourceSink);
    mediaSourceSink.onFragmentMetadata("AmitKey","AmitValue",true);

MushMal, 2020-04-29:
I think there is some confusion. The sample you reference is only streaming the video (and can embed metadata). There is another sample that's doing face detection - that's the one that should act when parsing the stream and perform certain tasks on metadata.

amitsharmakks, 2020-04-30:
I want to pass meta data on video Streaming . and i shared code to pass meta data in stream can you confirm me ? it's correct or not .If not then please share me sample for that .

amitsharmakks, 2020-04-30:
I am sending meta data using below code.

         

```
   MediaSourceSink mediaSourceSink = mCameraMediaSource.getMediaSourceSink();
            mCameraMediaSource.initialize(mediaSourceSink);
            mediaSourceSink.onFragmentMetadata("device","AmitDeviceValue",true);
            mediaSourceSink.onFragmentMetadata("stream","AmitStreamValue",true);

```

 I am sending two tags (also already on  Kinesis Video Stream > Tags ) 

**Can you please help me to see metadata on aws console ?** As of now i am trying to see meta data on " CloudWatch > Log Groups > Streams for /aws/lambda/videokinesis" but not able to see metadata tags here.

Please let me know if i need to see meta data on some other place.

zhiyua-git, 2020-04-30:
The metadata added in onFragmentMetadata() is sent inside the streaming data. You can access it using parser library. One sample you can try on is the [render example](https://github.com/aws/amazon-kinesis-video-streams-parser-library#kinesisvideorendererexample), modify the stream name [here](https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/test/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoRendererExampleTest.java#L56) to view the metadata.

Do you really want metadata to come along with the streaming data or you just want the metadata (tagging) for stream itself? If you want the latter, then you would need to use [createKinesisVideoClient()](https://github.com/aws-amplify/aws-sdk-android/blob/3b9f66afab09acbd233b9269b9cf0b9906651f54/aws-android-sdk-kinesisvideo/src/main/java/com/amazonaws/mobileconnectors/kinesisvideo/client/KinesisVideoAndroidClientFactory.java#L124-L128) which accepts DeviceInfo and provide your tags info in your device info.

MushMal, 2020-05-20:
Resolving this issue, Please feel free to cut a separate issue or reopen if you need further assistance

closed 2020-05-20 by MushMal